matthias-bs/BresserWeatherSensorReceiver #112: Simple example for ESP32 DOIT devkit v1

elprocrastinador, 2023-11-27:
Hello.
I'm trying to get data from my Bresser 5in1 station with no luck. I'm testing both CC1101 and RFM95W with a ESP32 DOIT devkit v1 module.
PLEASE
Can you add an example for this module?
Thanks a lot.

matthias-bs, 2023-11-27:
Hi,

I cannot provide examples for all possible board/transceiver combinations.

Please follow this recipe:
* Refer to [Manual Configuration](https://github.com/matthias-bs/BresserWeatherSensorReceiver#manual-configuration).
* Please check that you select GPIO pins which are not connected to anything else (board's schematic)
* Check the SPI pins (board documentation, schematic, https://github.com/espressif/arduino-esp32/tree/master/variants)
* Check your own wiring (at least twice!)
* **Keep in mind that the ArduinoIDE might chose to use .cpp/.h files from `Arduino/libraries/BresserWeatherSensorReceiver` rather than from your sketch directory!**
* Modify WeatherSensorCfg.h
   * Set the correct radio transceiver in the section "Radio Transceiver"
   * Define the pins according to your setup in the section starting with
     ```
     #elif defined(ESP32)
        // Generic pinning for ESP32 development boards
     ```
* Check the files actually used and the pin definitions actually used in the compiler output - 
   enable  "Show verbose output during compile" in Preferences for this

If you need more help, please create a new issue and provide as much information on your setup (schematic, wiring diargram, `WeatherSensorCfg.h`, ArduinoIDE settings, folder structure, ...) as possible.

elprocrastinador, 2023-11-27:
Thanks Matthias.
I will follow your recipe hoping a tasty result.

matthias-bs, 2023-11-28:
Please refer to https://github.com/matthias-bs/BresserWeatherSensorReceiver/issues/91#issuecomment-1830529398

matthias-bs, 2023-11-29:
See https://github.com/matthias-bs/BresserWeatherSensorReceiver/issues/91#issuecomment-1832549188

elprocrastinador, 2023-12-02:
[image]

matthias-bs, 2023-12-02:
Happy to read this!

elprocrastinador, 2023-12-02:
### HARDWARE
ESP32 DEVKIT V1 + CC1101
Connections
ESP32 - CC1101
3.3V - VCC
GND - GND
23 - MOSI
18 - SCLK
19 - MISO
21 - GDO0
33 - GDO2
27 - CSN

### SOFTWARE
Visual Studio Code + Platformio

BresserWeatherSensorBasic.ino

### ------WeatherSensorCfg.h------

### Weather Sensor
Line 83:
```
// Select appropriate sensor message format(s)
// Comment out unused decoders to save operation time/power
#define BRESSER_5_IN_1
//#define BRESSER_6_IN_1
//#define BRESSER_7_IN_1
//#define BRESSER_LIGHTNING
//#define BRESSER_LEAKAGE
```
### Board
Line 198:
```
#elif defined(ARDUINO_ESP32_DEV)
    //#define LORAWAN_NODE
    //#define FIREBEETLE_ESP32_COVER_LORA
```

### Radio Transceiver
Line 222:
```
#if ( !defined(USE_CC1101) && !defined(USE_SX1276) && !defined(USE_SX1262) )
    #define USE_CC1101
    //#define USE_SX1276
    //#define USE_SX1262
#endif
```

###platformio.ini

```
[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino
monitor_speed = 115200 
lib_deps =
  matthias-bs/BresserWeatherSensorReceiver @ ^0.19.6
  jgromes/RadioLib @ 6.1.0
```

matthias-bs, 2023-12-14:
Seems to be solved.

closed 2023-12-14 by matthias-bs

elprocrastinador, 2024-03-11:
Sorry for reopen this thread.

Notes for using RFM95W with ESP32 DOIT devkit v1.

HARDWARE

ESP32 DEVKIT V1 +RFM95W
Connections
ESP32 - RFM95W
3.3V - VCC
GND - GND
23 - MOSI
18 - SCK
19 - MISO
21 - DIO0
33 - DIO2
27 - NSS
32 - RESET

One more connection than CC1101

WeatherSensorCfg.h  -  LINE 198
    #elif defined(ARDUINO_ESP32_DEV)
    //#define LORAWAN_NODE
    //#define FIREBEETLE_ESP32_COVER_LORA

WeatherSensorCfg.h - LINE 222
#if ( !defined(USE_CC1101) && !defined(USE_SX1276) && !defined(USE_SX1262) )
    //#define USE_CC1101
    #define USE_SX1276
    //#define USE_SX1262
#endif

matthias-bs, 2024-03-11:
Could you please explain what you would like to be modified or create a PR?

elprocrastinador, 2024-03-11:
The entry is not about a PR or a bug. 
Its only some info for beginners like me.
If you think that this is not the correct place, just tell me.

matthias-bs, 2024-03-11:
> The entry is not about a PR or a bug. 
> Its only some info for beginners like me.
> If you think that this is not the correct place, just tell me.

I think an info should rather be put in the README file.