matthias-bs/BresserWeatherSensorReceiver #117: Upgraded to 0.20.0 from 0.19.5, now no data is received

Calimerorulez, 2023-12-09:
Hi,

Before running 0.20.0 in my LilyGo TTGo Lora32 Oled v2.1, with my 7 in 1 sensor I was receiving data:

```
[ 83153][D][WeatherSensor.cpp:251] getMessage(): [SX1276] R [D4] RSSI: -73.0
[ 83154][D][WeatherSensor.h:494] log_message():           Byte #: 00 01 02 03 04 05 06 07 08 09 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 
[ 83162][D][WeatherSensor.h:504] log_message(): De-whitened Data: 4C C3 AB E9 10 80 B2 02 80 25 00 12 92 00 07 70 92 00 03 77 00 00 00 00 00 AA 
[ 83175][D][WeatherSensor.cpp:403] findSlot(): sensor[0]: v=0 id=0x0000ABE9 t=1 c=0
[ 95653][D][WeatherSensor.cpp:251] getMessage(): [SX1276] R [D4] RSSI: -70.0
[ 95654][D][WeatherSensor.h:494] log_message():           Byte #: 00 01 02 03 04 05 06 07 08 09 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 
[ 95662][D][WeatherSensor.h:504] log_message(): De-whitened Data: 20 CF AB E9 13 80 B2 03 60 34 00 12 92 00 07 70 92 00 03 77 00 00 00 00 00 AA 
[ 95674][D][WeatherSensor.cpp:403] findSlot(): sensor[0]: v=0 id=0x0000ABE9 t=1 c=0
[ 95684][I][BresserWeatherSensorMQTTWifiMgr-bak.ino:803] publishWeatherdata(): ESPWeather-9EB8C4/abe9/data: {"id":44009,"ch":0,"battery_ok":1,"temp_c":7.7,"humidity":92,"wind_gust":3.6,"wind_avg":3.4,"wind_dir":138.0,"uv":0.0,"light_klx":0.4,"rain":129.2,"rain_h":0.0,"rain_d":0.0,"rain_w":0.0,"rain_m":0.0}
```

Now running 0.20.0 I only see the Byte # and the De-whitened Data, but no weather data anymore:

```
[  9173][D][WeatherSensor.cpp:301] getMessage(): [SX1276] R [D4] RSSI: -74.5
[  9174][D][WeatherSensor.h:475] log_message():           Byte #: 00 01 02 03 04 05 06 07 08 09 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 
[  9182][D][WeatherSensor.h:485] log_message(): De-whitened Data: 16 55 AB E9 13 80 B2 02 80 25 00 12 96 00 07 70 92 00 02 55 00 00 00 00 00 AA 
[  9194][D][WeatherSensor.cpp:472] findSlot(): sensor[0]: v=0 id=0x00000000 t=0 c=0
```

In both versions I've disabled sleep, and in 0.20.0 it seems that the findSlot function has always values whit zeroes.

matthias-bs, 2023-12-10:
Hi,

which RadioLib version are you using? Could you please provide a longer debug logfile and a compile log?

Calimerorulez, 2023-12-11:
Hi @matthias-bs,

I will provide the info a.s.a.p. as I'm away a couple of days. If I recall correctly, my Arduino IDE did update the RadioLib automatically to the latest available version.

Calimerorulez, 2023-12-15:
Ok, small update. Running 0.19.5 with RadioLib 6.3.0 works, 0.20.0 with RadioLib 6.3.0 doesn't.

In function findSlot() sensor[0] stays empty:

```
[  7849][V][WeatherSensor.cpp:299] getMessage(): [SX1276] Data: D4 D9 43 01 43 9A CA 18 AA AA AA AA BC 98 AA BA BA 28 AA EB 88 AA AA AA AA AA 00 
[  7851][D][WeatherSensor.cpp:301] getMessage(): [SX1276] R [D4] RSSI: -74.5
[  7858][D][WeatherSensor.h:475] log_message():           Byte #: 00 01 02 03 04 05 06 07 08 09 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 
[  7871][D][WeatherSensor.h:485] log_message(): De-whitened Data: 73 E9 AB E9 30 60 B2 00 00 00 00 16 32 00 10 10 82 00 41 22 00 00 00 00 00 AA 
[  7883][V][WeatherSensor.cpp:429] findSlot(): find_slot(): ID=0000ABE9
[  7890][D][WeatherSensor.cpp:472] findSlot(): sensor[0]: v=0 id=0x00000000 t=0 c=0
[  7897][V][WeatherSensor.cpp:497] findSlot(): find_slot(): Storing into slot #0
```

matthias-bs, 2023-12-16:
Hi,

I cannot reproduce this. For testing, I fed the message data from your message above to SensorTransmitter and correctly received it using [BresserWeatherSensorBasic.ino](examples/BresserWeatherSensorBasic/BresserWeatherSensorBasic.ino):

```
[1086505][D][WeatherSensor.cpp:301] getMessage(): [SX1276] R [D4] RSSI: -107.0
[1086505][D][WeatherSensor.h:475] log_message():           Byte #: 00 01 02 03 04 05 06 07 08 09 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 
[1086514][D][WeatherSensor.h:485] log_message(): De-whitened Data: 8A EE AB E9 21 00 B2 00 00 00 00 16 32 00 10 00 82 00 30 60 00 00 00 00 00 AA 
[1086526][D][WeatherSensor.cpp:472] findSlot(): sensor[0]: v=0 id=0x39582376 t=1 c=0
[1086534][D][WeatherSensor.cpp:472] findSlot(): sensor[1]: v=0 id=0x00000000 t=0 c=0
[1086541][D][WeatherSensor.cpp:472] findSlot(): sensor[2]: v=0 id=0x00000000 t=0 c=0
[1086549][D][WeatherSensor.cpp:472] findSlot(): sensor[3]: v=0 id=0x00000000 t=0 c=0
Id: [    ABE9] Typ: [1] Ch: [0] St: [0] Bat: [OK ] RSSI: [-107.0dBm] Temp: [ 10.0C] Hum: [ 82%] Wmax: [ 0.0m/s] Wavg: [ 0.0m/s] Wdir: [210.0deg] Rain: [  163.2mm] UVidx: [0.0] Light: [3.1klx] 
[1090379][D][WeatherSensor.cpp:301] getMessage(): [SX1276] R [D4] RSSI: -104.5
[1090379][D][WeatherSensor.cpp:1051] decodeBresser7In1Payload(): Digest check failed - [061D] vs [3FE5] (39F8)
[1090385][D][WeatherSensor.cpp:846] decodeBresser6In1Payload(): Digest check failed - [ACB7] != [2FBE]
[1090394][D][WeatherSensor.cpp:643] decodeBresser5In1Payload(): Parity wrong at column 0
[1090402][D][WeatherSensor.cpp:1186] decodeBresserLightningPayload(): Digest check failed - [061D] vs [0E42] (085F)
[1090412][D][WeatherSensor.cpp:472] findSlot(): sensor[0]: v=0 id=0x0000ABE9 t=1 c=0
[1090419][D][WeatherSensor.cpp:472] findSlot(): sensor[1]: v=0 id=0x00000000 t=0 c=0
[1090427][D][WeatherSensor.cpp:472] findSlot(): sensor[2]: v=0 id=0x00000000 t=0 c=0
[1090434][D][WeatherSensor.cpp:472] findSlot(): sensor[3]: v=0 id=0x00000000 t=0 c=0
```

Calimerorulez, 2023-12-17:
Hi,

I'm using BresserWeatherSensorMQTTWifiMgr.ino

Does that make any difference?

Calimerorulez, 2023-12-17:
OK,

I just removed and reinstalled the Arduino IDE on my PC, downloaded the code again, and now the receiver works :)

But only one message is received:

```
19:56:59.078 -> Saving config file...
19:56:59.078 -> Saving config file failed
19:57:10.080 -> [ 16020][V][WeatherSensor.cpp:299] getMessage(): [SX1276] Data: D4 1D C3 01 43 AC CA 18 AA 2A A2 AA BC 98 AA AD BA 28 AA AA AA AA AA AA AA AA 00 
19:57:10.152 -> [ 16021][D][WeatherSensor.cpp:301] getMessage(): [SX1276] R [D4] RSSI: -67.0
19:57:10.152 -> [ 16028][D][WeatherSensor.h:475] log_message():           Byte #: 00 01 02 03 04 05 06 07 08 09 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 
19:57:10.152 -> [ 16041][D][WeatherSensor.h:485] log_message(): De-whitened Data: B7 69 AB E9 06 60 B2 00 80 08 00 16 32 00 07 10 82 00 00 00 00 00 00 00 00 AA 
19:57:10.164 -> [ 16053][V][WeatherSensor.cpp:429] findSlot(): find_slot(): ID=0000ABE9
19:57:10.164 -> [ 16060][D][WeatherSensor.cpp:472] findSlot(): sensor[0]: v=0 id=0x00000000 t=0 c=0
19:57:10.164 -> [ 16067][V][WeatherSensor.cpp:497] findSlot(): find_slot(): Storing into slot #0
19:57:10.164 -> [ 16077][I][BresserWeatherSensorMQTTWifiMgr.ino:823] publishWeatherdata(): ESPWeather-9EB8C4/abe9/data: {"id":44009,"ch":0,"battery_ok":1,"temp_c":7.1,"humidity":82,"wind_gust":0.8,"wind_avg":0.8,"wind_dir":66.0,"uv":0.0,"light_klx":0.0,"rain":163.2,"rain_h":0.0,"rain_d":0.0,"rain_w":0.0,"rain_m":0.0}
19:57:10.179 -> 
19:57:10.179 -> [ 16105][I][BresserWeatherSensorMQTTWifiMgr.ino:839] publishWeatherdata(): ESPWeather-9EB8C4/extra: {"wind_dir_txt":"ENE","wind_gust_bft":0,"wind_avg_bft":0,"dewpoint_c":4.2,"perceived_temp_c":7.1}
19:57:10.212 -> 
19:58:40.203 -> [106119][I][BresserWeatherSensorMQTTWifiMgr.ino:974] loop(): ESPWeather-9EB8C4/status: online
19:58:40.203 -> 
19:58:40.203 -> [106122][I][BresserWeatherSensorMQTTWifiMgr.ino:856] publishRadio(): ESPWeather-9EB8C4/radio: {"rssi":-67}
19:58:40.203 -> 
20:00:10.193 -> [196127][I][BresserWeatherSensorMQTTWifiMgr.ino:974] loop(): ESPWeather-9EB8C4/status: online
20:00:10.226 -> 
20:00:10.226 -> [196130][I][BresserWeatherSensorMQTTWifiMgr.ino:856] publishRadio(): ESPWeather-9EB8C4/radio: {"rssi":-67}
```

Only the line with `publishRadio()` is shown.

matthias-bs, 2023-12-17:
I will try to look into this later.

matthias-bs, 2023-12-18:
Did you set `#define SLEEP_EN false`?
If so, you could try `#define SLEEP_EN true` or comment out all `radio.sleep();` statements in `getData()` (`WeatherSensor.cpp`).

I'll investigate this further and come up with a proper solution if my suspicion is correct.

Calimerorulez, 2023-12-18:
> Did you set `#define SLEEP_EN false`? If so, you could try `#define SLEEP_EN true` or comment out all `radio.sleep();` statements in `getData()` (`WeatherSensor.cpp`).
> 

Hi,

Yes, I've set `#define SLEEP_EN true`. I will comment out all  `radio.sleep();` statements in `getData()` (`WeatherSensor.cpp`) and report back.

> I'll investigate this further and come up with a proper solution if my suspicion (receiver will not be woken up by `radio.startReceive()` is correct.

Calimerorulez, 2023-12-18:
Yes :)

Commenting out seems to solve it 💯 

```
08:48:00.451 -> [ 49794][D][WeatherSensor.cpp:472] findSlot(): sensor[0]: v=0 id=0x0000ABE9 t=1 c=0
08:48:00.451 -> [ 49801][V][WeatherSensor.cpp:497] findSlot(): find_slot(): Storing into slot #0
08:48:00.451 -> [ 49811][I][BresserWeatherSensorMQTTWifiMgr.ino:823] publishWeatherdata(): ESPWeather-9EB8C4/abe9/data: {"id":44009,"ch":0,"battery_ok":1,"temp_c":8.0,"humidity":87,"wind_gust":0.8,"wind_avg":0.8,"wind_dir":312.0,"uv":0.0,"light_klx":0.5,"rain":163.2,"rain_h":0.0,"rain_d":0.0,"rain_w":0.0,"rain_m":0.0}
08:48:00.462 -> 
08:48:00.462 -> [ 49839][I][BresserWeatherSensorMQTTWifiMgr.ino:839] publishWeatherdata(): ESPWeather-9EB8C4/extra: {"wind_dir_txt":"NW","wind_gust_bft":0,"wind_avg_bft":0,"dewpoint_c":6.0,"perceived_temp_c":8.0}
08:48:00.462 -> 
08:48:12.865 -> [ 62254][V][WeatherSensor.cpp:299] getMessage(): [SX1276] Data: D4 C0 D9 01 43 9B 8A 18 AA CA AC AA BC 98 AA A2 AA 2D AA AE F8 AA AA AA AA AA 00 
08:48:12.865 -> [ 62255][D][WeatherSensor.cpp:301] getMessage(): [SX1276] R [D4] RSSI: -68.5
08:48:12.865 -> [ 62262][D][WeatherSensor.h:475] log_message():           Byte #: 00 01 02 03 04 05 06 07 08 09 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 
08:48:12.865 -> [ 62275][D][WeatherSensor.h:485] log_message(): De-whitened Data: 6A 73 AB E9 31 20 B2 00 60 06 00 16 32 00 08 00 87 00 04 52 00 00 00 00 00 AA 
08:48:12.973 -> [ 62287][V][WeatherSensor.cpp:429] findSlot(): find_slot(): ID=0000ABE9
08:48:12.973 -> [ 62294][D][WeatherSensor.cpp:472] findSlot(): sensor[0]: v=0 id=0x0000ABE9 t=1 c=0
08:48:12.973 -> [ 62301][V][WeatherSensor.cpp:497] findSlot(): find_slot(): Storing into slot #0
08:48:25.347 -> [ 74753][V][WeatherSensor.cpp:299] getMessage(): [SX1276] Data: D4 D7 52 01 43 99 AA 18 AA AA AA AA BC 98 AA A2 AA 2D AA AE F8 AA AA AA AA AA 00 
08:48:25.347 -> [ 74755][D][WeatherSensor.cpp:301] getMessage(): [SX1276] R [D4] RSSI: -68.5
08:48:25.442 -> [ 74762][D][WeatherSensor.h:475] log_message():           Byte #: 00 01 02 03 04 05 06 07 08 09 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 
08:48:25.442 -> [ 74775][D][WeatherSensor.h:485] log_message(): De-whitened Data: 7D F8 AB E9 33 00 B2 00 00 00 00 16 32 00 08 00 87 00 04 52 00 00 00 00 00 AA 
08:48:25.442 -> [ 74787][V][WeatherSensor.cpp:429] findSlot(): find_slot(): ID=0000ABE9
08:48:25.455 -> [ 74793][D][WeatherSensor.cpp:472] findSlot(): sensor[0]: v=0 id=0x0000ABE9 t=1 c=0
08:48:25.455 -> [ 74801][V][WeatherSensor.cpp:497] findSlot(): find_slot(): Storing into slot #0
08:48:25.455 -> [ 74810][I][BresserWeatherSensorMQTTWifiMgr.ino:823] publishWeatherdata(): ESPWeather-9EB8C4/abe9/data: {"id":44009,"ch":0,"battery_ok":1,"temp_c":8.0,"humidity":87,"wind_gust":0.0,"wind_avg":0.0,"wind_dir":330.0,"uv":0.0,"light_klx":0.5,"rain":163.2,"rain_h":0.0,"rain_d":0.0,"rain_w":0.0,"rain_m":0.0}
08:48:25.468 -> 
08:48:25.468 -> [ 74839][I][BresserWeatherSensorMQTTWifiMgr.ino:839] publishWeatherdata(): ESPWeather-9EB8C4/extra: {"wind_dir_txt":"NNW","wind_gust_bft":0,"wind_avg_bft":0,"dewpoint_c":6.0,"perceived_temp_c":8.0}
08:48:25.468 -> 
08:48:25.468 -> [ 74853][I][BresserWeatherSensorMQTTWifiMgr.ino:974] loop(): ESPWeather-9EB8C4/status: online
08:48:25.487 -> 
08:48:25.487 -> [ 74861][I][BresserWeatherSensorMQTTWifiMgr.ino:856] publishRadio(): ESPWeather-9EB8C4/radio: {"rssi":-68.5}
08:48:25.487 -> 
08:48:37.836 -> [ 87254][V][WeatherSensor.cpp:299] getMessage(): [SX1276] Data: D4 9F A6 01 43 A3 CA 18 AB DA BF AA BC 98 AA A2 AA 2D AA AE F8 AA AA AA AA AA 00 
08:48:37.932 -> [ 87255][D][WeatherSensor.cpp:301] getMessage(): [SX1276] R [D4] RSSI: -68.5
08:48:37.932 -> [ 87262][D][WeatherSensor.h:475] log_message():           Byte #: 00 01 02 03 04 05 06 07 08 09 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 
08:48:37.932 -> [ 87275][D][WeatherSensor.h:485] log_message(): De-whitened Data: 35 0C AB E9 09 60 B2 01 70 15 00 16 32 00 08 00 87 00 04 52 00 00 00 00 00 AA 
08:48:37.932 -> [ 87287][V][WeatherSensor.cpp:429] findSlot(): find_slot(): ID=0000ABE9
08:48:37.932 -> [ 87294][D][WeatherSensor.cpp:472] findSlot(): sensor[0]: v=0 id=0x0000ABE9 t=1 c=0
08:48:37.932 -> [ 87301][V][WeatherSensor.cpp:497] findSlot(): find_slot(): Storing into slot #0

```

matthias-bs, 2023-12-18:
Thank you for pointing this out!
Fixed in [v0.20.1](https://github.com/matthias-bs/BresserWeatherSensorReceiver/releases/tag/v0.20.1)

Calimerorulez, 2023-12-18:
Thank you for the quick fix. I Will update tomorrow 😄 

But I'm confident it will work just fine.

Calimerorulez, 2023-12-19:
@matthias-bs 

Works ok :)

closed 2023-12-19 by Calimerorulez