matthias-bs/BresserWeatherSensorReceiver #120: Compile Error Wemos D1 mini with CC1101 with MQTT custom

kroon040, 2023-12-29:
**Context**
- Which example sketch are you using?
 BresserWeatherSensorMQTTCustom.ino
- Which changes to the code did you make (if any)?
username password, only the defaukts 
- Which sensor are you using?
`// ------------------------------------------------------------------------------------------------
// --- Radio Transceiver ---
// ------------------------------------------------------------------------------------------------
// Select type of receiver module (if not yet defined based on the assumptions above)
#if ( !defined(USE_CC1101) && !defined(USE_SX1276) && !defined(USE_SX1262) )
    #define USE_CC1101
    //#define USE_SX1276
    //#define USE_SX1262
#endif`
**Describe the bug**
A clear and concise description of what the bug is.
`/home/richard/Arduino/BresserWeatherSensorMQTTCustom/RainGauge.cpp:58:70: note: '#pragma message: RainGauge with SLEEP_EN only supported on ESP32!'
   58 |    #pragma message("RainGauge with SLEEP_EN only supported on ESP32!")
      |                                                                      ^
In file included from /home/richard/Arduino/BresserWeatherSensorMQTTCustom/WeatherSensor.cpp:87:
/home/richard/Arduino/BresserWeatherSensorMQTTCustom/WeatherSensorCfg.h:518:48: note: '#pragma message: Receiver chip: [CC1101]'
  518 | #pragma message("Receiver chip: " RECEIVER_CHIP)
      |                                                ^
/home/richard/Arduino/BresserWeatherSensorMQTTCustom/WeatherSensorCfg.h:519:170: note: '#pragma message: Pin config: RST->RADIOLIB_NC, CS->15, GD0/G0/IRQ->4, GDO2/G1/GPIO->5'
  519 | #pragma message("Pin config: RST->" STR(PIN_RECEIVER_RST) ", CS->" STR(PIN_RECEIVER_CS) ", GD0/G0/IRQ->" STR(PIN_RECEIVER_IRQ) ", GDO2/G1/GPIO->" STR(PIN_RECEIVER_GPIO) )
      |                                                                                                                                                                          ^
In file included from /home/richard/Arduino/BresserWeatherSensorMQTTCustom/BresserWeatherSensorMQTTCustom.ino:190:
/home/richard/Arduino/BresserWeatherSensorMQTTCustom/src/WeatherSensorCfg.h:518:48: note: '#pragma message: Receiver chip: [CC1101]'
  518 | #pragma message("Receiver chip: " RECEIVER_CHIP)
      |                                                ^
/home/richard/Arduino/BresserWeatherSensorMQTTCustom/src/WeatherSensorCfg.h:519:170: note: '#pragma message: Pin config: RST->RADIOLIB_NC, CS->15, GD0/G0/IRQ->4, GDO2/G1/GPIO->5'
  519 | #pragma message("Pin config: RST->" STR(PIN_RECEIVER_RST) ", CS->" STR(PIN_RECEIVER_CS) ", GD0/G0/IRQ->" STR(PIN_RECEIVER_IRQ) ", GDO2/G1/GPIO->" STR(PIN_RECEIVER_GPIO) )
      |                                                                                                                                                                          ^
/home/richard/Arduino/BresserWeatherSensorMQTTCustom/src/RainGauge.cpp:58:70: note: '#pragma message: RainGauge with SLEEP_EN only supported on ESP32!'
   58 |    #pragma message("RainGauge with SLEEP_EN only supported on ESP32!")
      |                                                                      ^
In file included from /home/richard/Arduino/BresserWeatherSensorMQTTCustom/src/WeatherSensor.cpp:87:
/home/richard/Arduino/BresserWeatherSensorMQTTCustom/src/WeatherSensorCfg.h:518:48: note: '#pragma message: Receiver chip: [CC1101]'
  518 | #pragma message("Receiver chip: " RECEIVER_CHIP)
      |                                                ^
/home/richard/Arduino/BresserWeatherSensorMQTTCustom/src/WeatherSensorCfg.h:519:170: note: '#pragma message: Pin config: RST->RADIOLIB_NC, CS->15, GD0/G0/IRQ->4, GDO2/G1/GPIO->5'
  519 | #pragma message("Pin config: RST->" STR(PIN_RECEIVER_RST) ", CS->" STR(PIN_RECEIVER_CS) ", GD0/G0/IRQ->" STR(PIN_RECEIVER_IRQ) ", GDO2/G1/GPIO->" STR(PIN_RECEIVER_GPIO) )
      |                                                                                                                                                                          ^
/home/richard/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /tmp/arduino/sketches/84A2F82A7D12FB5A4436CFA77D535BD8/sketch/src/RainGauge.cpp.o: in function `_ZN9RainGauge5resetEh':
/home/richard/Arduino/BresserWeatherSensorMQTTCustom/src/RainGauge.cpp:182: multiple definition of `_ZN9RainGauge5resetEh'; /tmp/arduino/sketches/84A2F82A7D12FB5A4436CFA77D535BD8/sketch/RainGauge.cpp.o:/home/richard/Arduino/BresserWeatherSensorMQTTCustom/RainGauge.cpp:182: first defined here
/home/richard/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /tmp/arduino/sketches/84A2F82A7D12FB5A4436CFA77D535BD8/sketch/src/RainGauge.cpp.o: in function `_ZN9RainGauge9timeStampE2tm':
/home/richard/Arduino/BresserWeatherSensorMQTTCustom/src/RainGauge.cpp:229: multiple definition of `_ZN9RainGauge9timeStampE2tm'; /tmp/arduino/sketches/84A2F82A7D12FB5A4436CFA77D535BD8/sketch/RainGauge.cpp.o:/home/richard/Arduino/BresserWeatherSensorMQTTCustom/RainGauge.cpp:229: first defined here
/home/richard/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /tmp/arduino/sketches/84A2F82A7D12FB5A4436CFA77D535BD8/sketch/src/RainGauge.cpp.o: in function `_ZN9RainGauge4initE2tmf':
/home/richard/Arduino/BresserWeatherSensorMQTTCustom/src/RainGauge.cpp:213: multiple definition of `_ZN9RainGauge4initE2tmf'; /tmp/arduino/sketches/84A2F82A7D12FB5A4436CFA77D535BD8/sketch/RainGauge.cpp.o:/home/richard/Arduino/BresserWeatherSensorMQTTCustom/RainGauge.cpp:213: first defined here
/home/richard/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /tmp/arduino/sketches/84A2F82A7D12FB5A4436CFA77D535BD8/sketch/src/RainGauge.cpp.o: in function `_ZN9RainGauge6updateE2tmfbf':
/home/richard/Arduino/BresserWeatherSensorMQTTCustom/src/RainGauge.cpp:248: multiple definition of `_ZN9RainGauge6updateE2tmfbf'; /tmp/arduino/sketches/84A2F82A7D12FB5A4436CFA77D535BD8/sketch/RainGauge.cpp.o:/home/richard/Arduino/BresserWeatherSensorMQTTCustom/RainGauge.cpp:248: first defined here
/home/richard/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /tmp/arduino/sketches/84A2F82A7D12FB5A4436CFA77D535BD8/sketch/src/RainGauge.cpp.o: in function `_ZN9RainGauge8pastHourEv':
/home/richard/Arduino/BresserWeatherSensorMQTTCustom/src/RainGauge.cpp:341: multiple definition of `_ZN9RainGauge8pastHourEv'; /tmp/arduino/sketches/84A2F82A7D12FB5A4436CFA77D535BD8/sketch/RainGauge.cpp.o:/home/richard/Arduino/BresserWeatherSensorMQTTCustom/RainGauge.cpp:341: first defined here
/home/richard/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /tmp/arduino/sketches/84A2F82A7D12FB5A4436CFA77D535BD8/sketch/src/RainGauge.cpp.o: in function `_ZN9RainGauge10currentDayEv':
/home/richard/Arduino/BresserWeatherSensorMQTTCustom/src/RainGauge.cpp:348: multiple definition of `_ZN9RainGauge10currentDayEv'; /tmp/arduino/sketches/84A2F82A7D12FB5A4436CFA77D535BD8/sketch/RainGauge.cpp.o:/home/richard/Arduino/BresserWeatherSensorMQTTCustom/RainGauge.cpp:348: first defined here
/home/richard/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /tmp/arduino/sketches/84A2F82A7D12FB5A4436CFA77D535BD8/sketch/src/RainGauge.cpp.o: in function `_ZN9RainGauge11currentWeekEv':
/home/richard/Arduino/BresserWeatherSensorMQTTCustom/src/RainGauge.cpp:354: multiple definition of `_ZN9RainGauge11currentWeekEv'; /tmp/arduino/sketches/84A2F82A7D12FB5A4436CFA77D535BD8/sketch/RainGauge.cpp.o:/home/richard/Arduino/BresserWeatherSensorMQTTCustom/RainGauge.cpp:354: first defined here
/home/richard/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /tmp/arduino/sketches/84A2F82A7D12FB5A4436CFA77D535BD8/sketch/src/RainGauge.cpp.o: in function `_ZN9RainGauge12currentMonthEv':
[... cut ...]
 - Serial console log file (see [Debug Output Configuration in Arduino IDE](https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/DEBUG_OUTPUT.md))

**Embedded Device (please complete the following information):**
 - Arduino IDE Version: 2.2.1
 -
 - ESP32/ESP8266 Board Support Package Version: 3.1.2
 - Library Version :??
 - Your Board: Wemos D1 mini
 - Selected Board in the Arduino IDE: [Wemos D1 mini
 - Your Radio Transceiver Module (if external): CC1101
 - **Pinning/Wiring**

**Additional context**
Add any other context about the problem here.

matthias-bs, 2023-12-29:
The linker finds many symbols twice - once from a previous build and once from the current build.
Please close the Arduino IDE, delete the folder `/tmp/arduino/sketches/84A2F82A7D12FB5A4436CFA77D535BD8/` (as shown in your log file - or everything in `/tmp/arduino/sketches/`) and try again.

kroon040, 2023-12-29:
It's now compiling. Wifi is connecting and than connecting to mqtt it reboots.

I change the port form 8883 to 1883 and use_wifi instead of Secure_wifi

```
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3424, room 16 
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8 
tail 0
chksum 0x2b
csum 0x2b
v000521b0
~ld
����g�r��o|�$�l`c��|{�d�o��o�d`��{�l�$��fpm close 1 
mode : sta(68:c6:3a:f4:4e:fe)
add if0
...scandone
state: 0 -> 2 (b0)
.state: 2 -> 3 (0)
state: 3 -> 2 (7a0)
.state: 2 -> 0 (4)
reconnect
..scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 17
cnt 

connected with kroonen, channel 8
dhcp client start...
ip:192.168.170.82,mask:255.255.255.0,gw:192.168.170.254
.Checking wifi...
MQTT connecting... 
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Soft WDT reset
```

matthias-bs, 2023-12-30:
Could you please provide the exception decoder output?
Which version of the library are you using?
Which "Core Debug Level" did you set in the Arduino IDE?

The only glitch I am currently aware of is in `BresserWeatherSensorMQTTCustom/src/WeatherSensor.cpp` (l. 195):

Please change

```
log_e("%s startReceive() failed, code %d", state);
```

to

```
log_e("%s startReceive() failed, code %d", RECEIVER_CHIP, state);
```

I doubt this is the cause.

kroon040, 2023-12-30:
I'm a bit further.

I now stuck on connect to the broker

`connected with kroonen, channel 8
dhcp client start...
..ip:192.168.170.39,mask:255.255.255.0,gw:192.168.170.254
.Checking wifi...
MQTT connecting... pm open,type:2 0..................`

I have changed thte port to 1883 and also
```

// #define USE_SECUREWIFI        // use secure WIFI
#define USE_WIFI              // use non-secure WIFI
```

matthias-bs, 2023-12-30:
If you are running your own broker, you can check the log files to see why the connection fails.

kroon040, 2023-12-30:
Looking in the broker, and also MQTT explorer, I see the everything is working and getting data. I think the logging on the serial must be increased?

kroon040, 2023-12-30:
I notriced taht the reception is not so weel, I ordered and TTG Lora esp32 , hopefully that will do better

closed 2023-12-30 by kroon040