matthias-bs/BresserWeatherSensorReceiver #121: Trouble with Heltec ESP32 WiFi LoRa 32(V3) / Wireless shell(V3) / Wireless stick lite (V3) SX1262

yalgoo, 2024-01-03:
I have trouble to receive the weather data from Bresser 3-1 7002531. With the basic sample file the serial out put of Arduino ID is:

21:58:17.859 -> [1278568][D][WeatherSensor.cpp:304] getMessage(): [SX1262] R [D4] RSSI: -95.0
21:58:17.859 -> [1278568][D][WeatherSensor.cpp:475] findSlot(): sensor[0]: v=0 id=0xFD300512 t=1 c=0
21:58:17.859 -> [1278571][D][WeatherSensor.cpp:979] decodeBresser6In1Payload(): Flags: Temp=1  Hum=1  Wind=1  Rain=0  UV=1
21:58:17.859 -> Id: [FD300512] Typ: [1] Ch: [0] St: [0] Bat: [OK ] RSSI: [ -95.0dBm] Temp: [  7.6C] Hum: [ 91%] Wmax: [ 3.1m/s] Wavg: [ 3.0m/s] Wdir: [  0.0deg] Rain: [-----.-mm] UVidx: [0.0]

When I start the mqtt the actual data values are cut-off and are missing:

22:48:05.985 -> [232979][D][WeatherSensor.cpp:304] getMessage(): [SX1262] R [D4] RSSI: -95.0
22:48:05.985 -> [232980][D][WeatherSensor.cpp:475] findSlot(): sensor[0]: v=0 id=0xFD300512 t=1 c=0
22:48:05.985 -> [232982][D][WeatherSensor.cpp:979] decodeBresser6In1Payload(): Flags: Temp=1  Hum=1  Wind=1  Rain=0  UV=1

When I enable "#define _DEBUG_MQTT_" all values are transmitted are received by MQTT Explorer.  I have no idea to get forward.

matthias-bs, 2024-01-04:
Hi,

I think I recently found and fixed this bug locally, but I did not publish it yet.
Could you please check if https://github.com/matthias-bs/BresserWeatherSensorReceiver/commit/2490a1d5f6647afe385d5f4f8edb78f00e194593 fixes the issue?

(You can simply copy the sketch from the main branch.)

Regards,
Matthias

yalgoo, 2024-01-06:
Thank you for your response. I added your new mqtt example file in my project and also updated the library. I still do not receive the sensor data. With activated  _DEBUG_MQTT_ values are transmitted and received of MQTT host.

Following the Debug print out:
```js
15:40:37.822 -> ESP-ROM:esp32s3-20210327
15:40:37.856 -> Build:Mar 27 2021
15:40:37.856 -> rst:0x8 (TG1WDT_SYS_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
15:40:37.856 -> Saved PC:0x403743c0
15:40:37.856 -> SPIWP:0xee
15:40:37.856 -> mode:DIO, clock div:1
15:40:37.856 -> load:0x3fce3808,len:0x43c
15:40:37.856 -> load:0x403c9700,len:0xbec
15:40:37.856 -> load:0x403cc700,len:0x2a3c
15:40:37.856 -> SHA-256 comparison failed:
15:40:37.856 -> Calculated: dcde8d8a4817d9bf5d5d69a7247667264e4e10ac7493514868b61f5aa6146539
15:40:37.856 -> Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
15:40:37.889 -> Attempting to boot anyway...
15:40:37.889 -> entry 0x403c98d8
15:40:38.021 -> [   205][I][BresserWeatherSensorMQTT_04-01-2024.ino:696] setup(): 
15:40:38.053 -> 
15:40:38.053 -> BresserWeatherSensorMQTT 20231030
15:40:38.053 -> 
15:40:38.053 -> [   206][I][BresserWeatherSensorMQTT_04-01-2024.ino:368] printDateTime(): 2024-01-06 15:40:33
15:40:38.053 -> [   212][I][BresserWeatherSensorMQTT_04-01-2024.ino:399] mqtt_setup(): Attempting to connect to SSID: Macquarie-IoT
15:40:38.053 -> [   232][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 0 - WIFI_READY
15:40:38.085 -> [   264][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 2 - STA_START
15:40:38.085 -> .[   329][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
15:40:38.182 -> [   346][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
15:40:38.182 -> [   346][D][WiFiGeneric.cpp:991] _eventCallback(): STA IP: 192.168.40.254, MASK: 255.255.255.0, GW: 192.168.40.1
15:40:39.106 -> [  1271][I][BresserWeatherSensorMQTT_04-01-2024.ino:404] mqtt_setup(): connected!
15:40:39.106 -> [  1272][I][BresserWeatherSensorMQTT_04-01-2024.ino:408] mqtt_setup(): Setting time using SNTP
15:40:39.106 -> [  1276][I][BresserWeatherSensorMQTT_04-01-2024.ino:423] mqtt_setup(): 
15:40:39.138 -> done!
15:40:39.138 -> [  1283][I][BresserWeatherSensorMQTT_04-01-2024.ino:428] mqtt_setup(): Current time (GMT): Sat Jan  6 14:40:35 2024
15:40:39.138 -> 
15:40:39.138 -> Checking wifi...
15:40:39.138 -> MQTT connecting... [  1346][I][BresserWeatherSensorMQTT_04-01-2024.ino:482] mqtt_connect(): 
15:40:39.170 -> connected!
15:40:39.170 -> [  1352][I][BresserWeatherSensorMQTT_04-01-2024.ino:484] mqtt_connect(): ESPWeather-E489FC/status: online
15:40:39.203 -> 
15:40:39.203 -> [  1353][D][WeatherSensor.cpp:126] begin(): [SX1276] Initializing ... 
15:40:40.094 -> ESP-ROM:esp32s3-20210327
```

Thank you advance.  Regards, Christoph

matthias-bs, 2024-01-06:
Hi,

Could you please provide the compiler output?
What happens after the message `[  1353][D][WeatherSensor.cpp:126] begin(): [SX1276] Initializing ... `?

Regards,
Matthias

yalgoo, 2024-01-06:
Hi Matthias,
Thank you for your mail. 

Compiler output:

￼

What happens after… :  It starts again with "-> ESP-ROM:esp32s3-20210327“ and runs in endless loop.

Regards,
Christoph

> 
> 
> Hi,
> 
> Could you please provide the compiler output?
> What happens after the message [ 1353][D][WeatherSensor.cpp:126] begin(): [SX1276] Initializing ... ?
> 
> Regards,
> Matthias

In file included from /Users/gromit/Documents/Arduino/my/1/BresserWeatherSensorMQTT_04-01-2024/BresserWeatherSensorMQTT_04-01-2024.ino:192:
/Users/gromit/Documents/Arduino/libraries/BresserWeatherSensorReceiver/src/WeatherSensorCfg.h:518:48: note: #pragma message: Receiver chip: [SX1276]
 #pragma message("Receiver chip: " RECEIVER_CHIP)
                                                ^
/Users/gromit/Documents/Arduino/libraries/BresserWeatherSensorReceiver/src/WeatherSensorCfg.h:519:170: note: #pragma message: Pin config: RST->32, CS->27, GD0/G0/IRQ->21, GDO2/G1/GPIO->33
 #pragma message("Pin config: RST->" STR(PIN_RECEIVER_RST) ", CS->" STR(PIN_RECEIVER_CS) ", GD0/G0/IRQ->" STR(PIN_RECEIVER_IRQ) ", GDO2/G1/GPIO->" STR(PIN_RECEIVER_GPIO) )
                                                                                                                                                                          ^
Sketch uses 796681 bytes (60%) of program storage space. Maximum is 1310720 bytes.
Global variables use 40424 bytes (12%) of dynamic memory, leaving 287256 bytes for local variables. Maximum is 327680 bytes.
esptool.py v3.3
Serial port /dev/cu.SLAB_USBtoUART
Connecting....
Chip is ESP32-S3
Features: WiFi, BLE
Crystal is 40MHz
MAC: 48:27:e2:e4:89:fc
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00003fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x000d2fff...
Flash params set to 0x023f
Compressed 15008 bytes to 10328...
Writing at 0x00000000... (100 %)
Wrote 15008 bytes (10328 compressed) at 0x00000000 in 1.0 seconds (effective 117.3 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 129...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (129 compressed) at 0x00008000 in 0.0 seconds (effective 679.7 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 1259.2 kbit/s)...
Hash of data verified.
Compressed 797232 bytes to 502130...
Writing at 0x00010000... (3 %)
Writing at 0x0001c5c4... (6 %)
Writing at 0x00026c6a... (9 %)
Writing at 0x00038dda... (12 %)
Writing at 0x0003e4ca... (16 %)
Writing at 0x00044730... (19 %)
Writing at 0x0004a20d... (22 %)
Writing at 0x0004fad3... (25 %)
Writing at 0x000558f2... (29 %)
Writing at 0x0005b308... (32 %)
Writing at 0x00060911... (35 %)
Writing at 0x00065a3f... (38 %)
Writing at 0x0006a9df... (41 %)
Writing at 0x0006f87c... (45 %)
Writing at 0x00074751... (48 %)
Writing at 0x000796db... (51 %)
Writing at 0x0007e810... (54 %)
Writing at 0x00083a2d... (58 %)
Writing at 0x000895df... (61 %)
Writing at 0x0008e56a... (64 %)
Writing at 0x00093559... (67 %)
Writing at 0x000985b8... (70 %)
Writing at 0x0009da02... (74 %)
Writing at 0x000a2e46... (77 %)
Writing at 0x000a8889... (80 %)
Writing at 0x000ae02c... (83 %)
Writing at 0x000b37a1... (87 %)
Writing at 0x000bab0e... (90 %)
Writing at 0x000c2902... (93 %)
Writing at 0x000c9389... (96 %)
Writing at 0x000ceb5c... (100 %)
Wrote 797232 bytes (502130 compressed) at 0x00010000 in 44.6 seconds (effective 143.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

matthias-bs, 2024-01-06:
Hi Christoph,

something is wrong with your configuration: The devices listed in the topic of this issue all have a SX1262 transceiver - which was already used in your first test (see your first comment). Now you are compiling for a SX1276 transceiver (see above). Which board do you actually use and which do you set in the Arduino IDE?

If you are using the Heltec Wireless Stick, you simply have to change `WeatherSensorCfg.h` to
```
#elif defined(ARDUINO_heltec_wireless_stick)
    #pragma message("ARDUINO_heltec_wireless_stick defined; using on-board transceiver")
    #pragma message("Radio transceiver chip has to be configured manually: V2 -> USE_SX1276 / V3 -> USE_SX1262")
    //#define USE_SX1276 // Heltec Wireless Stick V2
    #define USE_SX1262 // Heltec Wireless Stick V3
```

With the wrong transceiver selected, the code fails to initialize and gets in an endless loop, which leads to a watchdog reset.
 
Cheers,
Matthias

yalgoo, 2024-01-08:
Hi Matthias,
I accidentally overwrote the weather config. Now it is correct - but the result keeps the same. With the basis file values are transmitted and with the mqtt not. 
The board library is: https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/releases/download/0.0.9/package_heltec_esp32_index.json
and the board runs as: Heltec ESP32 WiFi LoRa 32(V3) / Wireless shell(V3) / Wireless stick lite (V3) 
Photos of the board and compiling files are attached.
[image]
[image]
[output compiling BresserWeatherSensorBasic.txt](https://github.com/matthias-bs/BresserWeatherSensorReceiver/files/13864793/output.compiling.BresserWeatherSensorBasic.txt)
[oupur compiling BresserWeatherSensorMQTT.txt](https://github.com/matthias-bs/BresserWeatherSensorReceiver/files/13864795/oupur.compiling.BresserWeatherSensorMQTT.txt)

Thank you - Christoph

matthias-bs, 2024-01-08:
Hi Christoph,

In short: In `BresserWeatherSensorMQTT.ino`, replace

```
#define RX_STRATEGY DATA_COMPLETE
```

by

```
#define RX_STRATEGY 0
```

Reason: The 6-in-1 protocol normally uses two alternating messages for the complete weather sensor data - except for the 3-in-1 sensor, which will not send rain data. Unfortunately there is no way to distinguish between the 3-in-1 and other sensors; both have the same type code in the message.

See https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/2490a1d5f6647afe385d5f4f8edb78f00e194593/src/WeatherSensor.h#L114

and following lines for details.

Cheers,
Matthias

matthias-bs, 2024-01-08:
BTW: I only checked the Espressif board package, not the one by Heltec. But there is no need to change it, since it seems to work in your case as well.

yalgoo, 2024-01-08:
Hi Matthias,
Thank you for your immediate  response. Just a quick update. Now I receive these topics:
data ={"id":4247782674,"ch":0,"battery_ok":1,"temp_c":-6.8,"humidity":72,"wind_gust":4.6,"wind_avg":4.2,"wind_dir":0.0,"uv":0.0}
rassi = -109.5
extra =
{"wind_dir_txt":"N","wind_gust_bft":2,"wind_avg_bft":2,"dewpoint_c":-11.1,"perceived_temp_c":-10.7}
status = online or offline

Christoph

tracteur12, 2024-01-09:
Hello,

Thank for you good job.
I have a question.
I looklike to use alone transceiver for multifrequence (SOMFY and Bresser). I hope to use RFplayer. Your system, is it work with this dongle?

Thank for your feedback.

Romain.

https://myrfplayer.com/
https://github.com/crazymikefra/domoticz-rfplayer
[rfplayer-guide-utilisation.pdf](https://github.com/matthias-bs/BresserWeatherSensorReceiver/files/13871896/rfplayer-guide-utilisation.pdf)

matthias-bs, 2024-01-09:
> Hi Matthias, Thank you for your immediate response. Just a quick update. Now I receive these topics: data [...]

Hi Christoph,

fine, the issue seems to be solved then, right? The UV value is another thing which cannot be distinguished automatically from the protocol. Your 3-in-1 sensor cannot provide it, so you can simply ignore it.

Cheers,
Matthias

yalgoo, 2024-01-11:
Hi Matthias, 
Thank you one more for patience  and support. I just reported all transmitted topics. I will collect only the relevant and calculate average, distribution, change units and so on . Cheers, Christoph

matthias-bs, 2024-01-12:
Hi Christoph,
You are welcome. Along the way, we got a small improvement of the documentation and another board confirmed to be working! 
Cheers,
Matthias

closed 2024-01-12 by matthias-bs

mothra78, 2024-01-13:
If you want to use the onboard LED on Wirless Stick V3 you need to define : 
#define LED_GPIO LED