matthias-bs/BresserWeatherSensorReceiver #122: Does it work with Bresser 3in1 as well?

iz3man, 2024-01-04:
Sorry, found no other place to ask. I only saw version 5in1 6in1 7in1 mentioned, but there is a Wind/Temp/Humidity sensor from Bresser as well. Is this supported? Wanted to be sure before I start building.

Thanks!

https://www.amazon.de/dp/B07PDQCY7Q?psc=1&ref=ppx_yo2ov_dt_b_product_details

[image]

matthias-bs, 2024-01-04:
Hi,

No problem. Yes, it is supported. You can find a list of supported devices in the table at the top of the [README]( https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/README.md).

Best regards
Matthias

iz3man, 2024-01-04:
Danke dir! Thank you!

closed 2024-01-04 by matthias-bs