matthias-bs/BresserWeatherSensorReceiver #130: Config isn't saved

Calimerorulez, 2024-01-28:
Hi,

Using `BresserWeatherSensorMQTTWifiMgr`.

I'm having a hard time to save my config a couple of versions now. When configuring Wifi and MQTT via the integrated AP, after pressing 'save', the info is retained after reboot, but after a power cycle the info is gone, and I have to connect to the AP again to configure the ESP again.

Seeing the following error on mounting and saving:

```
10:27:08.233 -> [ 74179][D][WeatherSensor.cpp:128] begin(): [SX1276] Initializing ... 
10:27:08.264 -> [ 74199][D][WeatherSensor.cpp:144] begin(): success!
10:27:08.264 -> [ 74200][D][WeatherSensor.cpp:189] begin(): [SX1276] Setup complete - awaiting incoming messages...
10:27:08.264 -> Stop doubleResetDetecting
10:27:08.264 -> [ 74209][E][vfs_api.cpp:24] open(): File system is not mounted
10:27:08.264 -> Saving config file...
10:27:08.297 -> Saving config file failed
10:27:08.297 -> [ 74220][I][BresserWeatherSensorMQTTWifiMgr.ino:998] loop(): ESPWeather-9EB8C4/status: online
10:27:08.297 -> 
10:27:08.297 -> [ 74223][I][BresserWeatherSensorMQTTWifiMgr.ino:880] publishRadio(): ESPWeather-9EB8C4/radio: {"rssi":-56}
10:27:08.297 -> 

```

I've tried formatting flash before upload, and also skipping that, put it does not solve it. Using a TTGo Lora32 v2.1 1.6.1.

matthias-bs, 2024-01-28:
Hi,

I have replaced SPIFFS by LittleFS on the 'fix-wifimanager' branch, see https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/fix-wifimanager/examples/BresserWeatherSensorMQTTWifiMgr/BresserWeatherSensorMQTTWifiMgr.ino.

I don't have time for testing now, but you might want to give this a try.

Cheers,
Matthias

Calimerorulez, 2024-01-29:
Hi @matthias-bs 

Tried your fix, but sadly I do not see any difference:
```

16:55:25.401 -> [ 59558][I][BresserWeatherSensorMQTTWifiMgr.ino:563] wifimgr_setup(): saving config
16:55:25.401 -> [ 59565][E][vfs_api.cpp:24] open(): File system is not mounted
16:55:25.401 -> [ 59571][E][BresserWeatherSensorMQTTWifiMgr.ino:574] wifimgr_setup(): failed to open config file for writing
16:55:25.433 -> {"mqtt_server":"192.168.5.110","mqtt_port":"1883","mqtt_user":"espweather","mqtt_pass":"espweather"}[ 59593][I][BresserWeatherSensorMQTTWifiMgr.ino:584] wifimgr_setup(): local ip: 192.168.3.134
16:55:25.433 -> Checking wifi...
16:55:25.433 -> MQTT connecting... [ 59635][I][BresserWeatherSensorMQTTWifiMgr.ino:667] mqtt_connect(): 
16:55:25.467 -> connected!
16:55:25.497 -> [ 59643][I][BresserWeatherSensorMQTTWifiMgr.ino:669] mqtt_connect(): ESPWeather-9EB8C4/status: online
16:55:25.497 -> 
16:55:25.497 -> [ 59645][D][WeatherSensor.cpp:128] begin(): [SX1276] Initializing ... 
16:55:25.497 -> [ 59664][D][WeatherSensor.cpp:144] begin(): success!
16:55:25.497 -> [ 59666][D][WeatherSensor.cpp:189] begin(): [SX1276] Setup complete - awaiting incoming messages...
16:55:25.530 -> Stop doubleResetDetecting
16:55:25.530 -> [ 59674][E][vfs_api.cpp:24] open(): File system is not mounted
16:55:25.530 -> Saving config file...
16:55:25.530 -> Saving config file failed
```

matthias-bs, 2024-01-29:
Hi @Calimerorulez,

I think I fixed it. Initially or after erasing the flash, the LittleFS partition must be formatted. This is now done automatically if mounting the file system failed. My test just succeeded.

Could you please try again with the update from the [fix-wifimanager](https://github.com/matthias-bs/BresserWeatherSensorReceiver/tree/fix-wifimanager) branch?

Cheers,
Matthias

Calimerorulez, 2024-01-30:
Hi @matthias-bs 

I tested it, by power cycling, and it's working. Config is saved and the errors on the filesystem and saving the config are gone :)

Thanks!

closed 2024-01-30 by matthias-bs