matthias-bs/BresserWeatherSensorReceiver #133: New library with Bug

tonbor, 2024-01-30:
#pragma message("Receiver chip: " RECEIVER_CHIP)
                                                ^
e:\Arduino files\libraries\BresserWeatherSensorReceiver\src/WeatherSensorCfg.h:538:170: note: #pragma message: Pin config: RST->RST_LoRa, CS->SS, GD0/G0/IRQ->DIO0, GDO2/G1/GPIO->DIO1
 #pragma message("Pin config: RST->" STR(PIN_RECEIVER_RST) ", CS->" STR(PIN_RECEIVER_CS) ", GD0/G0/IRQ->" STR(PIN_RECEIVER_IRQ) ", GDO2/G1/GPIO->" STR(PIN_RECEIVER_GPIO) )
                                                                                                                                                                          ^
E:\Arduino files\BresserWeatherSensorMQTT_Display_nosleep_copy_20240124113954\BresserWeatherSensorMQTT_Display_nosleep_copy_20240124113954.ino: In function 'void publishWeatherdata(bool)':
E:\Arduino files\BresserWeatherSensorMQTT_Display_nosleep_copy_20240124113954\BresserWeatherSensorMQTT_Display_nosleep_copy_20240124113954.ino:600:36: error: invalid conversion from 'int' to 'bool*' [-fpermissive]
             if (lightning.pastHour(events)) {
                                    ^~~~~~
In file included from E:\Arduino files\BresserWeatherSensorMQTT_Display_nosleep_copy_20240124113954\BresserWeatherSensorMQTT_Display_nosleep_copy_20240124113954.ino:198:
e:\Arduino files\libraries\BresserWeatherSensorReceiver\src/Lightning.h:222:24: note:   initializing argument 1 of 'int Lightning::pastHour(bool*, int*)'
     int pastHour(bool *valid = nullptr, int *quality = nullptr);
                  ~~~~~~^~~~~~~~~~~~~~~

exit status 1

Compilation error: invalid conversion from 'int' to 'bool*' [-fpermissive]

matthias-bs, 2024-01-30:
That's what I marked as **breaking change** in the [v0.22.0 Release Notes](https://github.com/matthias-bs/BresserWeatherSensorReceiver/releases/tag/v0.22.0)! ;-)
If you update the sketch together with the library, everything will work again...

tonbor, 2024-01-31:
So your new library is not backwards compatible. So the homemade sketches
must all be adapted. Can you tell us what part?

matthias-bs, 2024-01-31:
See https://github.com/matthias-bs/BresserWeatherSensorReceiver/commit/cd7c14fc176a4cff224c1ad4f474519caccb2f97 for changes in the library and https://github.com/matthias-bs/BresserWeatherSensorReceiver/commit/beac20e61911112c05d15ec730c3d711322377ab for changes in the sketch.

tonbor, 2024-02-01:
> See cd7c14f
> for changes in the library and beac20e
> for changes in the sketch.

Thanks

closed 2024-02-01 by matthias-bs

tonbor, 2024-03-17:
the domoticz exaple is not correct (any more) because of newvinpkemantation of rain calculation
String(weatherSensor.sensor[i].w.rain_mm * 100, 1); should be rain mm per hour
if (weatherSensor.sensor[i].w.rain_ok)
    {
        domo2_payload = String("{\"idx\":") + String(DOMO_RAIN_IDX) + String(",\"nvalue\":0,\"svalue\":\"") + String(weatherSensor.sensor[i].w.rain_mm * 100, 1);
        domo2_payload += String(";") + String(weatherSensor.sensor[i].w.rain_mm, 1);
        domo2_payload += String("\"}");
        Serial.printf("%s: %s\n", MQTT_PUB_DOMO, domo2_payload.c_str());
        client.publish(MQTT_PUB_DOMO, domo2_payload.c_str(), false, 0);
    }