matthias-bs/BresserWeatherSensorReceiver #143: What receiver do you recommend?

elprocrastinador, 2024-03-10:
Only that, what receiver do you recommend?
CC1101, SX1276/RFM95W or SX1262?
Thanks.

matthias-bs, 2024-03-11:
With a SX1276/RFW95W you have a large variety of modules to chose from and probably the least chance of trouble. Just make sure to get an 868 MHz version.

elprocrastinador, 2024-03-11:
Any experience about reliability? range?

matthias-bs, 2024-03-11:
Some thoughts about range: https://github.com/matthias-bs/BresserWeatherSensorReceiver/wiki/06.-Radio-Transceiver-Range

No reliability issues encountered so far, but three deployed units might not be a relevant sample...

nosfaratu, 2024-03-20:
Thanks for the development!
But I need help...
I have a Lora Lilygo T3 V1.6.1 and no matter how much I follow your tutorial I can't flash this :/
I know I shouldn't put this here, but I don't know where else to turn, sorry

matthias-bs, 2024-03-21:
Please create a new issue using the template and describe your attempt as detailed as possible.

closed 2024-03-25 by matthias-bs