matthias-bs/BresserWeatherSensorReceiver #144: rain implementation -> renew examples

tonbor, 2024-03-18:
the domoticz example is not correct (any more) because of new implementation of rain calculation
String(weatherSensor.sensor[i].w.rain_mm * 100, 1); should be rain mm per hour
domo2_payload = String("{"idx":") + String(DOMO_RAIN_IDX) + String(","nvalue":0,"svalue":"") + String(rainGauge.pastHour() * 100, 1);

if (weatherSensor.sensor[i].w.rain_ok)
{
domo2_payload = String("{"idx":") + String(DOMO_RAIN_IDX) + String(","nvalue":0,"svalue":"") + String(weatherSensor.sensor[i].w.rain_mm * 100, 1);
domo2_payload += String(";") + String(weatherSensor.sensor[i].w.rain_mm, 1);
domo2_payload += String(""}");
Serial.printf("%s: %s\n", MQTT_PUB_DOMO, domo2_payload.c_str());
client.publish(MQTT_PUB_DOMO, domo2_payload.c_str(), false, 0);
}

matthias-bs, 2024-03-19:
Do you have an example of how the MQTT payload is supposed to look like - or even better - a link to the Domoticz specification?
I don't use this myself, I just implemented this upon a user's request.

tonbor, 2024-03-19:
https://www.domoticz.com/wiki/Domoticz_API/JSON_URL%27s#Rain

matthias-bs, 2024-03-25:
Can you please test it?
https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/fix-domoticz-rain/examples/BresserWeatherSensorDomoticz/BresserWeatherSensorDomoticz.ino

matthias-bs, 2024-04-09:
Added in https://github.com/matthias-bs/BresserWeatherSensorReceiver/pull/152

closed 2024-04-09 by matthias-bs