matthias-bs/BresserWeatherSensorReceiver #145: Can't flash in Lilygo T3 V1.6.1

matthias-bs, 2024-03-21:
Thanks for the development!
But I need help...
I have a Lora Lilygo T3 V1.6.1 and no matter how much I follow your tutorial I can't flash this :/
I know I shouldn't put this here, but I don't know where else to turn, sorry

_Ursprünglich gepostet von @nosfaratu in https://github.com/matthias-bs/BresserWeatherSensorReceiver/issues/143#issuecomment-2010911201_

nosfaratu, 2024-03-21:
Mathias, thanks for opening the topic.
I'm a bit of a code lammer and I tried to follow all your steps, but without success.

Board - lilygo model T3 V1.6.1

So I'm using Arduino IE as you can see in this link to have a chance of doing everything right, but not :/
https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/BUILD.md

**code for BresserWeatherSensorMQTT2.ino**
https://gist.github.com/nosfaratu/a1c46ffddb8e8281a7018ab253c5f4f0

**code for WeatherSensorcfg.h**
https://gist.github.com/nosfaratu/6b9cf40bd2a2667aee947fadecb6ac58

I compile the code and save it in another folder.
Then I'll check ok, when I upload it gives an error, but the upload finishes.
Set it to reset but the board does not reset.

**[image]**
https://gist.github.com/nosfaratu/3bd3a127fa0f1e063de87ca436db1833

Thanks in advance!

matthias-bs, 2024-03-22:
Hi,

in your `WeatherSensorCfg.h`, you have uncommented two defines in the board section:
```
// This define is set by selecting "Board: TTGO LoRa32-OLED" / "Board Revision: TTGO LoRa32 V2.1 (1.6.1)"
// in the Arduino IDE:
#define ARDUINO_TTGO_LoRa32_V21new

[...]

// DFRobot Firebeetle32
// https://github.com/espressif/arduino-esp32/tree/master/variants/firebeetle32/pins_arduino.h
//
// This define (not very specific...) is set by selecting "FireBeetle-ESP32" in the Arduino IDE:
#define ARDUINO_ESP32_DEV
```
Please leave those defines commented out as it was before. If you select this board and board variant in the IDE (as you did as I see in the screenshot), the correct settings are used automatically.

For the first test, I recommend using the basic example. I any case, please use the examples provided when trying the first steps. You are trying `BresserWeatherSensorMQTT2.ino` - how should I know which changes you made?

After flashing the board, please switch to the serial console (same serial port as for flashing).

Regards,
Matthias

nosfaratu, 2024-03-23:
@matthias-bs 
Success! Using just BresserWeatherSensorBasic.ino from the examples, without changing anything it flashed!
[image]

The log appears on the serial monitor!
[image]

But I can't pass anything to HA xD

matthias-bs, 2024-03-23:
Did you check the part related to secure/non-secure MQTT and the related port and WiFi settings?
You have to change
```
#define USE_SECUREWIFI        // use secure WIFI
// #define USE_WIFI              // use non-secure WIFI
```
to
```
//#define USE_SECUREWIFI        // use secure WIFI
#define USE_WIFI              // use non-secure WIFI
```
in order to try a non-secure connection on port 1883.

Check the MQTT broker's logfile to see why the connection fails.

matthias-bs, 2024-03-23:
A secure connection requires `USE_SECUREWIFI`, port 8883 and some specific credentials.

matthias-bs, 2024-03-23:
Regarding the display - have a look at [find_transmitter.ino](https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/feature-m5stack-core2/extras/hw_test/find_transmitter/find_transmitter.ino) how to use it.

nosfaratu, 2024-03-23:
mqtt is already resolved! I had to set sleep to false, because it almost always appeared offline
I can now listen to the topic on homeassistant
[image]
Now it remains to create the sensors.

Regarding the screen, do I just put that file and flash it along with the others?

matthias-bs, 2024-03-23:
No, you have to modify the MQTT sketch to use the display. You can use the other sketch I mentioned as an example for using the Adafruit SSD1306 library.

matthias-bs, 2024-04-14:
Seems to be solved - closing.

closed 2024-04-14 by matthias-bs