matthias-bs/BresserWeatherSensorReceiver #146: Help with setting up Waldbeck Huygens weather station

jdukat, 2024-03-21:
Hi, I'm trying to make sense from this:
https://github.com/matthias-bs/BresserWeatherSensorReceiver/issues/41
and from this:
https://github.com/matthias-bs/BresserWeatherSensorReceiver/issues/44

The latter I really don't understand :)
I'm using an USB SDR stick like this one: https://www.aliexpress.com/item/1005001815260665.html

The weather station seem to be a variant of Sainlogic WS3500 or other Ecowitt device, but I'm completely new to rtl_433 and after hours of staring at various parameters, I can't figure out what to do next.

Here's what I've found:

`rtl_433 -f 868.3M -vvv`
returns:

trying device  0:  Realtek, RTL2838UHIDIR, SN: 00000001
Detached kernel driver
Found Fitipower FC0012 tuner
Using device 0: Generic RTL2832U OEM
Exact sample rate is: 1000000.026491 Hz
Sample rate set to 1000000 S/s.
Bit detection level set to 0.0 (Auto).
Tuner gain set to Auto.
Reading samples in async mode...
Tuned to 868.300MHz.
Allocating 15 zero-copy buffers
Fineoffset_WH24: Raw: 12 04 b2 c1 37 1c 80 00 00 18 00 00 00 07 40 17 a3  @ bit_offset [59]
M-Bus: CRC error: Calculated 0xFFFF, Read: 0x0
ambientweather_whx_decode: WH31E/WH40 detected, buffer is 198 bits length
ambientweather_whx_decode: unknown message type 12 (expected 0x30/0x40/0x68)
bresser_5in1_decode: bit_per_row 52 out of range
directv: incorrect number of bits in bitbuffer: 1 (expected between 44 and 99).
WARNING: Undeclared field "msg" in [130] "IKEA Sparsnas Energy Meter Monitor"
WARNING: Undeclared field "num_rows" in [130] "IKEA Sparsnas Energy Meter Monitor"
WARNING: Undeclared field "codes" in [130] "IKEA Sparsnas Energy Meter Monitor"
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-03-21 23:32:54                    msg       : ikea_sparsnas_decode:                   num_rows  : 1
codes     : {402}cccccccd99b3333333333033cfe66030c0061bc33c01879f81f860000000000003fff000000000000001ffcc0000cfcc01890
ikea_sparsnas_decode: Too short or too long packet received. Expected 160, received 402
Fineoffset_WH51: Msg family unknown: 12
Fine Offset WH1080 data {80} ff 12 04 b2 c1 37 1c 80 00 00
insteon_callback: new buffer 1 rows
lacrosse_breezepro_decode: Wrong packet length: 99
lacrosse_wr1_decode: Packet too short: 101 bits
lacrosse_th_decode: Packet too short: 101 bits
bresser_6in1_decode: bit_per_row 52 out of range
bresser_7in1_decode: to few bits (52)
ecodhome_decode: to few bits (12)
lacrosse_r1_decode: Packet too short: 101 bits
Pulse data: 245 pulses
[  0] Pulse:    0, Gap:    2, Period:    2
[  1] Pulse:    3, Gap:    2, Period:    5
[  2] Pulse:   15, Gap:    1, Period:   16
[  3] Pulse:    4, Gap:    1, Period:    5
[  4] Pulse:   11, Gap:    1, Period:   12
[  5] Pulse:   15, Gap:    1, Period:   16
[  6] Pulse:   10, Gap:   59, Period:   69
[  7] Pulse:   59, Gap:   50, Period:  109
[  8] Pulse:    3, Gap:    2, Period:    5
[  9] Pulse:   61, Gap:    5, Period:   66
[ 10] Pulse:    2, Gap:   29, Period:   31
[ 11] Pulse:    3, Gap:   18, Period:   21
[ 12] Pulse:   58, Gap:   30, Period:   88
[ 13] Pulse:    1, Gap:    3, Period:    4
[ 14] Pulse:    1, Gap:   23, Period:   24
[ 15] Pulse:   29, Gap:    1, Period:   30
[ 16] Pulse:   28, Gap:   56, Period:   84
[ 17] Pulse:   62, Gap:   33, Period:   95
[ 18] Pulse:    1, Gap:   22, Period:   23
[ 19] Pulse:   57, Gap:   25, Period:   82
[... 293 lines cut ...]
 [ 1] count:   16,  width:    5 us [5;7]        (   5 S)
 [ 2] count:   48,  width:    3 us [3;3]        (   3 S)
 [ 3] count:  118,  width:    2 us [2;2]        (   2 S)
 [ 4] count:   91,  width:    1 us [1;1]        (   1 S)
 [ 5] count:   61,  width:   61 us [49;75]      (  61 S)
 [ 6] count:   32,  width:   23 us [19;29]      (  23 S)
 [ 7] count:   64,  width:   33 us [28;41]      (  33 S)
 [ 8] count:   19,  width:  115 us [100;141]    ( 115 S)
 [ 9] count:    6,  width:  182 us [173;211]    ( 182 S)
 [10] count:   26,  width:    4 us [4;4]        (   4 S)
 [11] count:    4,  width:   81 us [77;86]      (  81 S)
 [12] count:    6,  width:    8 us [8;9]        (   8 S)
 [13] count:   17,  width:   16 us [14;18]      (  16 S)
 [14] count:   10,  width:   11 us [10;13]      (  11 S)
Level estimates [high, low]:  11834,    489
RSSI: -1.4 dB SNR: 13.8 dB Noise: -15.3 dB
Frequency offsets [F1, F2]:     -43,   2232     (-0.7 kHz, +34.1 kHz)
Guessing modulation: Pulse Code Modulation (Not Return to Zero)
Attempting demodulation... short_width: 1, long_width: 1, reset_limit: 1024, sync_width: 0
Use a flex decoder with -X 'n=name,m=FSK_PCM,s=1,l=1,r=1024'
pulse_demod_pcm(): Analyzer Device
bitbuffer:: Number of rows: 1
[00] {2048} 3f 73 f7 77 33 bb bf 30 00 00 c0 00 00 00 21 ff ff ff ff ff ff ff f0 00 60 00 00 00 00 0f ff ff ef ff ff ff ff 01 80 00 00 00 00 00 ff ff ff ff ff ff ff f0 e4 00 00 00 00 00 0f ff ff ff ff ff ff ff c0 00 00 00 03 00 00 7f ff ff ff ff ff ff c0 00 00 03 00 00 00 03 ff ff ff ff ff ff fe 00 00 00 c0 00 00 00 7f ff ff ff ff ff ff f0 00 00 00 00 00 00 0f ff ff ff ff ff ff ff 00 06 00 00 00 00 00 7f ff ff ff ff ff ff fc 0c 00 00 00 00 00 07 ff ff ff ff fb ff ff 80 00 00 00 10 00 00 3f ff ff fb ff ff ff e8 00 00 00 01 c0 00 01 ff ff ff ff bf ff fe 40 00 00 60 60 00 00 1f ff ff ff ff ff ff e0 00 00 00 00 00 00 07 ff ff ff ff ff ff ff 80 02 00 00 00 00 01 bf ff ff ff ff ff ff fe 00 00 00 00 00 00 07 ff ff ff ff ff ff ff e0 00 00 00 00 00 00 1f ff ff ff ff ff ff fe 00

From here, I'm lost on what to try next... Any hints?

jdukat, 2024-03-21:
One more maybe useful output that I grabbed based on #1750

`rtl_433 -f 868.3M -R 0 -X 'n=name,m=FSK_PCM,s=1,l=1,r=1024'`

Returns such packets every 16 seconds, so it is likely the weather station output, but does a proper decoder exist for it?

time      : 2024-03-22 00:09:10
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 2048         data      : 1ffffffffffffdfe00f00000000001ffffffffffffffc000000000000007ffffffffffffff80000000000000bfffffffffffffd80000000198000ffffffffffffffc000010000000001fffffffffffff8000000000600007fffffffffffffe000400000000005fffffffffffffe000000000006007ffffffffffffff000018000000001ffffffffffffff80000000000c80fffffffffffffff800063000000001ffffffffffffffc0300000000000fffffffffffffff000000003000003ffffffffffffffc00000000006007ffffffffffffff000000300000001ffffffffffffffe0c000000000007ffffffffffffff800000000000001fffffffffffffff00
codes     : {2048}1ffffffffffffdfe00f00000000001ffffffffffffffc000000000000007ffffffffffffff80000000000000bfffffffffffffd80000000198000ffffffffffffffc000010000000001fffffffffffff8000000000600007fffffffffffffe000400000000005fffffffffffffe000000000006007ffffffffffffff000018000000001ffffffffffffff80000000000c80fffffffffffffff800063000000001ffffffffffffffc0300000000000fffffffffffffff000000003000003ffffffffffffffc00000000006007ffffffffffffff000000300000001ffffffffffffffe0c000000000007ffffffffffffff800000000000001fffffffffffffff00
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-03-22 00:09:26
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 2048         data      : 3ff777ee66eeee7d0000000001c0000fffffffffffffff00008000000001ffffffffffffff8e0000000000001ffffffffffffffc000000400000007ffffffffffffff00000000000001ffffffffffffffc000000000000007ffffffffffffff00400000000000ffffffffffffffe0000000c0000007ffffffffffffff000000000000007ffffffffffffff00000000000000fffffffffffffff000000000000007ffffffffffffffc00000000000067fffffffffffffe000000000000001ffffffffffffffc0000000000018fffffffffffffff000000000000001ffffffffffffffc00000000000007fffffffffffffe000000000800001ffffffffffffffc0
codes     : {2048}3ff777ee66eeee7d0000000001c0000fffffffffffffff00008000000001ffffffffffffff8e0000000000001ffffffffffffffc000000400000007ffffffffffffff00000000000001ffffffffffffffc000000000000007ffffffffffffff00400000000000ffffffffffffffe0000000c0000007ffffffffffffff000000000000007ffffffffffffff00000000000000fffffffffffffff000000000000007ffffffffffffffc00000000000067fffffffffffffe000000000000001ffffffffffffffc0000000000018fffffffffffffff000000000000001ffffffffffffffc00000000000007fffffffffffffe000000000800001ffffffffffffffc0

jdukat, 2024-03-21:
One more datapoint - I read the ecowitt data uploaded over http, it reports:
freq=868M
model=WS2900_V2.02.01

And the ecowitt app shows the device as EasyWeatherPro v5.1.5

matthias-bs, 2024-03-22:
Hi,

in your first log, you have
```
ambientweather_whx_decode: WH31E/WH40 detected, buffer is 198 bits length
ambientweather_whx_decode: unknown message type 12 (expected 0x30/0x40/0x68)
```
That means the preample was already matching.

You could try to modify `https://github.com/merbanan/rtl_433/blob/master/src/devices/ambientweather_wh31e.c` (e.g. by adding your message type) and see if can decode any data fields.

This is issue should better be placed in the rtl_433 project, though.

If you get it working in rtl_433, it might be possible to port the decoder to BresserWeatherSensorReceiver. This depends on the required radio parameters.

matthias-bs, 2024-03-22:
How about this:
```
Fineoffset_WH24: Raw: 12 04 b2 c1 37 1c 80 00 00 18 00 00 00 07 40 17 a3 @ bit_offset [59]
```

See https://github.com/merbanan/rtl_433/blob/master/src/devices/fineoffset.c

jdukat, 2024-03-22:
Ah, honestly I intended to submit this for rtl_433. Not sure when I changed the project, probably while searching for other sources where I could find anything about this device :)

Looked into ambientweather_wh31e.c, but this seems to be an accidental match of the preamble, or the protocol has diverged very significantly.
When looking for logs from this decoder only, the message type changes and the buffer size changes (and is longer than max 18 bytes decoded for the longest recognized message).

`rtl_433 -f 868.3M -vv -R 113` outputs:
ambientweather_whx_decode: WH31E/WH40 detected, buffer is 207 bits length
ambientweather_whx_decode: unknown message type 24 (expected 0x30/0x40/0x68)
ambientweather_whx_decode: WH31E/WH40 detected, buffer is 216 bits length
ambientweather_whx_decode: unknown message type 24 (expected 0x30/0x40/0x68)
ambientweather_whx_decode: WH31E/WH40 detected, buffer is 211 bits length
ambientweather_whx_decode: unknown message type 24 (expected 0x30/0x40/0x68)
ambientweather_whx_decode: WH31E/WH40 detected, buffer is 216 bits length
ambientweather_whx_decode: unknown message type 12 (expected 0x30/0x40/0x68)
ambientweather_whx_decode: WH31E/WH40 detected, buffer is 212 bits length
ambientweather_whx_decode: unknown message type 12 (expected 0x30/0x40/0x68)
ambientweather_whx_decode: WH31E/WH40 detected, buffer is 212 bits length
ambientweather_whx_decode: unknown message type 12 (expected 0x30/0x40/0x68)
ambientweather_whx_decode: WH31E/WH40 detected, buffer is 209 bits length
ambientweather_whx_decode: unknown message type 24 (expected 0x30/0x40/0x68)
ambientweather_whx_decode: WH31E/WH40 detected, buffer is 210 bits length
ambientweather_whx_decode: unknown message type 12 (expected 0x30/0x40/0x68)
ambientweather_whx_decode: WH31E/WH40 detected, buffer is 216 bits length
ambientweather_whx_decode: unknown message type 24 (expected 0x30/0x40/0x68)
ambientweather_whx_decode: WH31E/WH40 detected, buffer is 210 bits length
ambientweather_whx_decode: unknown message type 12 (expected 0x30/0x40/0x68)
ambientweather_whx_decode: WH31E/WH40 detected, buffer is 210 bits length
ambientweather_whx_decode: unknown message type 12 (expected 0x30/0x40/0x68)
ambientweather_whx_decode: WH31E/WH40 detected, buffer is 213 bits length
ambientweather_whx_decode: unknown message type 24 (expected 0x30/0x40/0x68)
ambientweather_whx_decode: WH31E/WH40 detected, buffer is 215 bits length
ambientweather_whx_decode: unknown message type 11 (expected 0x30/0x40/0x68)
ambientweather_whx_decode: WH31E/WH40 detected, buffer is 211 bits length
ambientweather_whx_decode: unknown message type 22 (expected 0x30/0x40/0x68)
ambientweather_whx_decode: WH31E/WH40 detected, buffer is 215 bits length
ambientweather_whx_decode: unknown message type 24 (expected 0x30/0x40/0x68)
ambientweather_whx_decode: WH31E/WH40 detected, buffer is 210 bits length
ambientweather_whx_decode: unknown message type 12 (expected 0x30/0x40/0x68)
ambientweather_whx_decode: WH31E/WH40 detected, buffer is 209 bits length
ambientweather_whx_decode: unknown message type 24 (expected 0x30/0x40/0x68)

jdukat, 2024-03-22:
> How about this:
> 
> ```
> Fineoffset_WH24: Raw: 12 04 b2 c1 37 1c 80 00 00 18 00 00 00 07 40 17 a3 @ bit_offset [59]
> ```

This seems more probable, as wh24 - device [78] is also referenced in https://github.com/matthias-bs/BresserWeatherSensorReceiver/issues/41, and when I was buying this weather station, I got some clues that it is another brand of Sainlogic WS3500, which was expected to work seamlessly (as mentioned on Home assistant forums).

But scanning with -R 78 shows all kind of problems:

`rtl_433 -f 868.3M -vv -R 78`
Fineoffset_WH24: Raw: 12 04 b3 c0 8e 07 20 00 00 00 0c 00 00 00 01 90 46  @ bit_offset [56]
Fineoffset_WH24: short package. Header index: 225
bitbuffer:: Number of rows: 1
[00] {201} aa aa ab 55 55 4b 75 04 80 4e 78 27 01 d0 00 00 00 18 00 00 00 01 90 8d 67 00
Fineoffset_WH24: short package. Header index: 223
bitbuffer:: Number of rows: 1
[00] {199} 55 55 55 55 55 45 b9 09 02 59 a0 9c 0e 40 00 00 01 f0 00 00 00 72 0d 4b cc
Fineoffset_WH24: Raw: 24 09 67 82 70 32 00 00 00 03 00 00 00 00 72 04 56  @ bit_offset [59]
Fineoffset_WH24: Checksum error: fd 31

jdukat, 2024-03-22:
Looking into the comment in fineoffset.c:

```
The sensor sends a package each ~16 s with a width of ~11 ms. The bits are PCM modulated with Frequency Shift Keying.

Example:

         [00] {196} d5 55 55 55 55 16 ea 12 5f 85 71 03 27 04 01 00 25 00 00 80 00 00 47 83 9
      aligned {199} 1aa aa aa aa aa 2d d4 24 bf 0a e2 06 4e 08 02 00 4a 00 01 00 00 00 8f 07 2
    Payload:                              FF II DD VT TT HH WW GG RR RR UU UU LL LL LL CC BB
    Reading: id: 191, temp: 11.8 C, humidity: 78 %, wind_dir 266 deg, wind_speed: 1.12 m/s, gust_speed 2.24 m/s, rainfall: 22.2 mm

```

This seems to make sense, as my device also sends packets every ~16 sec and if you look at the next byte after the preamble, it's 12 or 24, and in the log above the 12 and 24 are the most frequent msg unrecognized by wh31e decoder.

jdukat, 2024-03-22:
Can it be that my radio is receiving a lot of noise? The packets looks like they should make sense:
```
Fineoffset_WH24: Raw: 24 09 67 82 70 32 00 00 00 03 00 00 00 00 72 04 56 @ bit_offset [59]
Fineoffset_WH24: Checksum error: fd 31

```
24 is expected, but sometimes it is 12, sometimes 22 - look like single bit errors - bits shifted by 1 position
Next byte 09 - on my station there is a sticker saying 'sensor id 09' so it makes perfect sense, but sometimes it reads 04 instead

matthias-bs, 2024-03-25:
Could be noise or overloading the receiver if it is too close to the transmitter.

jdukat, 2024-03-29:
Moving things around doesn't help, same problem. I give up. I hoped to connect this weather station directly to Home Assistant and not using the head unit at all (use rtl_433 and send packets to mqtt) but I switched to Ecowitt integration and it works out of the box over local wifi. Feel free to close this issue, I will not debug it further.

closed 2024-03-30 by matthias-bs