matthias-bs/BresserWeatherSensorReceiver #154: Implementing new weather stations

dudzio12, 2024-04-09:
Hi,

I have a Bresser clone from Otio, which has identical sensor as 7003100. Both of those sensor are 7 in 1, but I have no readings with 7in1 sensor enabled and have full readings from it when I enable 5in1 sensor:

```
[ 26133][I][BresserWeatherSensorMQTTCustom.ino:702] publishWeatherdata(): ESPWeather-5ACF04/f2/data: {"id":242,"ch":0,"battery_ok":1,"temp_c":14.4,"humidity":75,"wind_gust":0.0,"wind_avg":0.2,"wind_dir":157.5,"rain":156.8,"rain_h":0.0,"rain_d":0.0,"rain_w":0.0,"rain_m":0.0}

[ 26150][I][BresserWeatherSensorMQTTCustom.ino:714] publishWeatherdata(): ESPWeather-5ACF04/extra: {"wind_dir_txt":"SSE","wind_gust_bft":0,"wind_avg_bft":0,"dewpoint_c":10.0,"perceived_temp_c":14.4}
```

My question is, how should I start implementing data from my sensor to make it recognizable as 7 in 1 sensor  (including missing UV sensor and barometer)? Is there a way to get data from indoor sensors of Genuine Bresser device?
Can I use my (Heltec v3 lite) board to grab data from my sensor  by dumping some data somewhere in the code to my console, as debug message and then implement it as 7 in 1?

The project is awesome, big thanks for the hard work :)

matthias-bs, 2024-04-12:
Hi,

this is strange... Are you sure you are receiving your own sensor and not a neighbors?

It is quite unlikely that the 5-in-1 protocol is used to transmit data from a 7-in-1 sensor. The 5-in-1 protocol has no space left for additional data (AFAIK) - see [WeatherSensor.cpp](https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/src/WeatherSensor.cpp). So the UV and the light intensity values would have to be transmitted in an additional message (similar to the 6-in-1 protocol). Please note that the barometer is integrated in the base station and cannot be received as a radio signal.

To see what is going on "on air", you have to enable debug level "verbose" (see [DEBUG_OUTPUT.md](https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/DEBUG_OUTPUT.md)).

What do you mean by "Is there a way to get data from indoor sensors of Genuine Bresser device?" ? The base station is just a receiver. If you mean additional indoor sensors transmitting at 868 MHz - those should be supported.

Best regards,
Matthias

matthias-bs, 2024-04-13:
Please use the example BresserWeatherSensorBasic for testing. This allows to monitor all available sensor messages continuously.

matthias-bs, 2024-04-24:
Closed due to inactivity.

dudzio12, 2024-04-29:
I've had some time to run BresserWeatherSensorBasic example and there's no difference. I've also upgraded lib to newest version.

In serial monitor I see:
```
[212540][D][WeatherSensor.cpp:1267] decodeBresser7In1Payload(): Digest check failed - [46A7] vs [0186] (4721)
[212549][D][WeatherSensor.cpp:994] decodeBresser6In1Payload(): Digest check failed - [EC0D] != [DFBD]
```

Is it normal that those those functions run when my config is like below? I'm not an expert of Arduino IDE, but I've placed main .ino and WeatherSensorCfg.h files inside one catalog and edited both of those from IDE. I assume that .ino file uses that .h file placed next to it and not from library itself. 
```
// #define BRESSER_5_IN_1
// #define BRESSER_6_IN_1
#define BRESSER_7_IN_1
// #define BRESSER_LIGHTNING
// #define BRESSER_LEAKAGE
```

Here you can see my full log. I have no UV recordings which should be available. Everything beside that is working as expected :)

```
[200518][V][WeatherSensor.cpp:344] getMessage(): [SX1262] Data: D4 EC 0D 7F FF CF FF EF F9 FE 6C ED DF FF 13 F2 80 00 30 00 10 06 01 93 12 20 00 
[200531][D][WeatherSensor.cpp:346] getMessage(): [SX1262] R [D4] RSSI: -84.0
[200538][D][WeatherSensor.cpp:1267] decodeBresser7In1Payload(): Digest check failed - [46A7] vs [0186] (4721)
[200548][D][WeatherSensor.cpp:994] decodeBresser6In1Payload(): Digest check failed - [EC0D] != [DFBD]
[200557][V][WeatherSensor.cpp:473] findSlot(): find_slot(): ID=000000F2
[200563][D][WeatherSensor.cpp:511] findSlot(): sensor[0]: v=0 id=0x000000F2 t=0 c=0
[200571][V][WeatherSensor.cpp:536] findSlot(): find_slot(): Storing into slot #0
[212520][V][WeatherSensor.cpp:344] getMessage(): [SX1262] Data: D4 EC 0D 7F FF CF FF EF F9 FE 6C ED DF FF 13 F2 80 00 30 00 10 06 01 93 12 20 00 
[212533][D][WeatherSensor.cpp:346] getMessage(): [SX1262] R [D4] RSSI: -83.5
[212540][D][WeatherSensor.cpp:1267] decodeBresser7In1Payload(): Digest check failed - [46A7] vs [0186] (4721)
[212549][D][WeatherSensor.cpp:994] decodeBresser6In1Payload(): Digest check failed - [EC0D] != [DFBD]
[212558][V][WeatherSensor.cpp:473] findSlot(): find_slot(): ID=000000F2
[212565][D][WeatherSensor.cpp:511] findSlot(): sensor[0]: v=0 id=0x000000F2 t=0 c=0
[212572][V][WeatherSensor.cpp:536] findSlot(): find_slot(): Storing into slot #0
[212588][D][RainGauge.cpp:194] prefs_load(): lastUpdate        =1714427498
[212594][D][RainGauge.cpp:195] prefs_load(): startupPrev       =0
[212600][D][RainGauge.cpp:196] prefs_load(): rainPreStartup    =201.199997
[212607][D][RainGauge.cpp:197] prefs_load(): tsDayBegin        =1
[212613][D][RainGauge.cpp:198] prefs_load(): rainDayBegin      =201.199997
[212620][D][RainGauge.cpp:199] prefs_load(): tsWeekBegin       =1
[212626][D][RainGauge.cpp:200] prefs_load(): rainWeekBegin     =201.199997
[212632][D][RainGauge.cpp:201] prefs_load(): wdayPrev          =1
[212638][D][RainGauge.cpp:202] prefs_load(): tsMonthBegin      =3
[212644][D][RainGauge.cpp:203] prefs_load(): rainMonthBegin    =201.199997
[212651][D][RainGauge.cpp:204] prefs_load(): rainPrev          =201.199997
[212658][D][RainGauge.cpp:205] prefs_load(): rainAcc           =0.000000
[212664][D][RainGauge.cpp:280] update(): rainDelta: 0.0
[212669][D][RainGauge.cpp:325] update(): t_delta: 24
[212674][D][RainGauge.cpp:345] update(): hist[8]=0 (upd)
[212679][D][RainGauge.cpp:383] update(): hist[]={-1, -1, -1, -1, -1, -1, -1, -1, 0, -1, }
[212698][I][BresserWeatherSensorMQTT.ino:704] publishWeatherdata(): ESPWeather-5ACF04/f2/data: {"id":242,"ch":0,"battery_ok":1,"temp_c":10.6,"humidity":93,"wind_gust":0.0,"wind_avg":0.0,"wind_dir":67.5,"rain":201.2,"rain_h":0.0,"rain_d":0.0,"rain_w":0.0,"rain_m":0.0}

[212725][I][BresserWeatherSensorMQTT.ino:716] publishWeatherdata(): ESPWeather-5ACF04/extra: {"wind_dir_txt":"ENE","wind_gust_bft":0,"wind_avg_bft":0,"dewpoint_c":9.5,"perceived_temp_c":10.6}

[212743][I][BresserWeatherSensorMQTT.ino:818] loop(): ESPWeather-5ACF04/status: online

[212752][I][BresserWeatherSensorMQTT.ino:733] publishRadio(): ESPWeather-5ACF04/radio: {"rssi":-83.5}
```

> Are you sure you are receiving your own sensor and not a neighbors?
Yes

> The base station is just a receiver. If you mean additional indoor sensors transmitting at 868 MHz - those should be supported.
Thanks for clarifing :)

matthias-bs, 2024-04-30:
Hi @dudzio12 ,

> Thanks for your anserws @matthias-bs
> 
> I've had some time to run BresserWeatherSensorBasic example and there's no difference. I've also upgraded lib to newest version.
> 
> In serial monitor I see:
> 
> ```
> [212540][D][WeatherSensor.cpp:1267] decodeBresser7In1Payload(): Digest check failed - [46A7] vs [0186] (4721)
> [212549][D][WeatherSensor.cpp:994] decodeBresser6In1Payload(): Digest check failed - [EC0D] != [DFBD]
> ```
> 
> Is it normal that those those functions run when my config is like below? I'm not an expert of Arduino IDE, but I've placed main .ino and WeatherSensorCfg.h files inside one catalog and edited both of those from IDE. I assume that .ino file uses that .h file placed next to it and not from library itself.

Yes, that's normal. Unfortunately the Arduino IDE still takes the header files from the library - it gave me a headache a long while ago, too. Therefore I made [examples/BresserWeatherSensorMQTTCustom](https://github.com/matthias-bs/BresserWeatherSensorReceiver/tree/main/examples/BresserWeatherSensorMQTTCustom) to show how to create a configuration specifically for a single sketch.

> ```
> // #define BRESSER_5_IN_1
> // #define BRESSER_6_IN_1
> #define BRESSER_7_IN_1
> // #define BRESSER_LIGHTNING
> // #define BRESSER_LEAKAGE
> ```
> 
> Here you can see my full log. I have no UV recordings which should be available. Everything beside that is working as expected :)

Here you can see the mapping of your data to the 5-in-1 protocol:
[image]

Even the two unused nibbles next to "T" and "W" are zero. You can cover the UV sensor and check if you find any changes in the raw data, but I doubt it.

> ```
> [200518][V][WeatherSensor.cpp:344] getMessage(): [SX1262] Data: D4 EC 0D 7F FF CF FF EF F9 FE 6C ED DF FF 13 F2 80 00 30 00 10 06 01 93 12 20 00 
> [200531][D][WeatherSensor.cpp:346] getMessage(): [SX1262] R [D4] RSSI: -84.0
> [200538][D][WeatherSensor.cpp:1267] decodeBresser7In1Payload(): Digest check failed - [46A7] vs [0186] (4721)
> [200548][D][WeatherSensor.cpp:994] decodeBresser6In1Payload(): Digest check failed - [EC0D] != [DFBD]
> [200557][V][WeatherSensor.cpp:473] findSlot(): find_slot(): ID=000000F2
> [200563][D][WeatherSensor.cpp:511] findSlot(): sensor[0]: v=0 id=0x000000F2 t=0 c=0
> [200571][V][WeatherSensor.cpp:536] findSlot(): find_slot(): Storing into slot #0
> [212520][V][WeatherSensor.cpp:344] getMessage(): [SX1262] Data: D4 EC 0D 7F FF CF FF EF F9 FE 6C ED DF FF 13 F2 80 00 30 00 10 06 01 93 12 20 00 
> [212533][D][WeatherSensor.cpp:346] getMessage(): [SX1262] R [D4] RSSI: -83.5
> [212540][D][WeatherSensor.cpp:1267] decodeBresser7In1Payload(): Digest check failed - [46A7] vs [0186] (4721)
> [212549][D][WeatherSensor.cpp:994] decodeBresser6In1Payload(): Digest check failed - [EC0D] != [DFBD]
> [212558][V][WeatherSensor.cpp:473] findSlot(): find_slot(): ID=000000F2
> [212565][D][WeatherSensor.cpp:511] findSlot(): sensor[0]: v=0 id=0x000000F2 t=0 c=0
> [212572][V][WeatherSensor.cpp:536] findSlot(): find_slot(): Storing into slot #0
> [212588][D][RainGauge.cpp:194] prefs_load(): lastUpdate        =1714427498
> [212594][D][RainGauge.cpp:195] prefs_load(): startupPrev       =0
> [212600][D][RainGauge.cpp:196] prefs_load(): rainPreStartup    =201.199997
> [212607][D][RainGauge.cpp:197] prefs_load(): tsDayBegin        =1
> [212613][D][RainGauge.cpp:198] prefs_load(): rainDayBegin      =201.199997
> [212620][D][RainGauge.cpp:199] prefs_load(): tsWeekBegin       =1
> [212626][D][RainGauge.cpp:200] prefs_load(): rainWeekBegin     =201.199997
> [212632][D][RainGauge.cpp:201] prefs_load(): wdayPrev          =1
> [212638][D][RainGauge.cpp:202] prefs_load(): tsMonthBegin      =3
> [212644][D][RainGauge.cpp:203] prefs_load(): rainMonthBegin    =201.199997
> [212651][D][RainGauge.cpp:204] prefs_load(): rainPrev          =201.199997
> [212658][D][RainGauge.cpp:205] prefs_load(): rainAcc           =0.000000
> [212664][D][RainGauge.cpp:280] update(): rainDelta: 0.0
> [212669][D][RainGauge.cpp:325] update(): t_delta: 24
> [212674][D][RainGauge.cpp:345] update(): hist[8]=0 (upd)
> [212679][D][RainGauge.cpp:383] update(): hist[]={-1, -1, -1, -1, -1, -1, -1, -1, 0, -1, }
> [212698][I][BresserWeatherSensorMQTT.ino:704] publishWeatherdata(): ESPWeather-5ACF04/f2/data: {"id":242,"ch":0,"battery_ok":1,"temp_c":10.6,"humidity":93,"wind_gust":0.0,"wind_avg":0.0,"wind_dir":67.5,"rain":201.2,"rain_h":0.0,"rain_d":0.0,"rain_w":0.0,"rain_m":0.0}
> 
> [212725][I][BresserWeatherSensorMQTT.ino:716] publishWeatherdata(): ESPWeather-5ACF04/extra: {"wind_dir_txt":"ENE","wind_gust_bft":0,"wind_avg_bft":0,"dewpoint_c":9.5,"perceived_temp_c":10.6}
> 
> [212743][I][BresserWeatherSensorMQTT.ino:818] loop(): ESPWeather-5ACF04/status: online
> 
> [212752][I][BresserWeatherSensorMQTT.ino:733] publishRadio(): ESPWeather-5ACF04/radio: {"rssi":-83.5}
> ```
> 
> > Are you sure you are receiving your own sensor and not a neighbors?
> 
> Yes
> 
> > The base station is just a receiver. If you mean additional indoor sensors transmitting at 868 MHz - those should be supported.
> 
> Thanks for clarifying :)