matthias-bs/BresserWeatherSensorReceiver #160: Good morning

matthias-bs, 2024-04-26:
Good morning
I am looking for the wiring to use BresserWeatherSensorReceiver it with an ESP32 and a CC1101.
Here are the pins of my CC1101
1 GND
2 3.3V
3 GDO0
4 CKN
5 SCK
6 MOSI
7 MISO/GDO1
8 GDO2

can you help me connect my ESP32 and my CC1101
Thank you

_Ursprünglich gepostet von @Piratejim13 in https://github.com/matthias-bs/BresserWeatherSensorReceiver/issues/14#issuecomment-2078997770_

matthias-bs, 2024-04-26:
@piratejim13 Which board are you using?

Did you read https://github.com/matthias-bs/BresserWeatherSensorReceiver?tab=readme-ov-file#configuration? If you use any of the predefined boards, you just have to select in in the Arduino IDE and do not have to change the code.

Otherwise you can use the section starting at https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/4b19519aa1ac30daf6b9cd71a07b50f255f97358/src/WeatherSensorCfg.h#L559C5-L559C50 as a template and use the default SPI pins of your board.

Piratejim13, 2024-04-27:
Good morning

I use an ESP32 Dev Module in IDE and a CC1101.
I have read the documents cited but I cannot find the pin equivalences.
Here are the pins of my CC1101:
1 GND
2 3.3V
3 GDO0
4 CKN
5 SCK
6 MOSI
7 MISO/GDO1
8 GDO2

Best Regard

matthias-bs, 2024-04-27:
Hi,

please find the connections from ESP32 GPIO pins in  the comments below: 

```
#elif defined(ESP32)
    // Generic pinning for ESP32 development boards
    #define PIN_RECEIVER_CS   27 // connect to CKN (this is unusual, should actually be CSN)

    // CC1101: GDO0 / RFM95W/SX127x: G0
    #define PIN_RECEIVER_IRQ  21 // connect to GDO0

    // CC1101: GDO2 / RFM95W/SX127x: G1
    #define PIN_RECEIVER_GPIO 33 // connect to GDO2

    // RFM95W/SX127x - GPIOxx / CC1101 - RADIOLIB_NC
    #define PIN_RECEIVER_RST  32 // leave unconnected
```

For that matter, we assume that your board has those pins and that they are available. You did not provide any information about your ESP32 board.

The SPI pins SCK, MISO and MOSI should be marked on your ESP32 board as well - connect those 1:1 to your board. There is no guarantee, but the assignment SCK - GPIO18, MOSI - GPIO23 and MISO - GPIO19 seems to be common.

The remaining pins 3.3V  and GND should also be marked on the ESP32 board and have to be connected 1:1 to the CC1101 module.

If things are still unclear, please provide some information about your ESP32 board. If nothing else is available, a photo with the pin markings (and maybe a hint about manufacturer/model) could help.

Cheers,
Matthias

Piratejim13, 2024-05-01:
Good morning
Excuse me for bothering you again but I have problems with the
management program for a Bresser 7_in_1.
the WS.Begin command cancels all printing on my serial monitor so I
cannot debug the program

could you guide me on the problem?
thanks in advance

best regard

matthias-bs, 2024-05-01:
Good evening,

Most likely the program enters an endless loop, because the configuration of the CC1101 fails. This can be caused by a wrong wiring or (less likely) a defective CC1101.

Maybe I can help if you provide some information about your boards, your wiring and your SW config.

Best regards,
Matthias

Piratejim13, 2024-05-02:
Good morning
Here is attached my configuration, the photos of my cards and the wiring

thank you for your help

Sincerely

matthias-bs, 2024-05-02:
I don't see any attachment.

Piratejim13, 2024-05-02:
***@***.*** sent you documents pour debug

2 items, 774 KB in total ・ Expires on 9 May, 2024

Download link:
https://wetransfer.com/downloads/35d958d9621b8b8dc5b382840348ab1e20240502094034/f63592c971312ce8c1a9d765ae4456bf20240502094117/16cee6?trk=TRN_TDL_01&utm_campaign=TRN_TDL_01&utm_medium=email&utm_source=sendgrid

2 items

WeatherSensorCfg.h - 23.2 KB
cablage.pdf - 751 KB

[Get more out of WeTransfer, get Pro](https://wetransfer.com/pricing?trk=TRN_TDL_01&utm_campaign=TRN_TDL_01&utm_medium=email&utm_source=sendgrid)

About WeTransfer: https://wetransfer.com/about
Help: https://wetransfer.zendesk.com/hc/en-us
Legal: https://wetransfer.com/legal/terms

To make sure our emails arrive, please add ***@***.*** to [your contacts](https://wetransfer.zendesk.com/hc/en-us/articles/204909429).

Piratejim13, 2024-05-02:
Hi

I sent the documents on wetranfert

matthias-bs, 2024-05-02:
The preferred way is to attach/post information via the Github website in the issue.

Piratejim13, 2024-05-02:
the documents
[cablage.pdf](https://github.com/matthias-bs/BresserWeatherSensorReceiver/files/15186764/cablage.pdf)
[WeatherSensorCfg.txt](https://github.com/matthias-bs/BresserWeatherSensorReceiver/files/15186781/WeatherSensorCfg.txt)

matthias-bs, 2024-05-03:
Hi, 

as can bee seen on your photo, you are using a 433 MHz version of the CC1101 module. This will not work, you need an 868 MHz version.

see https://github.com/matthias-bs/BresserWeatherSensorReceiver?tab=readme-ov-file#cc1101

Cheers,
Matthias

Piratejim13, 2024-05-03:
Hi

the CC1101 I use does both frequencies. moreover it already worked
with a previous version of your application but it no longer works
with the update.

[image: image.png]
 Best regard

matthias-bs, 2024-05-03:
Please make yourself familiar with posting images or code into Github issues. People are less willing to help if unnecessary effort is required on their side.

Can you please add your compile logfile and the output of the serial console (with debug enabled)?

matthias-bs, 2024-05-03:
Please change the code section in `BresserWeatherSensorCfg.h` as shown below (i.e. comment out the second define):
```
#elif defined(ARDUINO_ESP32_DEV) || defined(ARDUINO_DFROBOT_FIREBEETLE_ESP32)
    //#define LORAWAN_NODE
    //#define FIREBEETLE_ESP32_COVER_LORA
```

Piratejim13, 2024-05-06:
Good morning
thank you for your help
I finally manage to transmit the values ​​of my 7_in_1 in mqtt
[image]

best regard

closed 2024-05-06 by matthias-bs