matthias-bs/BresserWeatherSensorReceiver #161: Rain values reset unwanted behaviour.

Shadowpost, 2024-05-02:
- MQTT WiFiMgr
 
My daily rain sensor has reset itself at 14:29 yesterday.
My monthly rain sensor has not reset itself from april to may

I thought the Arduino code has a coupling to UTC to achieve just that?

Have I forgotten a step?

matthias-bs, 2024-05-03:
Hi,

no, you  actually found a bug.

You have probably set `USE_WIFI`. In this case, the RTC will not be set via SNTP. I forgot to change this after adding the rain statistics.

You can simply fix this in the *.ino file by moving the line `#ifdef USE_SECUREWIFI` (l. 602) at the begin of `mqtt_setup()` down after the line `log_i("Current time (GMT): %s", asctime(&timeinfo));` (l. 624). 

Best regards,
Matthias

matthias-bs, 2024-05-03:
Fixed in https://github.com/matthias-bs/BresserWeatherSensorReceiver/pull/162 / [v0.27.3](https://github.com/matthias-bs/BresserWeatherSensorReceiver/releases/tag/v0.27.3)

Shadowpost, 2024-05-06:
I have tested the revised sketches and the rain sensor resets at precisely 01:00. I did set the timezone in the sketch according the link.

I have tried to have the time printed with Serial.print, but I am not able to reverse engineer the sketch.

A workaround would be to add a helper in Home Assistant based on the total rain value, but that would only fix the issue for me.

matthias-bs, 2024-05-06:
I don't quite understand where the difference between your expected behaviour and the actual behaviour is.
The intended behaviour is described here: https://github.com/matthias-bs/BresserWeatherSensorReceiver/wiki/03.-Implementing-Rain-Gauge-Statistics.

If the reset time is one hour off, this could mean that summer time is not considered as desired.

By enabling debug messages (see https://github.com/matthias-bs/BresserWeatherSensorReceiver?tab=readme-ov-file#debug-output-configuration), you should also see date and time in the serial console after startup / wake-up. This is implemented in `printDateTime()`.

matthias-bs, 2024-06-02:
Closed due to inactivity.

closed 2024-06-02 by matthias-bs