matthias-bs/BresserWeatherSensorReceiver #165: Help with Firebeetle 2/Lora Cover and Bresser 7-in-1 Clearview

alucryd, 2024-05-07:
Hi there, I bought the Bresser ClearView station (https://www.bresser.de/en/Weather-Time/BRESSER-WIFI-ClearView-Weather-Station-with-7-in-1-Sensor.html) and had a Firebeetle 2 (not 1) lying around (https://www.dfrobot.com/product-2195.html) so I got the DFRobot 868 LoRa cover to go with it, only to realize it was designed for the first iteration of FireBeetle, not the second. I thought I could get it to work regardless but I'm stumped.

I used the same bridges on the LoRa cover, but had to use different pins on the FireBeetle itself, as D4 is actually used for the flash button. Instead I mapped D2, D3, D4 and D5 on the cover to A0, A1, A2, and A3 on the Beetle respectively using Dupont cables.

Using the MQTT custom example as a base, I chose `BRESSER_7_IN_1`, uncommented what's needed for the regular FireBeetle, and changed the pins to:
```
    #define PIN_RECEIVER_CS   34 // D4 -> A2

    // CC1101: GDO0 / RFM95W/SX127x: G0
    #define PIN_RECEIVER_IRQ  39 // D3 -> A1

    // CC1101: GDO2 / RFM95W/SX127x: G1
    #define PIN_RECEIVER_GPIO 35  // D5 -> A3

    // RFM95W/SX127x - GPIOxx / CC1101 - RADIOLIB_NC
    #define PIN_RECEIVER_RST  36 // D2 -> A0
```
SCK, MISO and MOSI are the same on both FireBeetle versions.

I can build and flash just fine, connect to Wifi and MQTT, but I don't see sensor data being picked up. The serial console only prints `Checking wifi` and `MQTT connecting`. Any idea what could be wrong?

matthias-bs, 2024-05-07:
Hi,

Yes, while selecting "FireBeetle-ESP32" sets the define `ARDUINO_ESP32_DEV` and we have to guess that this means FireBeetle, selecting "FireBeetle 2 ESP32-W" sets `ARDUINO_DFROBOT_FIREBEETLE_2_ESP32E` (which is actually meaningful).

This means that you can put your pin definitions in a block starting with
`#elif defined(ARDUINO_DFROBOT_FIREBEETLE_2_ESP32E)`

(`ARDUINO_DFROBOT_FIREBEETLE_ESP32` might become available with arduino-esp32 release 3.x.)

matthias-bs, 2024-05-07:
If you enable "Verbose Output" in the Arduino IDE's Preferences, you can check in the compile output if the correct code section and your desired pins are actually applied.

alucryd, 2024-05-08:
Thanks, they are correctly applied, not sure what's going on. I verified the cover is working at least, I flashed OpenMQTTGateway onto it and it picks up my existing LoRa devices, however if I try to flash OpenMQTTGateway with FSK instead of LoRa it doesn't pick up the weather station. Maybe that's a config issue as far as OpenMQTTGateway is concerned though. Will keep trying stuff. How often is the weather station supposed to broadcast its data?

matthias-bs, 2024-05-08:
You can find this in the datasheet. 24 sec is the transmit interval of the weather sensors I'm familiar with.

If you share the compile log and config file, I might be able to help. I'd like to add your HW config to the list of supported devices.

matthias-bs, 2024-05-08:
The transmission interval can be found in the sensor's manual. The weather sensors I'm familiar with have an interval of 24 seconds.

alucryd, 2024-05-08:
Thanks for the quick replies.

Managed to get OpenMQTTGateway to work with FSK. While building the firmware I saw it mapped RF_RECEIVER_GPIO to DIO2 in the code, so I tried plugging DIO2 on one of the FireBeetle pins and voila, I finally received some data.

I didn't see you mention DIO2 in your readme, could that be specific to OpenMQTTGateway?

It seems OMG can handle the data out of the box so I'm probably gonna stick with it since I already have one instance for LoRa and it integrates nicely into home assistant. Unless there's something your work does better that I'm missing?

Here's what the messages look like in OMG:

```
{"model":"Bresser-7in1","id":43970,"temperature_C":9.6,"humidity":92,"wind_max_m_s":0,"wind_avg_m_s":0,"wind_dir_deg":312,"rain_mm":4.8,"light_klx":0,"light_lux":0,"uv":0,"battery_ok":1,"mic":"CRC","protocol":"Bresser Weather Center 7-in-1","rssi":-93,"duration":38706}
```

matthias-bs, 2024-05-09:
This is strange, because DIO2 is not readily available on a few boards.

If OMG works for you, then keep it. The core functionality seems to be the same. Differences might exist in rain/lightning data post processing and power management.

I'll try to build a configuration for your HW setup next week and ask you to test it.

matthias-bs, 2024-05-14:
Hi,

the cover cannot be stacked directly on top of the FireBeetle 2 board, right? That would pretty much prevent using this combination unless you already have the HW available. 

I have added the pin configuration you proposed to https://github.com/matthias-bs/BresserWeatherSensorReceiver/tree/feat-firebeetle2. Could you please try it?

Thanks and regards,
Matthias

matthias-bs, 2024-06-02:
Closed due to inactivity.

closed 2024-06-02 by matthias-bs

alucryd, 2024-06-02:
Apologies, didn't see your previous message. Yeah, it can't be stacked, but I used a bunch of Dupont cables to connect the relevant pins.

I since switched to a board l was already for LoRa though, which has been working fine for FSK. I returned the LoRa cover.