matthias-bs/BresserWeatherSensorReceiver #178: Rain_m is equal to rain_d ?

Calimerorulez, 2024-06-18:
Hi,

Using the MQTT with the Wifi Manager .ino, I was just checking for the data from the rain gauge, as I want to implement it in my Home Assistant installation.

It occurs to me that the values of `rain_m` and `rain_d` are the same, and if `rain_d` increases, `rain_m` increases at the same time.

`{"id":44009,"ch":0,"battery_ok":1,"temp_c":16.8,"humidity":89,"wind_gust":0.0,"wind_avg":0.0,"wind_dir":324.0,"uv":0.7,"light_klx":9.0,"rain":996.0,"rain_h":0.8,"rain_d":2.0,"rain_w":15.2,"rain_m":2.0}`

matthias-bs, 2024-06-18:
Hi,rain_d, rain_w and rain_m are the daily, weekly and monthly rain counters which are reset when a new day/week/month begins. Week and month are handled according to the calendar.

matthias-bs, 2024-06-18:
BTW, there is some documentation, but the implementation description needs to be updated: https://github.com/matthias-bs/BresserWeatherSensorReceiver/wiki/03.-Implementing-Rain-Gauge-Statistics

closed 2024-06-30 by matthias-bs

Calimerorulez, 2024-06-30:
Thank you for your explanation, I will look into it.