matthias-bs/BresserWeatherSensorReceiver #181: MQTT Issue with ARDUINO_TTGO_LoRa32_v21new

m0rph3usX, 2024-07-13:
**Context**
- Which example sketch are you using?
BresserWeatherSensorMQTT.ino
and tried also: 
BresserWeatherSensorMQTTWiFiMgr.ino

- Which changes to the code did you make (if any)?
Added "#define ARDUINO_TTGO_LoRa32_v21new" to WeatherSensorCfg.h @line 103

Also added WIFI info and MQTT infos @sketch line 233:

const char ssid[] = "myWIFI";
const char pass[] = "xyz";

#define HOSTNAME "ESPWeather"
#define APPEND_CHIP_ID

//#define MQTT_PORT 8883 // 
#define MQTT_PORT 1883 // note: I first tried port 8883, changed then to 1883 as in MQTT at home assistant
const char MQTT_HOST[] = "192.168.178.66";
const char MQTT_USER[] = "mqtt-user"; // leave blank if no credentials used
const char MQTT_PASS[] = "mypassword"; // leave blank if no credentials used

- Which sensor are you using?
TTGO Lora32 from this link:
https://www.lilygo.cc/products/lora3?variant=42272562282677

**Describe the bug**
I can compile the sketch successfully with following output:

`In file included from C:\Users\x\Downloads\BresserWeatherSensorReceiver-0.28.8\examples\BresserWeatherSensorMQTT\BresserWeatherSensorMQTT.ino:203:
c:\Users\x\Documents\Arduino\libraries\BresserWeatherSensorReceiver\src/WeatherSensorCfg.h:184:81: note: '#pragma message: ARDUINO_TTGO_LoRa32_V1 defined; using on-board transceiver'
  184 |     #pragma message("ARDUINO_TTGO_LoRa32_V1 defined; using on-board transceiver")
      |                                                                                 ^
c:\Users\x\Documents\Arduino\libraries\BresserWeatherSensorReceiver\src/WeatherSensorCfg.h:549:48: note: '#pragma message: Receiver chip: [SX1276]'
  549 | #pragma message("Receiver chip: " RECEIVER_CHIP)
      |                                                ^
c:\Users\x\Documents\Arduino\libraries\BresserWeatherSensorReceiver\src/WeatherSensorCfg.h:550:170: note: '#pragma message: Pin config: RST->14, CS->18, GD0/G0/IRQ->26, GDO2/G1/GPIO->33'
  550 | #pragma message("Pin config: RST->" STR(PIN_RECEIVER_RST) ", CS->" STR(PIN_RECEIVER_CS) ", GD0/G0/IRQ->" STR(PIN_RECEIVER_IRQ) ", GDO2/G1/GPIO->" STR(PIN_RECEIVER_GPIO) )
      |                                                                                                                                                                          ^
Sketch uses 1028561 bytes (78%) of program storage space. Maximum is 1310720 bytes.
Global variables use 48576 bytes (16%) of dynamic memory, leaving 246336 bytes for local variables. Maximum is 294912 bytes.
esptool.py v4.6`

**To Reproduce**
Steps to reproduce the behavior:
1. Now the device is connected to my WIFI, I can see it in my router and I can ping it.
2. Then I check my MQTT "Mosquitto broker" at home assistant and see that the device is connected
3. After a few seconds, the device is suddenly disconnected without sending any data and then reconned error

MQTT Log:

`2024-07-13 17:07:22: New client connected from 192.168.178.147:56871 as ESPWeather-A3E100 (p2, c1, k10, u'mqtt-user').
2024-07-13 17:07:26: Client ESPWeather-A3E100 disconnected.
2024-07-13 17:11:13: New connection from 192.168.178.147:50043 on port 1883.
2024-07-13 17:11:13: New client connected from 192.168.178.147:50043 as ESPWeather-A3E100 (p2, c1, k10, u'mqtt-user').
2024-07-13 17:11:17: Client ESPWeather-A3E100 disconnected.
2024-07-13 17:14:34: New connection from 192.168.178.147:59625 on port 1883.
2024-07-13 17:14:34: New client connected from 192.168.178.147:59625 as ESPWeather-A3E100 (p2, c1, k10, u'mqtt-user').
2024-07-13 17:15:00: Client ESPWeather-A3E100 has exceeded timeout, disconnecting.
2024-07-13 17:19:44: New connection from 192.168.178.147:52112 on port 1883.
2024-07-13 17:19:44: New client connected from 192.168.178.147:52112 as ESPWeather-A3E100 (p2, c1, k10, u'mqtt-user').
2024-07-13 17:19:59: Client ESPWeather-A3E100 disconnected.`

**Expected behavior**
MQTT connection without timeout and receiving data of sensor

**Debug Log**
The TTGO Lora32 doesn't have a debugger interface

**Embedded Device (please complete the following information):**
 - Arduino IDE Version: [2.3.2]
 - ESP32/ESP8266 Board Support Package Version: [e.g. ESP32 2.0.7]
 - WiFiManager [2.0.17]
 - ArduinoJson [7.1.0]
 - Adafruit BusIO [1.16.1]
 - Adafruit GFX Library [1.11.9]
 - Adafruit SSD1306 [2.5.10]
 - BresserWeatherSensorReceiver [0.28.8]
 - ESP_DoubleResetDetector [1.3.2]
 - ESPSoftwareSerial [8.1.0]
 - MQTT [2.5.2]
 - Preferences [2.1.0]
 - RadioLib [6.6.0]
 - SdFat [2.2.3]
 - WifiManager [2.0.17]
  
 - Your Board: 
[image]
[brand/model/version] (if unsure, attach a photo)
[image]

 - Selected Board in the Arduino IDE: [TTGO loRa32-OLED]

matthias-bs, 2024-07-13:
Hi,

I reformatted your output a little - three backticks to enclose a block of verbatim text.

1. You don't have to define `#define ARDUINO_TTGO_LoRa32_v21new` manually. As described here https://github.com/matthias-bs/BresserWeatherSensorReceiver?tab=readme-ov-file#predefined-board-configurations, you just have to select the correct board (if provided)
2. Your compile output shows `#pragma message("ARDUINO_TTGO_LoRa32_V1 defined; using on-board transceiver")` while you selected `#define ARDUINO_TTGO_LoRa32_v21new` (which matches your photo), so you selected the wrong board variant.
3. Regarding debug output, please see https://github.com/matthias-bs/BresserWeatherSensorReceiver?tab=readme-ov-file#debug-output-configuration

Due do the wrong board selection (different pinning between ESP32 and radio chip), the configuration of the radio chip will fail, which leads to an endless loop. That's why you encounter an MQTT timeout.

m0rph3usX, 2024-07-14:
HI matthias,
thanks for your response, ok makes sense when selected wrong board = timeout MQTT.
Ok, so reverted back to your release (undoing all my changes) = original sketch "BresserWeatherSensorMQTT.ino" from your examples.

Then changed the code (removed the # from line 142 @WeatherSensorCfg.h":

`// This define is set by selecting "Board: TTGO LoRa32-OLED" / "Board Revision: TTGO LoRa32 V2.1 (1.6.1)"
// in the Arduino IDE:
#define ARDUINO_TTGO_LoRa32_V21new`

And selected (as before) the board "TTGO LoRa32-OLED" in Arduino IDE (as also described in your comment in the code):

[image]

So, in my opinion the selected board should be correct?!

`In file included from C:\Users\x\Downloads\aaa\BresserWeatherSensorReceiver-0.28.8\examples\BresserWeatherSensorMQTT\BresserWeatherSensorMQTT.ino:203:
c:\Users\x\Documents\Arduino\libraries\BresserWeatherSensorReceiver\src/WeatherSensorCfg.h:184:81: note: '#pragma message: ARDUINO_TTGO_LoRa32_V1 defined; using on-board transceiver'
  184 |     #pragma message("ARDUINO_TTGO_LoRa32_V1 defined; using on-board transceiver")
      |                                                                                 ^
c:\Users\x\Documents\Arduino\libraries\BresserWeatherSensorReceiver\src/WeatherSensorCfg.h:549:48: note: '#pragma message: Receiver chip: [SX1276]'
  549 | #pragma message("Receiver chip: " RECEIVER_CHIP)
      |                                                ^
c:\Users\x\Documents\Arduino\libraries\BresserWeatherSensorReceiver\src/WeatherSensorCfg.h:550:170: note: '#pragma message: Pin config: RST->14, CS->18, GD0/G0/IRQ->26, GDO2/G1/GPIO->33'
  550 | #pragma message("Pin config: RST->" STR(PIN_RECEIVER_RST) ", CS->" STR(PIN_RECEIVER_CS) ", GD0/G0/IRQ->" STR(PIN_RECEIVER_IRQ) ", GDO2/G1/GPIO->" STR(PIN_RECEIVER_GPIO) )
      |                                                                                                                                                                          ^
In file included from c:\Users\x\Documents\Arduino\libraries\BresserWeatherSensorReceiver\src\Lightning.h:72,
                 from c:\Users\x\Documents\Arduino\libraries\BresserWeatherSensorReceiver\src\Lightning.cpp:75:
c:\Users\x\Documents\Arduino\libraries\BresserWeatherSensorReceiver\src\WeatherSensorCfg.h:184:81: note: '#pragma message: ARDUINO_TTGO_LoRa32_V1 defined; using on-board transceiver'
  184 |     #pragma message("ARDUINO_TTGO_LoRa32_V1 defined; using on-board transceiver")
      |                                                                                 ^
c:\Users\x\Documents\Arduino\libraries\BresserWeatherSensorReceiver\src\WeatherSensorCfg.h:549:48: note: '#pragma message: Receiver chip: [SX1276]'
  549 | #pragma message("Receiver chip: " RECEIVER_CHIP)
      |                                                ^
c:\Users\x\Documents\Arduino\libraries\BresserWeatherSensorReceiver\src\WeatherSensorCfg.h:550:170: note: '#pragma message: Pin config: RST->14, CS->18, GD0/G0/IRQ->26, GDO2/G1/GPIO->33'
  550 | #pragma message("Pin config: RST->" STR(PIN_RECEIVER_RST) ", CS->" STR(PIN_RECEIVER_CS) ", GD0/G0/IRQ->" STR(PIN_RECEIVER_IRQ) ", GDO2/G1/GPIO->" STR(PIN_RECEIVER_GPIO) )
      |                                                                                                                                                                          ^
In file included from c:\Users\x\Documents\Arduino\libraries\BresserWeatherSensorReceiver\src\RainGauge.cpp:67:
c:\Users\x\Documents\Arduino\libraries\BresserWeatherSensorReceiver\src\WeatherSensorCfg.h:184:81: note: '#pragma message: ARDUINO_TTGO_LoRa32_V1 defined; using on-board transceiver'
  184 |     #pragma message("ARDUINO_TTGO_LoRa32_V1 defined; using on-board transceiver")
      |                                                                                 ^
c:\Users\x\Documents\Arduino\libraries\BresserWeatherSensorReceiver\src\WeatherSensorCfg.h:549:48: note: '#pragma message: Receiver chip: [SX1276]'
  549 | #pragma message("Receiver chip: " RECEIVER_CHIP)
      |                                                ^
c:\Users\x\Documents\Arduino\libraries\BresserWeatherSensorReceiver\src\WeatherSensorCfg.h:550:170: note: '#pragma message: Pin config: RST->14, CS->18, GD0/G0/IRQ->26, GDO2/G1/GPIO->33'
  550 | #pragma message("Pin config: RST->" STR(PIN_RECEIVER_RST) ", CS->" STR(PIN_RECEIVER_CS) ", GD0/G0/IRQ->" STR(PIN_RECEIVER_IRQ) ", GDO2/G1/GPIO->" STR(PIN_RECEIVER_GPIO) )
      |                                                                                                                                                                          ^
In file included from c:\Users\x\Documents\Arduino\libraries\BresserWeatherSensorReceiver\src\WeatherSensorDecoders.cpp:58:
c:\Users\x\Documents\Arduino\libraries\BresserWeatherSensorReceiver\src\WeatherSensorCfg.h:184:81: note: '#pragma message: ARDUINO_TTGO_LoRa32_V1 defined; using on-board transceiver'
  184 |     #pragma message("ARDUINO_TTGO_LoRa32_V1 defined; using on-board transceiver")
      |                                                                                 ^
c:\Users\x\Documents\Arduino\libraries\BresserWeatherSensorReceiver\src\WeatherSensorCfg.h:549:48: note: '#pragma message: Receiver chip: [SX1276]'
  549 | #pragma message("Receiver chip: " RECEIVER_CHIP)
      |                                                ^
[... cut ...]
c:\Users\x\Documents\Arduino\libraries\BresserWeatherSensorReceiver\src\WeatherSensorCfg.h:549:48: note: '#pragma message: Receiver chip: [SX1276]'
  549 | #pragma message("Receiver chip: " RECEIVER_CHIP)
      |                                                ^
c:\Users\x\Documents\Arduino\libraries\BresserWeatherSensorReceiver\src\WeatherSensorCfg.h:550:170: note: '#pragma message: Pin config: RST->14, CS->18, GD0/G0/IRQ->26, GDO2/G1/GPIO->33'
  550 | #pragma message("Pin config: RST->" STR(PIN_RECEIVER_RST) ", CS->" STR(PIN_RECEIVER_CS) ", GD0/G0/IRQ->" STR(PIN_RECEIVER_IRQ) ", GDO2/G1/GPIO->" STR(PIN_RECEIVER_GPIO) )`

matthias-bs, 2024-07-14:
Please look for the menu entry 'Board variant' and select the appropriate one!

grafik.png (view on web)

m0rph3usX, 2024-07-14:
Oh man, thanks for this hint!
Didn't know that Arduino IDE has a submenu for selecting the "Board Revision".
Changed the revision depending my board "TTGO lora32 V.2.1 (1.6.1):
[image]

Still getting the pragma message when compiling:
`In file included from C:\Users\x\Downloads\BresserWeatherSensorReceiver\examples\BresserWeatherSensorMQTTWifiMgr\BresserWeatherSensorMQTTWifiMgr.ino:225:
c:\Users\x\Documents\Arduino\libraries\BresserWeatherSensorReceiver\src/WeatherSensorCfg.h:193:85: note: '#pragma message: ARDUINO_TTGO_LoRa32_V21new defined; using on-board transceiver'
  193 |     #pragma message("ARDUINO_TTGO_LoRa32_V21new defined; using on-board transceiver")
      |                                                                                     ^
c:\Users\x\Documents\Arduino\libraries\BresserWeatherSensorReceiver\src/WeatherSensorCfg.h:549:48: note: '#pragma message: Receiver chip: [SX1276]'
  549 | #pragma message("Receiver chip: " RECEIVER_CHIP)
      |                                                ^
c:\Users\x\Documents\Arduino\libraries\BresserWeatherSensorReceiver\src/WeatherSensorCfg.h:550:170: note: '#pragma message: Pin config: RST->23, CS->18, GD0/G0/IRQ->26, GDO2/G1/GPIO->33'
  550 | #pragma message("Pin config: RST->" STR(PIN_RECEIVER_RST) ", CS->" STR(PIN_RECEIVER_CS) ", GD0/G0/IRQ->" STR(PIN_RECEIVER_IRQ) ", GDO2/G1/GPIO->" STR(PIN_RECEIVER_GPIO) )
      |     `

matthias-bs, 2024-07-14:
Yes, this time the pragma message shows that everything is fine.

m0rph3usX, 2024-07-14:
Hm ok thanks for your help.
So I uploaded the project and the issue remains still the same, MQTT timeout at home assistant:

`2024-07-14 13:06:38: Client ESPWeather-A3E100 has exceeded timeout, disconnecting.
2024-07-14 13:11:27: New connection from 192.168.178.147:52673 on port 1883.
2024-07-14 13:11:27: New client connected from 192.168.178.147:52673 as ESPWeather-A3E100 (p2, c1, k10, u'mqtt-user').
2024-07-14 13:11:39: Client ESPWeather-A3E100 disconnected.
2024-07-14 13:16:36: New connection from 192.168.178.147:56551 on port 1883.
2024-07-14 13:16:36: New client connected from 192.168.178.147:56551 as ESPWeather-A3E100 (p2, c1, k10, u'mqtt-user').
2024-07-14 13:16:47: Client ESPWeather-A3E100 disconnected.
2024-07-14 13:22:37: New connection from 192.168.178.147:57449 on port 1883.
2024-07-14 13:22:37: New client connected from 192.168.178.147:57449 as ESPWeather-A3E100 (p2, c1, k10, u'mqtt-user').
2024-07-14 13:22:56: Client ESPWeather-A3E100 has exceeded timeout, disconnecting.
2024-07-14 13:27:31: New connection from 192.168.178.147:58567 on port 1883.
2024-07-14 13:27:31: New client connected from 192.168.178.147:58567 as ESPWeather-A3E100 (p2, c1, k10, u'mqtt-user').
2024-07-14 13:27:37: Client ESPWeather-A3E100 disconnected.
2024-07-14 13:32:35: New connection from 192.168.178.147:54564 on port 1883.
2024-07-14 13:32:35: New client connected from 192.168.178.147:54564 as ESPWeather-A3E100 (p2, c1, k10, u'mqtt-user').
2024-07-14 13:32:37: Client ESPWeather-A3E100 disconnected.`

I also enabled
`#define _DEBUG_MQTT_`
to try to get some dummy data.... but still getting MQTT timeout.

Hm I'll try to modify the code, to get the debug infos via OLED instead of non existing debugging interface of this bord...

BTW it is a Bresser WSX3001.

matthias-bs, 2024-07-14:
Would you please kindly follow my advice and read the following regarding debug output via USB?
https://github.com/matthias-bs/BresserWeatherSensorReceiver?tab=readme-ov-file#debug-output-configuration

m0rph3usX, 2024-07-14:
Ok thanks got it. I already enabled the debug messages, but the "serial monitor" was hidden, instead "gdb-server1" tab was showing with an error... So I thought debugging wouldn't be available.

Here's the log, which seems to get valid data from the sensor, and the timeout @ MQTT is because of the sleep of 300000ms, but that would be ok. I thought the log of MQTT (mosquitto broker) would also show the received data...

`[ 18833][D][WeatherSensorDecoders.cpp:104] findSlot(): sensor[0]: v=0 id=0x00000000 t=0 c=0
[ 18851][D][RainGauge.cpp:194] prefs_load(): lastUpdate        =1720958307
[ 18858][D][RainGauge.cpp:195] prefs_load(): startupPrev       =1
[ 18864][D][RainGauge.cpp:196] prefs_load(): rainPreStartup    =0.000000
[ 18870][D][RainGauge.cpp:197] prefs_load(): tsDayBegin        =0
[ 18876][D][RainGauge.cpp:198] prefs_load(): rainDayBegin      =2023.300049
[ 18883][D][RainGauge.cpp:199] prefs_load(): tsWeekBegin       =6
[ 18889][D][RainGauge.cpp:200] prefs_load(): rainWeekBegin     =144.500000
[ 18896][D][RainGauge.cpp:201] prefs_load(): wdayPrev          =0
[ 18902][D][RainGauge.cpp:202] prefs_load(): tsMonthBegin      =6
[ 18908][D][RainGauge.cpp:203] prefs_load(): rainMonthBegin    =144.500000
[ 18914][D][RainGauge.cpp:204] prefs_load(): rainPrev          =2023.300049
[ 18921][D][RainGauge.cpp:205] prefs_load(): rainAcc           =2023.300049
[ 18928][D][RainGauge.cpp:280] update(): rainDelta: 0.0
[ 18933][D][RainGauge.cpp:325] update(): t_delta: 1061191424
[ 18939][D][RainGauge.cpp:349] update(): hist[0]=0 (new)
[ 18944][D][RainGauge.cpp:383] update(): hist[]={0, 0, -1, 0, -1, -1, -1, -1, 0, 0, }
[ 18964][I][BresserWeatherSensorMQTTWifiMgr.ino:899] publishWeatherdata(): ESPWeather-A3E100/abbb/data: {"id":43963,"ch":0,"battery_ok":1,"temp_c":24.3,"humidity":46,"wind_gust":2.8,"wind_avg":2.5,"wind_dir":252.0,"uv":10.5,"light_klx":117.7,"rain":0.0,"rain_h":0.0,"rain_d":0.0,"rain_w":1878.8,"rain_m":1878.8}

[ 18997][I][BresserWeatherSensorMQTTWifiMgr.ino:911] publishWeatherdata(): ESPWeather-A3E100/extra: {"wind_dir_txt":"WSW","wind_gust_bft":2,"wind_avg_bft":2,"dewpoint_c":11.9,"perceived_temp_c":25.5}

[ 19016][D][BresserWeatherSensorMQTTWifiMgr.ino:1097] loop(): Data forwarding completed.
[ 19024][I][BresserWeatherSensorMQTTWifiMgr.ino:1100] loop(): Sleeping for 300000 ms

[ 19032][I][BresserWeatherSensorMQTTWifiMgr.ino:1101] loop(): ESPWeather-A3E100/status: offline

`