matthias-bs/BresserWeatherSensorReceiver #182: Every second battery status is "low"

pcfritz, 2024-07-15:
**Context**
SensorBasic on TTGO LaRo OLED

**Describe the bug**
It seems that the battery status is being decoded incorrectly for some stations: 

BRESSER 5-in-1 7002510/7002511/7002512/7002513

Id: [19C01B29] Typ: [1] Ch: [0] St: [0] Bat: [OK ] RSSI: [-106.0dBm] Temp: [ 19.2C] Hum: [ 85%] Wmax: [ 0.2m/s] Wavg: [ 0.2m/s] Wdir: [158.0deg] Rain: [-----.-mm] UVidx: [0.0] Light: [0.0klx] 
Id: [19C01B29] Typ: [1] Ch: [0] St: [0] Bat: [Low] RSSI: [-109.0dBm] Temp: [---.-C] Hum: [---%] Wmax: [ 0.4m/s] Wavg: [ 0.4m/s] Wdir: [158.0deg] Rain: [   28.0mm] UVidx: [--.-] Light: [--.-klx] 
Id: [19C01B29] Typ: [1] Ch: [0] St: [0] Bat: [OK ] RSSI: [-110.5dBm] Temp: [ 19.2C] Hum: [ 85%] Wmax: [ 0.4m/s] Wavg: [ 0.4m/s] Wdir: [158.0deg] Rain: [-----.-mm] UVidx: [0.0] Light: [--.-klx] 
Id: [19C01B29] Typ: [1] Ch: [0] St: [0] Bat: [Low] RSSI: [-107.0dBm] Temp: [---.-C] Hum: [---%] Wmax: [ 0.7m/s] Wavg: [ 0.7m/s] Wdir: [158.0deg] Rain: [   28.0mm] UVidx: [--.-] Light: [--.-klx]

It works for others: 
Bresser 7002520
Id: [ 2868402] Typ: [1] Ch: [0] St: [0] Bat: [OK ] RSSI: [-107.0dBm] Temp: [---.-C] Hum: [---%] Wmax: [ 0.0m/s] Wavg: [ 0.0m/s] Wdir: [292.0deg] Rain: [  100.4mm] UVidx: [--.-] Light: [--.-klx] 
Id: [ 2868402] Typ: [1] Ch: [0] St: [0] Bat: [OK ] RSSI: [-110.0dBm] Temp: [ 23.5C] Hum: [ 61%] Wmax: [ 0.0m/s] Wavg: [ 0.0m/s] Wdir: [292.0deg] Rain: [-----.-mm] UVidx: [44.4] Light: [--.-klx] 
Id: [ 2868402] Typ: [1] Ch: [0] St: [0] Bat: [OK ] RSSI: [-108.0dBm] Temp: [---.-C] Hum: [---%] Wmax: [ 0.0m/s] Wavg: [ 0.0m/s] Wdir: [292.0deg] Rain: [  100.4mm] UVidx: [--.-] Light: [--.-klx] 
Id: [ 2868402] Typ: [1] Ch: [0] St: [0] Bat: [OK ] RSSI: [-107.0dBm] Temp: [ 23.5C] Hum: [ 61%] Wmax: [ 0.0m/s] Wavg: [ 0.0m/s] Wdir: [292.0deg] Rain: [-----.-mm] UVidx: [44.4] Light: [--.-klx] 
Id: [ 2868402] Typ: [1] Ch: [0] St: [0] Bat: [OK ] RSSI: [-107.0dBm] Temp: [---.-C] Hum: [---%] Wmax: [ 0.0m/s] Wavg: [ 0.0m/s] Wdir: [292.0deg] Rain: [  100.4mm] UVidx: [--.-] Light: [--.-klx]

**Expected behavior**
The output of the battery status should work for all stations

matthias-bs, 2024-07-16:
[BresserWeatherSensorBasic](examples/BresserWeatherSensorBasic) just shows each message as it is received from a sensor by using the function https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/80b5c59f86fb8e14af1defdc9001bd6e145d6311/src/WeatherSensor.cpp#L319
Sensors which using the 6-in-1 protocol are actually transmitting two kind of messages alternately:
1. Temperature, Humidity, Wind and Battery status
2. Wind and Rain

All other protocols use a single type of message which contains a complete set of data.

Other examples are using the function
https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/80b5c59f86fb8e14af1defdc9001bd6e145d6311/src/WeatherSensor.cpp#L245
which buffers and combines messages from the 6-in-1 protocol until a complete set of data (with some configuration options regarding completeness, see [BresserWeatherSensorOptions](examples/BresserWeatherSensorOptions)) is available.

So you actually see the peculiarity 6-in-1 protocol with the first sensor and the straightforward decoding with the second one.

It's not a bug in the decoder, but this behaviour could probably described in more details in the documentation.

matthias-bs, 2024-07-16:
See #185

matthias-bs, 2024-07-30:
Added note to [README.md](/matthias-bs/BresserWeatherSensorReceiver/blob/main/README.md)

closed 2024-07-30 by matthias-bs