matthias-bs/BresserWeatherSensorReceiver #183: 5-in-1 (7002580) fails on parity but works with 6-in-1 decode (though rainfall is incorrect)

simonsh, 2024-07-16:
**Context**
- Which example sketch are you using? - BresserWeatherCallback.ino
- Which changes to the code did you make (if any)? - just config changes to 6-in-1
- Which sensor are you using? 7002580

**Describe the bug**
5-in-1 decoder returns a parity error. 6-in-1 decoder receives data but rainfall appears incorrect i.e. with no rain it reports **83.2mm**

**To Reproduce**
Steps to reproduce the behavior:
Run sketch on a my 7002580 model (about 4 years old) with 5-in-1 configured then 6-in-1 configured.

**Expected behavior**
5-in-1 decoder should work on 7002580 and rainfall should be correct.

**Debug Log**
Output from BresserWeatherCallback.ino with 6-in-1 configured (there is no UV sensor in the 7002580) :

Id: [1930031E] Typ: [1] Ch: [0] St: [0] Bat: [OK ] RSSI: [ -77.5dBm] Temp: [ 30.6C] Hum: [ 41%] Wmax: [ 0.0m/s] Wavg: [ 0.0m/s] Wdir: [180.0deg] Rain: [   **83.2mm**] UVidx: [0.0] 
[ 56524][V][WeatherSensor.cpp:345] getMessage(): [SX1276] Data: D4 3A B7 19 30 03 1E 18 FF FF FF 18 08 30 72 41 FF F0 8E 00 00 00 00 00 00 00 00 
[ 56537][D][WeatherSensor.cpp:347] getMessage(): [SX1276] R [D4] RSSI: -77.5
[ 56544][V][WeatherSensorDecoders.cpp:66] findSlot(): find_slot(): ID=1930031E
[ 56551][D][WeatherSensorDecoders.cpp:104] findSlot(): sensor[0]: v=0 id=0x1930031E t=1 c=0
[ 56559][V][WeatherSensorDecoders.cpp:129] findSlot(): find_slot(): Storing into slot #0
[ 56567][D][WeatherSensorDecoders.cpp:581] decodeBresser6In1Payload(): Flags: Temp=1  Hum=1  Wind=1  Rain=0  UV=1
[ 68524][V][WeatherSensor.cpp:345] getMessage(): [SX1276] Data: D4 BC 8E 19 30 03 1E 18 FF FF FF 18 08 FF F7 CD FF 01 9D 00 00 00 00 00 00 00 00 
[ 68537][D][WeatherSensor.cpp:347] getMessage(): [SX1276] R [D4] RSSI: -77.5
[ 68544][V][WeatherSensorDecoders.cpp:66] findSlot(): find_slot(): ID=1930031E
[ 68551][D][WeatherSensorDecoders.cpp:104] findSlot(): sensor[0]: v=1 id=0x1930031E t=1 c=0
[ 68559][V][WeatherSensorDecoders.cpp:122] findSlot(): find_slot(): Updating slot #0
[ 68567][D][WeatherSensorDecoders.cpp:581] decodeBresser6In1Payload(): Flags: Temp=0  Hum=0  Wind=1  Rain=1  UV=0

**Embedded Device (please complete the following information):**
 - Arduino IDE Version: 2.3.1
 - ESP32/ESP8266 Board Support Package Version: ESP32 2.0.15
 - Library Version: 0.28.8 
 - Your Board: LILYGO (TTGO) LoRa32 868Mhz ESP32 with OLED
 - Selected Board in the Arduino IDE: TTGO LoRa-32 OLED Revision: TTGO LoRa-32 V2.1 (1.6.1)
 - **Pinning/Wiring** none, just the USB serial connection

matthias-bs, 2024-07-16:
* Newer 5-in-1 sensors use the 6-in-1 protocol, that's why the 5-in-1 decoder reports a parity error and the 6-in-1 decoder succeeds
* Automatic detection of the UV value's availability is not fully implemented yet
* `BresserWeatherCallback.ino` reports the raw rain value provided in the sensor message. There is a post-processing function (see https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/README.md#rain-statistics) which is applied in the MQTT/Domiticz example sketches

simonsh, 2024-07-16:
Understood, working on SNTP now.

simonsh, 2024-07-16:
I will probably just use the Wunderground API for this, for now. Just need a daily summary for rain and it's already worked out by the station and posted to the API.

matthias-bs, 2024-07-16:
> Understood, working on SNTP now.

What do you mean by that? Please not that SNTP is used the MQTT examples - to allow using secure MQTT and to provide time to the rain/lightning statistics functions.

simonsh, 2024-07-16:
I meant I was working on it. I have now configured SNTP correctly in the sketch as per the MSTT example, but I will use the Wunderground API data for rainfall data, whilst I work on other sensor stuff. I'll get back to it shortly. Thanks again.

closed 2024-07-21 by matthias-bs