matthias-bs/BresserWeatherSensorReceiver #184: Lightning and PM2.5 sensors need pairing

simonsh, 2024-07-16:
Can I use these sensors with this code without the requisite station base unit? Both sensors need pairing to a base unit but can be bought individually?

simonsh, 2024-07-16:
It arrives tomorrow on a Prime day, so happy to send any debug data I can get, if it helps. Let me know. I am on a Mac with a TTGO LoRa-32 OLED board.

matthias-bs, 2024-07-16:
Yes, you can use all supported sensors without pairing and without a matching base station. The code uses all sensors which can be received unless you apply any of the include/exclude filters.

simonsh, 2024-07-16:
Thanks for your prompt reply! I'll check on it tomorrow.

simonsh, 2024-07-17:
It works fine. Mobile phones set it off for testing. Now zeroed and waiting for a storm :-) Will calibrate later at the same time as I work on the rain gauge. Keep up your excellent work.

matthias-bs, 2024-07-17:
Thanks for the nice feedback!

closed 2024-07-17 by matthias-bs