matthias-bs/BresserWeatherSensorReceiver #188: AQ sensor reports 1.0PM

simonsh, 2024-07-23:
Is this correct, as the sensor only advertises 2.5 PM and 10 PM? I know that 2.5PM is officially <= 2.5 PM.

matthias-bs, 2024-07-23:
Yes, I think so. Please see https://github.com/matthias-bs/BresserWeatherSensorReceiver/issues/97.

simonsh, 2024-07-23:
Looks like it, have you decoded PM4 yet?

Sensirion SPS-30

Particulate matter
--
Mass concentration precision | ±10 | % |  
Mass concentration range | 0 - 1000 | μg/m³ |  
Lifetime | > 10 | years |  
Particle size range | PM1.0, PM2.5, PM4 and PM10

matthias-bs, 2024-07-23:
As I wrote in https://github.com/matthias-bs/BresserWeatherSensorReceiver/issues/97, there is one additional PM value in the radio message, which must be either PM1.0 or PM4. After comparison with the internal sensor signals (UART), I came to the conclusion that it is PM1.0 - but maybe I was wrong.

simonsh, 2024-07-23:
If it is a similar reading to PM 2.5 then it is PM 1.0 but not if higher. I think you are correct and I will keep an eye on my log. Hasn’t gone over 2.5 so far even when PM 10 is higher. Will let you know.

matthias-bs, 2024-07-30:
Please follow up in #97 if required.

closed 2024-07-30 by matthias-bs