matthias-bs/BresserWeatherSensorReceiver #189: ESP32AnalogRead error

mczakk, 2024-08-08:
I'm trying to update my ESP32 Lora boards and am having an issue with the ESP32AnalogRead this is the error i get:
```home/matt/Arduino/libraries/ESP32AnalogRead/ESP32AnalogRead.cpp: In member function 'uint32_t ESP32AnalogRead::readMiliVolts()':
/home/matt/Arduino/libraries/ESP32AnalogRead/ESP32AnalogRead.cpp:187:35: error: 'ADC_WIDTH_12Bit' was not declared in this scope; did you mean 'ADC_WIDTH_MAX'?
  187 |                 adc1_config_width(ADC_WIDTH_12Bit);
      |                                   ^~~~~~~~~~~~~~~
      |                                   ADC_WIDTH_MAX
```
All my libraries are up to to date but the code is failing to compile with this error?

matthias-bs, 2024-08-08:
Please see https://github.com/madhephaestus/ESP32AnalogRead/issues/13#issuecomment-2207225345
Did you encounter this in https://github.com/matthias-bs/BresserWeatherSensorTTN?

mczakk, 2024-08-08:
It's in my bespoke code based on yours. I'd not updated the esp32 boards for a year or so, all I wanted to do s change the text on the display! I'll try it without defining the bit that uses espanalogread, see what that does!

matthias-bs, 2024-08-09:
There are several options:
- Disable the analog inputs, but then you cannot read the battery voltage
- Switch to an arduino-esp32 version < 3.0.0
- Have a look at https://github.com/matthias-bs/BresserWeatherSensorLW/blob/main/src/adc/adc.cpp and copy the related code
- Wait until I update https://github.com/matthias-bs/BresserWeatherSensorTTN (I planned this anyway, maybe I could do this by end of next week)

matthias-bs, 2024-08-09:
Created https://github.com/matthias-bs/BresserWeatherSensorTTN/issues/92

mczakk, 2024-08-09:
I'm going to close this, i have opened another issue in BresserWeatherSensorLW if you have the time to help me there?

closed 2024-08-09 by mczakk

matthias-bs, 2024-08-13:
I just noticed that I already updated the ADC code in https://github.com/matthias-bs/BresserWeatherSensorTTN/releases/tag/v0.13.0. You seem to have missed this when comparing your code to the latest version.