matthias-bs/BresserWeatherSensorReceiver #190: Configuration of RainGauge / Lightning statistics update rate at run time

matthias-bs, 2024-08-26:
**Is your feature request related to a problem? Please describe.**
Currently the classes `Lightning` and `RainGauge` expect a fixed update rate. While a shorter update rate than expected works fine, the method `pastHour()` cannot provide valid data with a longer update rate. 

See [New Hourly Rainfall Algorithm](https://github.com/matthias-bs/BresserWeatherSensorReceiver/wiki/04.-Implementing-Rain-Gauge-Statistics#new-hourly-rainfall-algorithm)

This feature is required by [BresserWeatherSensorLW](https://github.com/matthias-bs/BresserWeatherSensorLW), which allows to control the uplink interval for energy management and uplink data traffic control.

**Describe the solution you'd like**
`LIGHTNING_HIST_SIZE` and `RAIN_HIST_SIZE` remain constant at  3600 [sec] / <update_rate> [sec], with <update_rate> being the shortest update rate implemented.

`LIGHTNING_UPD_RATE` and `RAINGAUGE_UPD_RATE` are changed to allow configuration at run time. The algorithm will be modified to use only a part of the history array if the update interval is increased.  The index calculation will be modified. The history is reset if the update rate changes, i.e. occasional switching of the update rate is supported, dynamically changing the update rate is not supported.   

Example:

Maximum update rate: 6 minutes -> HIST_SIZE = 60 min / 6 min = 10

Reduced update rate: 10 minutes -> Only the first 6 (60 min / 10 min) entries  are used.

**Describe alternatives you've considered**
--

**Additional context**
--

matthias-bs, 2024-08-26:
Setting the default update rate to the largest expected value would decrease accuracy for shorter intervals.

tonbor, 2024-11-27:
Well after compiling the rain is compleetly disoriented: 12:54:59.187 -> {"id":43961,"ch":0,"battery_ok":1,"temp_c":10.5,"humidity":92,"wind_gust":1.4,"wind_avg":1.4,"wind_dir":132.0,"uv":0.0,"light_klx":6.2,"rain":991.6,"rain_h":0.0,"rain_d":4974.0,"rain_w":4978.6,"rain_m":5062.1}

matthias-bs, 2024-11-27:
@tonbor You are commenting on an issue which I did not even start to implement...

To fix this, either set "Erase All Flash Before Sketch Upload: Enabled" in the board settings or send a "reset" message via MQTT.

tonbor, 2024-11-27:
I will try

tonbor, 2024-11-27:
Did erase All Flash, but keep my wifi in memory too. So erased all put back my wifi. Uploaded my mqtt ino result -> data = {"id":43961,"ch":0,"battery_ok":1,"temp_c":10.6,"humidity":95,"wind_gust":1.7,"wind_avg":1.5,"wind_dir":211.0,"uv":0.0,"light_klx":1.2,"rain":993.3,"rain_h":0.0,"rain_d":-72727896295666880632779.7,"rain_w":12953.1,"rain_m":-72727896295666880632779.7}
rssi v3 = -97.0
expected data = {"id":43961,"ch":0,"battery_ok":1,"temp_c":10.6,"humidity":95,"wind_gust":1.7,"wind_avg":1.5,"wind_dir":98.0,"uv":0.0,"light_klx":0.9,"rain":993.9,"rain_h":2.1,"rain_d":10.2,"rain_w":14.8,"rain_m":98.3}
rssi v3 = -113.5

tonbor, 2024-11-27:
Still a problem, erased memory with Erase All Flash Before Sketch Upload: Enabled, no more ssi in memory, still false rain info ->
data = {"id":43961,"ch":0,"battery_ok":1,"temp_c":10.8,"humidity":96,"wind_gust":3.7,"wind_avg":3.6,"wind_dir":205.0,"uv":0.0,"light_klx":0.3,"rain":994.6,"rain_h":0.0,"rain_d":22940.6,"rain_w":22940.6,"rain_m":22940.6}

matthias-bs, 2024-11-27:
Please use this issue instead: https://github.com/matthias-bs/BresserWeatherSensorReceiver/issues/202

closed 2025-03-25 by matthias-bs