matthias-bs/BresserWeatherSensorReceiver #191: Compile for esp32 S3 no error but constant reboot after start basic example

tonbor, 2024-08-31:
**Context**
- Which example sketch are you using
- basic
- Which changes to the code did you make (if any)?
- none
- Which sensor are you using?
- sx1276 connected to esp32 S3

**Describe the bug**
A clear and concise description of what the bug is.
wrong spi pins I suppose, s3 have other then esp32

**To Reproduce**
Steps to reproduce the behavior:
1. Step1 '...' load
2. Step2 '....' compile 
3. Step3 '....' run
4. Reboot after start. 

**Expected behavior**
A clear and concise description of what you expected to happen.
to run as with sx1276 with esp32 wroom

**Debug Log**
If applicable, add log file to help explain your problem.
 - Build log file (enable verbose output in the Arduino IDE: File->Preferences - "Show verbose output during: compile")
 - Serial console log file (see [Debug Output Configuration in Arduino IDE](https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/DEBUG_OUTPUT.md))

**Embedded Device (please complete the following information):**
 - Arduino IDE Version: [e.g. 1.8.19]
 - ESP32/ESP8266 Board Support Package Version: [e.g. ESP32 2.0.7]
 - Library Version [e.g. 0.6.1]
 - Your Board: [brand/model/version] (if unsure, attach a photo)
 - Selected Board in the Arduino IDE: [brand/model/version]
 - Your Radio Transceiver Module (if external): [brand/model/chipset/frequency] 
 - **Pinning/Wiring**

**Additional context**
Add any other context about the problem here.

tonbor, 2024-09-01:
See https://github.com/jgromes/RadioLib/discussions/1051, add to this library

tonbor, 2024-09-02:
Put this before #elif defined(ESP32), the pins are chosen, still constant reboot ESP32 S3, any Help?

#elif defined(CONFIG_IDF_TARGET_ESP32S3)
    // Generic pinning for ESP32 S3 development boards
#pragma message("Pin config: ESP32 S3")
    #define PIN_RECEIVER_CS   10

    // CC1101: GDO0 / RFM95W/SX127x: G0
    #define PIN_RECEIVER_IRQ  21

    // CC1101: GDO2 / RFM95W/SX127x: G1
    #define PIN_RECEIVER_GPIO 8

    // RFM95W/SX127x - GPIOxx / CC1101 - RADIOLIB_NC
    #define PIN_RECEIVER_RST  9

tonbor, 2024-09-03:
The good news is that my hardware and testings are accepted in using the Radiolib library. Tested with the examples RadioLib SX127x Settings Example and RadioLib SX127x Blocking Receive Example (SX1278 radio = new Module(10, 21, 9, 8);) So Matthias what part in your library must be adapted besides the cfg file, see above?

tonbor, 2024-09-03:
But #define PIN_RECEIVER_CS   10

    // CC1101: GDO0 / RFM95W/SX127x: G0
    #define PIN_RECEIVER_IRQ  21

    // CC1101: GDO2 / RFM95W/SX127x: G1
    #define PIN_RECEIVER_GPIO 8

    // RFM95W/SX127x - GPIOxx / CC1101 - RADIOLIB_NC
    #define PIN_RECEIVER_RST  9
SX1278 radio = new Module(PIN_RECEIVER_CS, PIN_RECEIVER_IRQ, PIN_RECEIVER_RST, PIN_RECEIVER_GPIO);

// or using RadioShield
// https://github.com/jgromes/RadioShield
//SX1278 radio = RadioShield.ModuleA;

void setup() {
  Serial.begin(9600);

  // initialize SX1278 with default settings
  Serial.print(F("[SX1278] Initializing ... "));
  //int state = radio.beginFSK();
  int state = radio.beginFSK(868.3, 8.21, 57.136417, 250, 10, 32);
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);

fails with [SX1278] Initializing ... failed, code -12

replace with only
int state = radio.beginFSK(); 
�)[SX1278] Initializing ... success!
[SX1278] Waiting for incoming transmission ... timeout!
[SX1278] Waiting for incoming transmission ... timeout!

matthias-bs, 2024-09-03:
Hi @tonbor,

I have a few questions:
- Which board did you select in the Arduino IDE?
- Which radio module are you using?

PIN_RECEIVER_CS is used as GPIO pin, so you can select any available GPIO pin, i.e. you are not restricted to a dedicated CS pin (according to the pinout drawing).

matthias-bs, 2024-09-03:
Here is a draft config assuming you are selecting "ESP32S3 Dev Module" in the Arduino IDE:
https://github.com/matthias-bs/BresserWeatherSensorReceiver/tree/feat-esp32-s3-devkitc-1

Please let me know which radio module you are using (see above).

matthias-bs, 2024-09-03:
The problem with ESP32 S3 or even Espressif ESP32-S3-DevkitC-1 is that this is just one of a hundred quite generic boards. There is nothing specific such as an integrated radio module or an interface for easy integration of a radio module (such as Adafruit Feather or FireBeetle Cover) or an outstanding power management which would make it preferable over any other board.
If anyone chooses such a board, I would expect him to integrate it by himself. Otherwise I would be permanently busy integration some board or other.

tonbor, 2024-09-03:
> Hi @tonbor,
> 
> I have a few questions:
> 
> * Which board did you select in the Arduino IDE? ESP32S3 Dev Module
> * Which radio module are you using? Hoperf RFM95W
> 
> PIN_RECEIVER_CS is used as GPIO pin, so you can select any available GPIO pin, i.e. you are not restricted to a dedicated CS pin (according to the pinout drawing).

tonbor, 2024-09-03:
> https://github.com/matthias-bs/BresserWeatherSensorReceiver/tree/feat-esp32-s3-devkitc-1

great it compiles, no reboot, and I get readings...
Id: [    ABB9] Typ: [1] Ch: [0] St: [0] Bat: [OK ] RSSI: [ -83.5dBm] Temp: [ 18.1C] Hum: [ 94%] Wmax: [ 0.0m/s] Wavg: [ 0.0m/s] Wdir: [312.0deg] Rain: [  692.9mm] UVidx: [0.0] Light: [0.0klx] 
Id: [72181262] Typ: [2] Ch: [4] St: [0] Bat: [OK ] RSSI: [ -99.5dBm] Temp: [ 24.2C] Hum: [ 67%] Wmax: [44.4m/s] Wavg: [44.4m/s] Wdir: [1221.0deg] Rain: [-----.-mm] UVidx: [44.4] Light: [--.-klx] 
Id: [    ABB9] Typ: [1] Ch: [0] St: [0] Bat: [OK ] RSSI: [ -96.5dBm] Temp: [ 18.1C] Hum: [ 94%] Wmax: [ 0.0m/s] Wavg: [ 0.0m/s] Wdir: [354.0deg] Rain: [  692.9mm] UVidx: [0.0] Light: [0.0klx]

matthias-bs, 2024-09-03:
Which radio module *board* (brand, model or link) are you using? HopeRF RFM95W is just the radio chip

tonbor, 2024-09-03:
https://www.tinytronics.nl/nl/communicatie-en-signalen/draadloos/lora/modules/hoperf-rfm95w-lora-module-868mhz

matthias-bs, 2024-09-03:
See https://github.com/matthias-bs/BresserWeatherSensorReceiver/compare/main...feat-esp32-s3-devkitc-1 for comparison (lines 91ff are not related).

I just selected the ESP32 board in the Arduino IDE, compiled with verbose output and copied the define set for this board into the preprocessor command. You already set the correct pin numbers.

matthias-bs, 2024-09-03:
> https://www.tinytronics.nl/nl/communicatie-en-signalen/draadloos/lora/modules/hoperf-rfm95w-lora-module-868mhz

Thanks for the link. This is again a generic module, meaning you may or may not be able to buy or use this part without any changes if you try again half a year later. Therefore I hesitate to add this variant to the source code.

tonbor, 2024-09-03:
Well they are all over Aliexpress https://nl.aliexpress.com/item/1005006440269881.html?spm=a2g0o.productlist.main.13.5d081ca11n0bRV&algo_pvid=9366666f-1dbe-4ac2-ac11-c984bcb5ca0a&algo_exp_id=9366666f-1dbe-4ac2-ac11-c984bcb5ca0a-6&pdp_npi=4%40dis%21EUR%2136.17%2110.49%21%21%21278.23%2180.69%21%40211b61a417253889860347119ea330%2112000037173992052%21sea%21NL%21102390509%21X&curPageLogUid=9mPHzbACEnaK&utparam-url=scene%3Asearch%7Cquery_from%3A

tonbor, 2024-09-03:
[Well you find them at Aliexpress, and ](https://nl.aliexpress.com/item/1005006440269881.html?spm=a2g0o.productlist.main.13.5d081ca11n0bRV&algo_pvid=9366666f-1dbe-4ac2-ac11-c984bcb5ca0a&algo_exp_id=9366666f-1dbe-4ac2-ac11-c984bcb5ca0a-6&pdp_npi=4%40dis%21EUR%2136.17%2110.49%21%21%21278.23%2180.69%21%40211b61a417253889860347119ea330%2112000037173992052%21sea%21NL%21102390509%21X&curPageLogUid=9mPHzbACEnaK&utparam-url=scene%3Asearch%7Cquery_from%3A)

tonbor, 2024-09-03:
[image]

tonbor, 2024-09-03:
Well for Lora use perhabs not ideal, but for scanning Bresser nodes at home ok, and I got a antenne soldered tot ant and GND

matthias-bs, 2024-09-03:
It depends on the range to your sensors. If works for you - fine by me. I'd rather not use it as an example for reasons mentioned above.

tonbor, 2024-09-03:
But adding the use of ESP32 S3 would be desireable because of using this with a cc1101, sx1276 or sx1268. Still I do not understand why I got a rebooting S3 with my changes.

tonbor, 2024-09-04:
Adafruit sells a shielded one https://www.adafruit.com/product/3072

matthias-bs, 2024-09-04:
> Adafruit sells a shielded one https://www.adafruit.com/product/3072

This is one of the boards I'd recommend, but it's not cheap. There others at [https://www.tinytronics.nl](https://www.tinytronics.nl) which are similar and made/sold by some established brands which also provide more or less helpful documentation.

matthias-bs, 2024-09-04:
> But adding the use of ESP32 S3 would be desireable because of using this with a cc1101, sx1276 or sx1268. Still I do not understand why I got a rebooting S3 with my changes.

Okay, I will add it as a generic ESP32 S3 target.

tonbor, 2024-09-04:
> > But adding the use of ESP32 S3 would be desireable because of using this with a cc1101, sx1276 or sx1268. Still I do not understand why I got a rebooting S3 with my changes.
> 
> Okay, I will add it as a generic ESP32 S3 target.

Thanks

matthias-bs, 2024-09-04:
Implemented in https://github.com/matthias-bs/BresserWeatherSensorReceiver/pull/192

tonbor, 2024-09-04:
Perhabs
Define | Radio Module | Configuration
-- | -- | --
ESP32 | user-defined | generic, used for ESP32 boards if none of the above is defined
ESP32 S3 | user-defined | generic, used for ESP32 S3 boards if none of the above is defined
ESP8266 | user-defined | generic, used for ESP8266 boards if none of the above is defined

</markdown-accessiblity-table>