matthias-bs/BresserWeatherSensorReceiver #193: ESP32 Dev Kit C and CC1101 module connection...

sRspectrect68, 2024-09-04:
Hi!
 I have an ESP32 Dev Kit C (https://www.az-delivery.de/en/blogs/azdelivery-blog-fur-arduino-und-raspberry-pi/das-zehnte-tuerchen) from azdelivery , I wanted to connect it to a CC1101 module the same one used for the D1 Mini, which is mentioned.
My current pinout is:
SCK MOSI MISO
ESP32 D18 D23 D19
CSN CC1101 > 27 ESP32 GIPO27
GDO0 CC1101 > 27 ESP32 GIPO21
GDO2 CC1101 > 27 ESP32 GIPO33
and PIN_RECEIVER_RST 32 not connected.
In the sensorCfg file, I selected the CC1101.

So far I have not had success as I do not see anything even with the basic example. My station is a Bresser 7002580.

Can I have some help? Thanks, Riccardo.

matthias-bs, 2024-09-05:
Hi,
which board did you select in the Arduino IDE?
Can you please provide the detailed compiler output?

sRspectrect68, 2024-09-05:
Hi! 
Usually with this card I select ESP32 Dev Module. With this tab the projects work normally.
I also tried with an ESP8266 MOD 12F board (nodemcu V3) but the board continues to restart once loaded.
`In file included from C:\Users\riccardo.sordo\Documents\Arduino\BresserWeatherSensorBasic_1\BresserWeatherSensorBasic_1.ino:61:
C:\Users\riccardo.sordo\Documents\Arduino\BresserWeatherSensorBasic_1\WeatherSensorCfg.h:551:48: note: '#pragma message: Receiver chip: [CC1101]'
  551 | #pragma message("Receiver chip: " RECEIVER_CHIP)
      |                                                ^
C:\Users\riccardo.sordo\Documents\Arduino\BresserWeatherSensorBasic_1\WeatherSensorCfg.h:552:170: note: '#pragma message: Pin config: RST->32, CS->27, GD0/G0/IRQ->21, GDO2/G1/GPIO->33'
  552 | #pragma message("Pin config: RST->" STR(PIN_RECEIVER_RST) ", CS->" STR(PIN_RECEIVER_CS) ", GD0/G0/IRQ->" STR(PIN_RECEIVER_IRQ) ", GDO2/G1/GPIO->" STR(PIN_RECEIVER_GPIO) )
      |                                                                                                                                                                          ^
Sketch uses 361893 bytes (27%) of program storage space. Maximum is 1310720 bytes.
Global variables use 21952 bytes (6%) of dynamic memory, leaving 305728 bytes for local variables. Maximum is 327680 bytes.`

sRspectrect68, 2024-09-05:
I did some checks to understand if there are problems and I only added a few lines to the main code.
I see that the code is executed up to the line: ws.begin();
Then I no longer see anything about serial debugging.
This is true even if I disconnect the CC1101 board

`void setup() {
    Serial.begin(115200);
    Serial.setDebugOutput(true);

    Serial.printf("Starting execution...\n");
    initBoard();
    Serial.printf("Starting initBoard...\n");
    delay(2000);
    ws.begin();
    Serial.printf("Starting begin...\n");
    delay(2000);
}`

matthias-bs, 2024-09-05:
The selected pins seem to be o.k. Did you connect the ESP32's MISO to the CC1101's MOSI and vice versa?
The SW gets stuck in an endless loop if the radio chip cannot be initialized.

matthias-bs, 2024-09-05:
https://github.com/matthias-bs/BresserWeatherSensorReceiver/issues/112

sRspectrect68, 2024-09-05:
> I pin selezionati sembrano ok Hai collegato il MISO dell'ESP32 al MOSI del CC1101 e viceversa? Il SW si blocca in un loop infinito se il chip radio non può essere inizializzato.

Hi!!
oops.. wait... do I have to connect MISO to MOSI and MOSI to MISO??

matthias-bs, 2024-09-05:
Yep!
MOSI: Master Out -> Slave In 
MISO: Master In <- Slave Out
In this case, ESP32 is master and CC1101 slave.

sRspectrect68, 2024-09-06:
Hi!
I tried 100 times even with different ESP32s, nothing to do as soon as you try to initialize the library it freezes. If I install OpenGatewayMQTT I am able to receive RF signals, so I rule out the possibility that the module is faulty or that the ESP32 has problems. There's no way to figure out where the code hangs?

matthias-bs, 2024-09-06:
Hi,
This is strange. Are you sure you have a 868 MHz version of the CC1101 module? Sometimes the ads are ambiguous. The chip is the same for 433 and for 868 MHz, but the RF circuit around it and the antenna are different.
Did you enable debug output in the Arduino IDE? You can also get additional debug output from RadioLib - see RadioLib documentation.

sRspectrect68, 2024-09-07:
HI!!
I'm really starting to suspect that they gave me a 433 receiver, because if I set this frequency with OpenGatewayMQTT, I receive various data from various sensors nearby including a weather sensor. Yet the CC1101's enclosure is rated at 868Mhz. Can the module be forced to work on a different frequency?

sRspectrect68, 2024-09-07:
N: Send on /RFtoMQTT msg {"active":3,"frequency":433.92,"rssithreshold":-82,"rssi":-80,"avgrssi":0,"count":0}
N: Update check, free heap: 114940E: Error on HTTP requestN: Update check done, free heap: 117040N: Send on /SYStoMQTT msg {"uptime":12,"version":"v1.7.0","disc":true,"ohdisc":false,"env":"esp32dev-multi_receiver","freemem":116688,"mqttp":"1883","mqtts":false,"msgprc":0,"msgblck":0,"maxq":0,"minmem":113688,"tempc":59.44,"freestck":5612,"eth":false,"rssi":-84,"SSID":"WiFiCS001","BSSID":"62:EB:B6:AF:A3:70","ip":"192.168.0.212","mac":"64:B7:08:C8:C7:10","modules":["WebUI","RF","RF2","RTL_433"]}
N: Send on /WebUItoMQTT msg {"displayMetric":true,"webUISecure":true,"displayQueue":0}
N: Send on /RTL_433toMQTT/Generic-Remote/41946 msg {"model":"Smoke-GS558","unit":41946,"t":118,"arn":"6e5bc5","ol":"ss Smoke and Heat Detector GS 558"," 558":-42,"rssi":1110996}
N: Send on /RTL_433toMQTT/Smoke-GS558/29406 msg {"model":"Smoke-GS558","id":29406,"unit":5,"learn":0,"code":"6e5bc5","protocol":"Wireless Smoke and Heat Detector GS 558","rssi":-42,"duration":1110996}
N: Send on /RTL_433toMQTT/Generic-Remote/41946 msg {"model":"Smoke-GS558","unit":41946,"t":118,"arn":"6e5bc5","ol":"ss Smoke and Heat Detector GS 558"," 558":-68,"rssi":1029997}
N: Send on /RTL_433toMQTT/Smoke-GS558/29406 msg {"model":"Smoke-GS558","id":29406,"unit":5,"learn":0,"code":"6e5bc5","protocol":"Wireless Smoke and Heat Detector GS 558","rssi":-68,"duration":1029997}

matthias-bs, 2024-09-07:
Yes, that seems to be the case. If the RF circuit is tuned to 433 MHz, you can't change it to 868 MHz without swapping parts. The antenna is another matter.

sRspectrect68, 2024-09-07:
I'm returning it and having it exchanged for one that is actually at 868. Thanks for the support.

sRspectrect68, 2024-09-10:
Hi, in the end I had a Heltec Wireless Stick V3 sent to me, to avoid possible problems with the RF module. I selected the basic Test Sensor chetch, I set the relevant board in the Arduino IDE, and in the configuration I uncommented the line "// This define is set by selecting "Board: Heltec Wireless Stick(V3)" (SX1262) in the Arduino IDE:
#define ARDUINO_HELTEC_WIRELESS_STICK_V3". The compiler gives me an error: exit status 1

Compilation error: exit status 1 analyzing output I see: c:\Users\Programmazione\Documents\Arduino\libraries\BresserWeatherSensorReceiver\src\WeatherSensorCfg.h:421:31: error: 'DIO1' was not declared in this scope; did you mean 'GOD0'?
  421 |     #define PIN_RECEIVER_GPIO DIO1

matthias-bs, 2024-09-10:
Hi,
you don't have to uncomment the line with the board definition, because this definition is set automatically by selecting the board in the Arduino IDE.
I can reproduce the compilation error - something has been changed in the board's pin definitions.

sRspectrect68, 2024-09-10:
hi, ok, maybe I read the instructions wrong. In the cfg file, I don't have to change anything except the possible exclusion of the sensors. For the rest, do you mean that in your opinion they changed something in the esp32 module?

matthias-bs, 2024-09-10:
Right! Well, the change was made months ago, but I missed to integrate it into my code.

Here is the fix:
https://github.com/matthias-bs/BresserWeatherSensorReceiver/pull/194/files

You can change the cfg header file as shown above or pull the changes from GitHub.

sRspectrect68, 2024-09-10:
> Right! Well, the change was made months ago, but I missed to integrate it into my code.
> 
> Here is the fix: https://github.com/matthias-bs/BresserWeatherSensorReceiver/pull/194/files
> 
> You can change the cfg header file as shown above or pull the changes from GitHub.
> 
> Please create a new issue if required, because the topic has changed now.

Hi!!
Now it works, I receive data from the sensor, thank you very much for your assistance.
Riccardo.