matthias-bs/BresserWeatherSensorReceiver #195: Added MAKERGO_C3_SUPERMINI and Heltec sx1262 shielded

tonbor, 2024-10-28:
**Is your feature request related to a problem? Please describe.**

A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

No support voor de ESP32 C3 (mini) and Heltec sx1262

#elif defined(ARDUINO_MAKERGO_C3_SUPERMINI)
//ESP32 C3 SUPERMINI

#pragma message("ARDUINO_MakerGO_ESP32_C3_SuperMini defined; this is a generic (i.e. non-specific) target")

#define USE_SX1262 //use Heltec sx1262 shielded
    
    #pragma message("Cross check if the selected GPIO pins are really available on your board.")
    #pragma message("Connect a radio module with a supported chip.")
    #pragma message("Select the chip by setting the appropriate define.")

#elif defined(ARDUINO_MAKERGO_C3_SUPERMINI)
        #define PIN_RECEIVER_CS   7
        // CC1101: GDO0 / RFM95W/SX127x: G0 / SX1262 DI01
    #define PIN_RECEIVER_IRQ  1

    // CC1101: GDO2 / RFM95W/SX127x: G1 / SX1262: BUSY
    #define PIN_RECEIVER_GPIO 2

    // RFM95W/SX127x - GPIOxx / CC1101 - RADIOLIB_NC / SX1262: RESET
    #define PIN_RECEIVER_RST  3

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
please add

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
none

**Additional context**
Add any other context or screenshots about the feature request here.

tonbor, 2024-10-28:
[image]
[image]

tonbor, 2024-10-28:
[image]
[image]

matthias-bs, 2024-10-29:
Hi @tonbor,

Thank you for the contribution, I will add it.
I'm just curious: Are there any specific reasons/features why you selected the MAKERGO_C3_SUPERMINI?

Regards,
Matthias

tonbor, 2024-10-29:
Small to use as a receiver for the sx1262 just enough pins, can put it in
foot of a case for the ESP32-8038s43. The ESP32-8038s43 has not enough pins
free for connecting the sx1262, besides there is a problem to use the spi
for two devices.

matthias-bs, 2024-10-29:
It seems the Heltec HT-RA62 is quite a bargain, good catch! But I personally don't like the 1,27 mm pitch.

closed 2024-10-30 by matthias-bs

tonbor, 2024-12-24:
> It seems the Heltec HT-RA62 is quite a bargain, good catch! But I personally don't like the 1,27 mm pitch.

https://github.com/user-attachments/assets/7020a30c-8bcc-46b4-9b2c-d127b7eaf2f4

fits fine to this board