matthias-bs/BresserWeatherSensorReceiver #196: Send data to mysql

asanbus, 2024-10-30:
Would you mind add "Send to mysql"

Thank you

closed 2024-10-30 by matthias-bs

asanbus, 2024-11-07:
Would you mind create ino? 
My system containt:
-TTGO LORA32 OLED 868
-Bresser 7-1
-BME280
-Mysql

asanbus, 2024-11-07:
Data delay one minute

asanbus, 2024-11-07:
It is imposible for me, Error and Error

Thank you

asanbus, 2024-11-07:
[image]

matthias-bs, 2024-11-08:
I suggest to make yourself familiar with some common rules regarding getting support on GitHub. For instance, there are some forms for creating an issue in this project which indicate which kind of information is required to understand a problem.

You cannot expect that a volunteer reads your mind, solves your issues or develops an entirely customized software for you.

asanbus, 2024-11-11:
Price?