matthias-bs/BresserWeatherSensorReceiver #199: DOMOTICZ - MQTT - CONFIGURATION

tracteur12, 2024-11-18:
Hello, 

Thank for your job.

My Configuration. ESP8266MOD+C1101. I use Arduino 2.3.3 to compil sketch.
I use Domoticz on my raspberry with Last Version. I have to install Mosquitto and nodejs.

**moquitto.conf:**
# A full description of the configuration file is at
# /usr/share/doc/mosquitto/examples/mosquitto.conf.example
pid_file /run/mosquitto/mosquitto.pid
persistence true
persistence_location /var/lib/mosquitto/
log_dest file /var/log/mosquitto/mosquitto.log
include_dir /etc/mosquitto/conf.d
#allow_anonymous false
#password_file /etc/mosquitto/pwfile
listener 8883
allow_anonymous true

**Configuration scketch "SensorDomoticz"**
#define HOSTNAME "BresserDomo"
#define APPEND_CHIP_ID
const int MQTT_PORT = 8883; // typically 8883 with TLS / 1883 without TLS
const char MQTT_HOST[] = "192.168.1.103";
const char MQTT_USER[] = ""; // leave blank if no credentials used
const char MQTT_PASS[] = ""; // leave blank if no credentials used

On Domoticz, I have to create Idx with virtual sensor and MQTT pluging
[image]
[image]
[image]

The "Basic" scketch work.
Log on Arduino

[image]

Actually, I try to use you scketch "SensorDomoticz" but it don't work
Log on Arduino. Some information try to send it.

[image]
[image]

Log on Domoticz, I have a error "Code 14".

[image]

I Try to find somme information (MQTT & Code 1') on internet, but I didn't answer.

Have you got some information track?

Thank you in advance.

Romain BERANRD

matthias-bs, 2024-11-19:
Hi,

It seems the connection between the ESP and the MQTT broker (Mosquitto) is working fine. Please have a look into the Mosquitto log file or confirmation.

As shown in your Domoticz log file, the connection to the MQTT broker cannot be established.
Possible reasons:

- Wrong IP address
- Access to port 8883 not possible (Firewall?)
- Wrong TLS version
- Incorrect TLS setup

You could try a non-secure connection between Domoticz and the broker first (port 1883).

Please refer to the Domoticz or Mosquitto documentation.

Best regards,
Matthias

tracteur12, 2024-11-21:
Hello, 
Thank for you feedback. I m testing step by step.

**Wrong IP adress**: Can confirm configuration: Arduino scketch : IP Raspberry with domoticz. Pluging MQTT domoticz: IP ESP8266.

**Firewall**
How I can test it?
I scan port of my ESP but I didn't find "open" port 8883 and 1883.
[image]
[image]

**Mosquitto log file**
[image]

matthias-bs, 2024-11-22:
Sorry, I don't have a clear picture of your configuration yet.
If Domoticz and Mosquitto are running on the same device (Raspberry Pi as far as I understood), you can use an non-secure connection between the two.

You add
```
listen 1883
protocol mqtt
```
to your Mosquitto configuration.

Then you set the MQTT broker to `localhost` and the port to `1883` in Domoticz.

In the Mosquitto logfile, you should see that the broker listens on port 1883 and that there is a connection attempt from some MQTT client (Domoticz).

tracteur12, 2024-11-22:
OK, I change my mosquitto config, I change IP in Domoticz, I reboot raspberry.

When we check mosquitto log and Domotizc log, I think is ok. But I didn't have any information of my Bresser weather station.

[image]
[image]

matthias-bs, 2024-11-22:
I don't know much about Domoticz, but I saw that you changed the IDX in the sketch. I think each IDX refers to a certain message format. Can you try to change the IDX back to the original and select the sensor/widget setting in Domoticz accordingly?

BTW: You could use different ports for connecting the ESP and Domoticz to Mosquitto.

tracteur12, 2024-11-27:
Hello, It's working. I have to recharge programme, update configuration mqtt on domoticz and recreat sensor on domoticz.
It was strange, I have to disable "ESP.deep" (ligne 640) because it didn't wake up. 

Regards,

Romain

matthias-bs, 2024-11-27:
That's good news!