matthias-bs/BresserWeatherSensorReceiver #200: Block not valid messages

tonbor, 2024-11-27:
**Context**
- Which example sketch are you using? MQTT
- Which changes to the code did you make (if any)?
- Which sensor are you using? sx1276

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Step1 recompile working MQTT ino 
2. Step2 now I get mqtt message that did not conform Bresser messages
3 I have channel 0 to 5 (4 latest)
4. See error
status = online
abb9
extra = {"wind_dir_txt":"S","wind_gust_bft":2,"wind_avg_bft":2,"dewpoint_c":9.7,"perceived_temp_c":10.6}
radio = {"rssi":-100.5}
72181262
data = {"id":1914180194,"ch":4,"battery_ok":1,"temp_c":15.5,"humidity":63,"uv":44.4}
rssi v3 = -113.0
18839309
data = {"id":411276041,"ch":2,"battery_ok":1,"temp_c":8.90,"moisture":99}
rssi v3 = -110.0
3fca6908
data = {"id":1070229768,"ch":105,"battery_ok":1}
rssi v3 = nan
3c104ff4
data = {"id":1007702004,"ch":78,"battery_ok":1}
rssi v3 = 0.0
abba1234
data = {"id":2881098292,"ch":5,"battery_ok":1}
rssi v3 = 0.0
3c104e40
data = {"id":1007701568,"ch":78,"battery_ok":1}
rssi v3 = 0.0
3c104e20
data = {"id":1007701536,"ch":85,"battery_ok":1}
rssi v3 = 0.0
3c1055fc
data = {"id":1007703548,"ch":78,"battery_ok":1}
rssi v3 = 0.0
80339449
data = {"id":2150863945,"ch":3,"battery_ok":1,"temp_c":8.80,"moisture":87}
rssi v3 = -104.5
3c10590c
data = {"id":1007704332,"ch":85,"battery_ok":1}
rssi v3 = 0.0
**Expected behavior**
A clear and concise description of what you expected to happen.
Block none Bresser messages
**Debug Log**
If applicable, add log file to help explain your problem.
 - Build log file (enable verbose output in the Arduino IDE: File->Preferences - "Show verbose output during: compile")
 - Serial console log file (see [Debug Output Configuration in Arduino IDE](https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/DEBUG_OUTPUT.md))

**Embedded Device (please complete the following information):**
 - Arduino IDE Version: EDI 2.3.3
 - ESP32 dev S3
 - Library Version [e.g. 0.6.1]
 - Your Board: [brand/model/version] (if unsure, attach a photo)
 - Selected Board in the Arduino IDE: [brand/model/version]
 - Your Radio Transceiver Module (if external): [brand/model/chipset/frequency] sx1276 868Mhz
 - **Pinning/Wiring**

**Additional context**
Add any other context about the problem here.

matthias-bs, 2024-12-02:
Can't reproduce. Not enough information provided for analysis.

closed 2024-12-02 by matthias-bs

tonbor, 2024-12-02:
> Can't reproduce. Not enough information provided for analysis.

Replaced the S3 Dev board and no more false readings. Seems a problem with the USB com port, no stable current. Sorry for the trouble, took me 3 days to find out :-(