matthias-bs/BresserWeatherSensorReceiver #202: Invalid rain statistics values

matthias-bs, 2024-11-27:
Still a problem, erased memory with Erase All Flash Before Sketch Upload: Enabled, no more ssi in memory, still false rain info ->
data = {"id":43961,"ch":0,"battery_ok":1,"temp_c":10.8,"humidity":96,"wind_gust":3.7,"wind_avg":3.6,"wind_dir":205.0,"uv":0.0,"light_klx":0.3,"rain":994.6,"rain_h":0.0,"rain_d":22940.6,"rain_w":22940.6,"rain_m":22940.6}

_Originally posted by @tonbor in https://github.com/matthias-bs/BresserWeatherSensorReceiver/issues/190#issuecomment-2504139641_

matthias-bs, 2024-11-27:
@tonbor I need your debug log file to analyze this.

tonbor, 2024-11-27:
18:54:39.536 -> [115316][D][RainGauge.cpp:194] prefs_load(): lastUpdate        =1732730077
18:54:39.568 -> [115323][D][RainGauge.cpp:195] prefs_load(): startupPrev       =0
18:54:39.568 -> [115329][D][RainGauge.cpp:196] prefs_load(): rainPreStartup    =995.100037
18:54:39.568 -> [115336][D][RainGauge.cpp:197] prefs_load(): tsDayBegin        =3
18:54:39.568 -> [115341][D][RainGauge.cpp:198] prefs_load(): rainDayBegin      =994.900024
18:54:39.568 -> [115348][D][RainGauge.cpp:199] prefs_load(): tsWeekBegin       =3
18:54:39.568 -> [115354][D][RainGauge.cpp:200] prefs_load(): rainWeekBegin     =994.900024
18:54:39.605 -> [115361][D][RainGauge.cpp:201] prefs_load(): wdayPrev          =3
18:54:39.605 -> [115367][D][RainGauge.cpp:202] prefs_load(): tsMonthBegin      =10
18:54:39.605 -> [115373][D][RainGauge.cpp:203] prefs_load(): rainMonthBegin    =994.900024
18:54:39.605 -> [115379][D][RainGauge.cpp:204] prefs_load(): rainPrev          =2346427760205696598016.000000
18:54:39.637 -> [115388][D][RainGauge.cpp:205] prefs_load(): rainAcc           =2346427760205696598016.000000
18:54:39.637 -> [115397][D][RainGauge.cpp:280] update(): rainDelta: 0.0
18:54:39.637 -> [115402][D][RainGauge.cpp:325] update(): t_delta: 1007513408
18:54:39.637 -> [115408][D][RainGauge.cpp:345] update(): hist[9]=0 (upd)
18:54:39.637 -> [115413][D][RainGauge.cpp:383] update(): hist[]={158, -1, 0, 0, -1, 0, 0, 0, 0, 0, }

tonbor, 2024-11-27:
19:18:59.586 -> [426578][D][RainGauge.cpp:194] prefs_load(): lastUpdate        =1732731517
19:18:59.586 -> [426585][D][RainGauge.cpp:195] prefs_load(): startupPrev       =1
19:18:59.586 -> [426591][D][RainGauge.cpp:196] prefs_load(): rainPreStartup    =17845015893711164469673984.000000
19:18:59.586 -> [426600][D][RainGauge.cpp:197] prefs_load(): tsDayBegin        =3
19:18:59.586 -> [426606][D][RainGauge.cpp:198] prefs_load(): rainDayBegin      =994.900024
19:18:59.586 -> [426613][D][RainGauge.cpp:199] prefs_load(): tsWeekBegin       =3
19:18:59.620 -> [426619][D][RainGauge.cpp:200] prefs_load(): rainWeekBegin     =994.900024
19:18:59.620 -> [426625][D][RainGauge.cpp:201] prefs_load(): wdayPrev          =3
19:18:59.620 -> [426631][D][RainGauge.cpp:202] prefs_load(): tsMonthBegin      =10
19:18:59.620 -> [426637][D][RainGauge.cpp:203] prefs_load(): rainMonthBegin    =994.900024
19:18:59.620 -> [426644][D][RainGauge.cpp:204] prefs_load(): rainPrev          =17847362088973039403270144.000000
19:18:59.661 -> [426653][D][RainGauge.cpp:205] prefs_load(): rainAcc           =2346427760205696598016.000000
19:18:59.661 -> [426661][D][RainGauge.cpp:280] update(): rainDelta: -17845015893711164469673984.0
19:18:59.661 -> [426669][D][RainGauge.cpp:325] update(): t_delta: 1007513408
19:18:59.661 -> [426675][D][RainGauge.cpp:345] update(): hist[3]=0 (upd)
19:18:59.661 -> [426680][D][RainGauge.cpp:383] update(): hist[]={0, 0, -1, 0, -1, 0, 0, 0, 0, 0, }

matthias-bs, 2024-11-27:
Which `SLEEP_INTERVAL` did you set?

tonbor, 2024-11-27:
non, comment out sleep

matthias-bs, 2024-11-27:
What is the interval between messages containing valid rain values then? The short log snippets you provided do not show this.

tonbor, 2024-11-27:
Will look in my source tomorrow. Did buy Heltec T190 with sx1262, will adjust source and adjust config file for this board. See if get the same outcome for rain and not blocking messages from other source then my weatherstation and Bresser sensors.

matthias-bs, 2024-11-28:
Another - most likely - cause for the wrong rain values could be the reception of another weather sensor's rain value (or some radio signal which is misinterpreted as that).
See https://github.com/matthias-bs/BresserWeatherSensorReceiver/issues/200.

To check this, please have a look at the message decoding in the debug log.

tonbor, 2024-11-28:
debug:10:30:12.544 -> [166314][I][BresserWeatherSensorMQTT_S3_V03_multi_mem.ino:848] publishRadio(): ESPWeather2-33737C/radio: {"rssi":-86}
10:30:12.544 -> 
10:30:12.544 -> weatherSensor.clearSlots()
10:30:12.544 -> Get weather Data
10:30:15.623 -> [169404][D][WeatherSensor.cpp:351] getMessage(): [SX1276] R [D4] RSSI: -91.5
10:30:15.656 -> [169411][D][WeatherSensor.h:598] log_message():           Byte #: 00 01 02 03 04 05 06 07 08 09 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 
10:30:15.656 -> [169424][D][WeatherSensor.h:608] log_message(): De-whitened Data: 7E 2F AB B9 23 90 B2 00 00 00 00 99 72 04 08 10 80 00 80 25 00 60 00 00 00 AA 
10:30:15.656 -> [169437][D][WeatherSensorDecoders.cpp:105] findSlot(): sensor[0]: v=0 id=0x72181262 t=2 c=0
10:30:15.690 -> decode_ok for get weather Data
10:30:15.690 -> weatherSensor.clearSlots()

matthias-bs, 2024-11-30:
We'd need more debug output like 
```
data = {"id":43961,"ch":0,"battery_ok":1,"temp_c":7.7,"humidity":81,"wind_gust":0.0,"wind_avg":0.0,"wind_dir":318.0,"uv":0.6,"light_klx":9.6,"rain":997.2,"rain_h":0.0,"rain_d":1.6,"rain_w":18.1,"rain_m":101.6}
```
and the parameters passed to `rainGauge.update()` to see what is going on.

tonbor, 2024-11-30:
A brand new Heltec T190 and this is the rain start 
abb9
data = {"id":43961,"ch":0,"battery_ok":1,"temp_c":6.1,"humidity":80,"wind_gust":0.0,"wind_avg":0.0,"wind_dir":168.0,"uv":0.0,"light_klx":3.9,"rain":997.2,"rain_h":0.0,"rain_d":0.0,"rain_w":987.3,"rain_m":987.3}
rssi = -104.5
debug see #203

matthias-bs, 2024-12-03:
@tonbor I noticed from a GitHub mail that you reported that the issue was caused by an unstable power supply, but I cannot find your comment in any of the issues you created. Can this issue be closed then?

tonbor, 2024-12-03:
Well the problem is starting the first time reading the sensor when the sensor has a value from the past and the module has none. Then hour, day week and month has the wrong values.

matthias-bs, 2024-12-04:
The software should handle this correctly. At a fresh start (reset command received or Flash erased), all entries are as in valid and the current raingauge value and the timestamps are stored. From this point on, the values are calculated.
The only thing which is not quite correct is the handling of `pasthour()`:
https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/617a1f159845c2654a8f04854eb13ef4d4debb58/src/RainGauge.h#L236
You should pass a pointer to a `bool valid;` variable to it and use the function's return value only if valid is true.

If there are values from a previous run stored in the flash, the algorithm will assume that it should proceed from there, i.e. use the stored timestamps and raingauge values as if recorded previously (from the same sensor!). This includes some tricks to handle potential resets (battery change) and counter overflow of the raingauge.

See https://github.com/matthias-bs/BresserWeatherSensorReceiver/wiki/04.-Implementing-Rain-Gauge-Statistics

Since I'm using some unit tests for this part of the software, nobody else reported any problems and I use the library in my own projects, I assume that it works.

I you still think there is a SW issue, please provide a complete log file spanning over a few transmission cycles and point me in the right direction... Otherwise I'm not inclined to spend my time on analyzing small bits of the big picture...

matthias-bs, 2025-01-25:
Closed due to inactivity.

closed 2025-01-25 by matthias-bs