matthias-bs/BresserWeatherSensorReceiver #203: Get Heltec Vision Master working

tonbor, 2024-11-29:
**Is your feature request related to a problem? Please describe.**
Try with a fresh start with a fresh Module the Heltec Vison Master T190 868mhz

**Describe the solution you'd like**
A working solution

**Describe alternatives you've considered**
I added the module at two places #elif defined(ARDUINO_HELTEC_WIFI_LORA_32_V3) || defined(ARDUINO_heltec_wifi_lora_32_V3) || defined(ARDUINO_HELTEC_VISION_MASTER_T190)
Has the same pinout

**Additional context**
Is not working :-(

14:54:09.629 -> done!
14:54:09.629 -> [  4305][I][BresserWeatherSensorMQTT_T190.ino:439] mqtt_setup(): Current time (GMT): Fri Nov 29 13:54:10 2024
14:54:09.629 -> 
14:54:09.629 -> Checking wifi...
14:54:09.629 -> MQTT connecting... [  4388][I][BresserWeatherSensorMQTT_T190.ino:493] mqtt_connect(): 
14:54:09.672 -> connected!
14:54:09.704 -> [  4412][I][BresserWeatherSensorMQTT_T190.ino:495] mqtt_connect(): ESPWeather-3C5268/status: online
14:54:09.704 -> 
14:54:09.704 -> [  4415][D][WeatherSensor.cpp:134] begin(): max_sensors: 6
14:54:09.704 -> [  4416][D][WeatherSensor.cpp:135] begin(): rx_flags: 1
14:54:09.736 -> [  4420][D][WeatherSensor.cpp:136] begin(): en_decoders: 255
14:54:09.736 -> [  4426][D][WeatherSensor.cpp:584] initList(): Key exc in preferences? 0
14:54:09.736 -> [  4432][D][WeatherSensor.cpp:606] initList(): Using sensor_ids_exc list from WeatherSensorCfg.h
14:54:09.736 -> [  4441][D][WeatherSensor.cpp:617] initList(): 792882A2
14:54:09.736 -> [  4446][D][WeatherSensor.cpp:584] initList(): Key inc in preferences? 0
14:54:09.782 -> [  4452][D][WeatherSensor.cpp:606] initList(): Using sensor_ids_inc list from WeatherSensorCfg.h
14:54:09.782 -> [  4460][D][WeatherSensor.cpp:613] initList(): (empty)
14:54:09.782 -> [  4465][D][WeatherSensor.cpp:149] begin(): [SX1262] Initializing ... 
14:54:09.817 -> [  4522][D][WeatherSensor.cpp:165] begin(): success!
14:54:09.817 -> [  4523][D][WeatherSensor.cpp:210] begin(): [SX1262] Setup complete - awaiting incoming messages...
14:54:09.817 -> [  4531][D][WeatherSensor.cpp:701] setSensorsCfg(): max_sensors: 6
14:54:09.852 -> [  4532][D][WeatherSensor.cpp:702] setSensorsCfg(): rx_flags: 1
14:54:09.852 -> [  4536][D][WeatherSensor.cpp:703] setSensorsCfg(): enabled_decoders: 255
14:54:09.852 -> =========== After Setup Start ============
14:54:09.852 -> INTERNAL Memory Info:
14:54:09.852 -> ------------------------------------------
14:54:09.852 ->   Total Size        :   362704 B ( 354.2 KB)
14:54:09.852 ->   Free Bytes        :   278668 B ( 272.1 KB)
14:54:09.852 ->   Allocated Bytes   :    75660 B (  73.9 KB)
14:54:09.852 ->   Minimum Free Bytes:   273704 B ( 267.3 KB)
14:54:09.884 ->   Largest Free Block:   262132 B ( 256.0 KB)
14:54:09.884 -> ------------------------------------------
14:54:09.884 -> GPIO Info:
14:54:09.884 -> ------------------------------------------
14:54:09.884 ->   GPIO : BUS_TYPE[bus/unit][chan]
14:54:09.884 ->   --------------------------------------  
14:54:09.884 ->      2 : GPIO
14:54:09.884 ->      8 : GPIO
14:54:09.884 ->      9 : SPI_MASTER_SCK[0]
14:54:09.884 ->     10 : SPI_MASTER_MOSI[0]
14:54:09.884 ->     11 : SPI_MASTER_MISO[0]
14:54:09.884 ->     12 : GPIO
14:54:09.884 ->     13 : GPIO
14:54:09.920 ->     14 : GPIO
14:54:09.920 ->     19 : USB_DM
14:54:09.921 ->     20 : USB_DP
14:54:09.921 ->     43 : UART_TX[0]
14:54:09.921 ->     44 : UART_RX[0]
14:54:09.921 -> ============ After Setup End =============
14:55:39.922 -> [ 94632][I][BresserWeatherSensorMQTT_T190.ino:827] loop(): ESPWeather-3C5268/status: online
14:55:39.922 -> 
14:55:39.922 -> [ 94635][I][BresserWeatherSensorMQTT_T190.ino:740] publishRadio(): ESPWeather-3C5268/radio: {"rssi":-98.5}
14:55:39.922 -> 
14:57:09.956 -> [184640][I][BresserWeatherSensorMQTT_T190.ino:827] loop(): ESPWeather-3C5268/status: online
14:57:09.956 -> 
14:57:09.957 -> [184643][I][BresserWeatherSensorMQTT_T190.ino:740] publishRadio(): ESPWeather-3C5268/radio: {"rssi":-96}
14:57:09.957 -> 
14:58:12.807 -> [247512][D][WeatherSensor.cpp:339] getMessage(): [SX1262] R [D4] RSSI: -98.5
14:58:12.807 -> [247513][D][WeatherSensor.cpp:1299] decodeBresser7In1Payload(): Digest check failed - [BDE6] vs [D97F] (6499)
14:58:12.839 -> [247518][D][WeatherSensor.cpp:1026] decodeBresser6In1Payload(): Digest check failed - [174C] != [D59]
14:58:12.839 -> [247527][D][WeatherSensor.cpp:823] decodeBresser5In1Payload(): Parity wrong at column 0
14:58:12.839 -> [247534][D][WeatherSensor.cpp:1455] decodeBresserLightningPayload(): Digest check failed - [BDE6] vs [4F8F] (F269)
14:58:12.839 -> [247544][D][WeatherSensor.cpp:1561] decodeBresserLeakagePayload(): CRC16 check failed - [2868] vs [174C]
14:58:39.937 -> [274648][I][BresserWeatherSensorMQTT_T190.ino:827] loop(): ESPWeather-3C5268/status: online
14:58:39.937 -> 
14:58:39.937 -> [274651][I][BresserWeatherSensorMQTT_T190.ino:740] publishRadio(): ESPWeather-3C5268/radio: {"rssi":-90}
14:58:39.937 -> 
14:58:39.937 -> 
15:00:09.982 -> [364656][I][BresserWeatherSensorMQTT_T190.ino:827] loop(): ESPWeather-3C5268/status: online
15:00:09.982 -> 
15:00:09.982 -> [364659][I][BresserWeatherSensorMQTT_T190.ino:740] publishRadio(): ESPWeather-3C5268/radio: {"rssi":-99}
15:00:09.982 -> 
15:00:11.063 -> [365758][D][WeatherSensor.cpp:339] getMessage(): [SX1262] R [D4] RSSI: -100.5
15:00:11.063 -> [365759][D][WeatherSensor.cpp:1299] decodeBresser7In1Payload(): Digest check failed - [8B41] vs [22F7] (A9B6)
15:00:11.063 -> [365764][D][WeatherSensor.cpp:1026] decodeBresser6In1Payload(): Digest check failed - [21EB] != [2003]
15:00:11.063 -> [365773][D][WeatherSensor.cpp:823] decodeBresser5In1Payload(): Parity wrong at column 0
15:00:11.108 -> [365780][D][WeatherSensor.cpp:1455] decodeBresserLightningPayload(): Digest check failed - [8B41] vs [C7ED] (4CAC)
15:00:11.108 -> [365790][D][WeatherSensor.cpp:1561] decodeBresserLeakagePayload(): CRC16 check failed - [5656] vs [21EB]
15:00:16.353 -> [371057][D][WeatherSensor.cpp:339] getMessage(): [SX1262] R [D4] RSSI: -99.5
15:00:16.353 -> [371057][D][WeatherSensor.cpp:1299] decodeBresser7In1Payload(): Digest check failed - [D5BB] vs [DF4F] (0AF4)
15:00:16.353 -> [371062][D][WeatherSensor.cpp:1026] decodeBresser6In1Payload(): Digest check failed - [7F11] != [9E02]
15:00:16.447 -> [371071][D][WeatherSensor.cpp:823] decodeBresser5In1Payload(): Parity wrong at column 0
15:00:16.447 -> [371079][D][WeatherSensor.cpp:1455] decodeBresserLightningPayload(): Digest check failed - [D5BB] vs [561A] (83A1)
15:00:16.447 -> [371089][D][WeatherSensor.cpp:1561] decodeBresserLeakagePayload(): CRC16 check failed - [5ADD] vs [7F11]
Ln 839, Col 32
Heltec Vision Master T190
aan COM27
2

matthias-bs, 2024-11-30:
I have checked the board specific define in the compiler output, the radio chip type according to Heltec's description and the pinning according to the schematic and the board's `pins_arduino.h`. If you have changed the code as you described, it should work.

The connection between ESP32 and HT-RA62 should be o.k, otherwise execution wouldn't pass the initialization phase. Furthermore, some signals with valid preamble and reasonable RSSI values are received. Maybe a signal quality issue after all? Which kind of antenna are you using?

Talking about antennas, according to Heltec's schematic, the radio chip's antenna pin is not connected at all (neither are some other pins...). On the other hand, I wonder if I've ever seen a flawless schematic by Heltec...

tonbor, 2024-11-30:
Well as I had problems with recompiling all my working scripts to get them working again I think there must be something wrong at my side or with the libraries. Because compiling for the sx1276 gives strange behaviors and compiling for the sx1262 I get no responce at all from getting info from the 5 sensors. So I went back to arduino 2.3.2 IDE, back to your libraries from june 2024 and Radio.lib. And yes all compiling and all working and gives good info. So where is the problem?

tonbor, 2024-11-30:
So the Heltec Vison Master T190 is working, but not with the (my) latest Arduino suite and libraries. My best guess is with Radioalib......
14:17:02.886 -> =========== Before Setup Start ===========
14:17:02.886 -> ..[  2936][I][BresserWeatherSensorMQTT.ino:408] mqtt_setup(): connected!
14:17:03.775 -> [  2936][I][BresserWeatherSensorMQTT.ino:412] mqtt_setup(): Setting time using SNTP
14:17:04.249 -> ......[  5939][I][BresserWeatherSensorMQTT.ino:427] mqtt_setup(): 
14:17:06.790 -> done!
14:17:06.790 -> [  5939][I][BresserWeatherSensorMQTT.ino:432] mqtt_setup(): Current time (GMT): Sat Nov 30 13:17:07 2024
14:17:06.790 -> 
14:17:06.790 -> Checking wifi...
14:17:06.790 -> MQTT connecting... [  5961][I][BresserWeatherSensorMQTT.ino:486] mqtt_connect(): 
14:17:06.790 -> connected!
14:17:06.790 -> [  5979][I][BresserWeatherSensorMQTT.ino:488] mqtt_connect(): ESPWeather-3C5268/status: online
14:17:06.790 -> 
14:17:06.822 -> [  5982][D][WeatherSensor.cpp:581] initList(): Key exc in preferences? 0
14:17:06.822 -> [  5983][D][WeatherSensor.cpp:602] initList(): Using sensor_ids_exc list from WeatherSensorCfg.h
14:17:06.822 -> [  5991][D][WeatherSensor.cpp:613] initList(): 792882A2
14:17:06.822 -> [  5996][D][WeatherSensor.cpp:581] initList(): Key inc in preferences? 0
14:17:06.822 -> [  6003][D][WeatherSensor.cpp:602] initList(): Using sensor_ids_inc list from WeatherSensorCfg.h
14:17:06.854 -> [  6011][D][WeatherSensor.cpp:609] initList(): (empty)
14:17:06.854 -> [  6016][D][WeatherSensor.cpp:157] begin(): [SX1262] Initializing ... 
14:17:06.891 -> [  6075][D][WeatherSensor.cpp:173] begin(): success!
14:17:06.891 -> [  6076][D][WeatherSensor.cpp:218] begin(): [SX1262] Setup complete - awaiting incoming messages...
14:17:06.924 -> =========== After Setup Start ============
14:17:06.924 -> INTERNAL Memory Info:
14:17:06.924 -> ------------------------------------------
14:17:06.924 ->   Total Size        :   362984 B ( 354.5 KB)
14:17:06.924 ->   Free Bytes        :   279244 B ( 272.7 KB)
14:17:06.924 ->   Allocated Bytes   :    75332 B (  73.6 KB)
14:17:06.924 ->   Minimum Free Bytes:   271164 B ( 264.8 KB)
14:17:06.924 ->   Largest Free Block:   262132 B ( 256.0 KB)
14:17:06.924 -> ------------------------------------------
14:17:06.924 -> GPIO Info:
14:17:06.958 -> ------------------------------------------
14:17:06.958 ->   GPIO : BUS_TYPE[bus/unit][chan]
14:17:06.958 ->   --------------------------------------  
4:17:06.958 -> ------------------------------------------
14:17:06.958 ->   GPIO : BUS_TYPE[bus/unit][chan]
14:17:06.958 ->   --------------------------------------  
14:17:06.958 ->      2 : GPIO
14:17:06.958 ->      8 : GPIO
14:17:06.958 ->      9 : SPI_MASTER_SCK[0]
14:17:06.958 ->     10 : SPI_MASTER_MOSI[0]
14:17:06.958 ->     11 : SPI_MASTER_MISO[0]
14:17:06.958 ->     12 : GPIO
14:17:06.958 ->     13 : GPIO
14:17:06.958 ->     14 : GPIO
14:17:06.958 ->     19 : USB_DM
14:17:06.958 ->     20 : USB_DP
14:17:06.958 ->     43 : UART_TX[0]
14:17:06.958 ->     44 : UART_RX[0]
14:17:06.958 -> ============ After Setup End =============
14:17:10.632 -> [  9813][D][WeatherSensor.cpp:346] getMessage(): [SX1262] R [D4] RSSI: -98.0
14:17:10.632 -> [  9813][D][WeatherSensor.h:573] log_message():           Byte #: 00 01 02 03 04 05 06 07 08 09 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 
14:17:10.675 -> [  9821][D][WeatherSensor.h:583] log_message(): De-whitened Data: E6 F2 AB B9 14 40 B2 00 60 06 00 99 72 04 06 20 79 00 71 27 00 00 00 00 00 AA 
14:17:10.675 -> [  9834][D][WeatherSensor.cpp:511] findSlot(): sensor[0]: v=0 id=0x00000000 t=0 c=0
14:17:23.150 -> [ 22310][D][WeatherSensor.cpp:346] getMessage(): [SX1262] R [D4] RSSI: -104.0
14:17:23.150 -> [ 22310][D][WeatherSensor.h:573] log_message():           Byte #: 00 01 02 03 04 05 06 07 08 09 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 
14:17:23.150 -> [ 22318][D][WeatherSensor.h:583] log_message(): De-whitened Data: CC 8E AB B9 14 40 B2 01 20 12 00 99 72 04 06 20 79 00 71 30 00 00 00 00 00 AA 
14:17:23.150 -> [ 22331][D][WeatherSensor.cpp:511] findSlot(): sensor[0]: v=0 id=0x0000ABB9 t=1 c=0
14:17:23.268 -> [ 22344][D][RainGauge.cpp:194] prefs_load(): lastUpdate        =1732972593
14:17:23.268 -> [ 22345][D][RainGauge.cpp:195] prefs_load(): startupPrev       =0
14:17:23.268 -> [ 22351][D][RainGauge.cpp:196] prefs_load(): rainPreStartup    =997.200012
14:17:23.268 -> [ 22357][D][RainGauge.cpp:197] prefs_load(): tsDayBegin        =6
14:17:23.268 -> [ 22363][D][RainGauge.cpp:198] prefs_load(): rainDayBegin      =997.200012
14:17:23.279 -> [ 22370][D][RainGauge.cpp:199] prefs_load(): tsWeekBegin       =5
14:17:23.279 -> [ 22376][D][RainGauge.cpp:200] prefs_load(): rainWeekBegin     =9.900000
4:17:23.279 -> [ 22382][D][RainGauge.cpp:201] prefs_load(): wdayPrev          =6
14:17:23.279 -> [ 22388][D][RainGauge.cpp:202] prefs_load(): tsMonthBegin      =10
14:17:23.279 -> [ 22394][D][RainGauge.cpp:203] prefs_load(): rainMonthBegin    =9.900000
14:17:23.279 -> [ 22400][D][RainGauge.cpp:204] prefs_load(): rainPrev          =997.200012
[... 2 lines cut ...]
14:17:23.279 -> [ 22418][D][RainGauge.cpp:325] update(): t_delta: 1007430616
14:17:23.279 -> [ 22423][D][RainGauge.cpp:345] update(): hist[2]=0 (upd)
14:17:23.279 -> [ 22429][D][RainGauge.cpp:383] update(): hist[]={0, 0, 0, -1, -1, -1, -1, -1, -1, -1, }
[... cut ...]
14:17:40.433 -> [ 39488][I][BresserWeatherSensorMQTT.ino:704] publishWeatherdata(): ESPWeather-3C5268/18839309/data: {"id":411276041,"ch":2,"battery_ok":1,"temp_c":6.70,"moisture":93}
14:17:40.433 -> 
14:17:48.277 -> [ 47308][D][WeatherSensor.cpp:346] getMessage(): [SX1262] R [D4] RSSI: -103.5
14:17:48.277 -> [ 47309][D][WeatherSensor.h:573] log_message():           Byte #: 00 01 02 03 04 05 06 07 08 09 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 
14:17:48.277 -> [ 47317][D][WeatherSensor.h:583] log_message(): De-whitened Data: 96 6D AB B9 15 6

matthias-bs, 2024-12-02:
I added the board definition to the code.
I don't have this board, so I cannot test it.
Regarding the other issues regarding potential library/tool problems - I cannot reproduce this or find any hints in your report to analyze.

tonbor, 2024-12-04:
> I added the board definition to the code. I don't have this board, so I cannot test it. Regarding the other issues regarding potential library/tool problems - I cannot reproduce this or find any hints in your report to analyze.

Well I found the problem (I think) compiling with Radiolib 7.1 the sx1262 does nog give proper results to work with in your library. Installing Radiolib 7.02 and older every things works fine. See debug result:
Radiolib 7.1
11:02:42.572 ->     14 : GPIO
11:02:42.572 ->     19 : USB_DM
11:02:42.572 ->     20 : USB_DP
11:02:42.572 ->     43 : UART_TX[0]
11:02:42.572 ->     44 : UART_RX[0]
11:02:42.572 -> ============ After Setup End =============
11:03:41.495 -> [ 64545][D][WeatherSensor.cpp:351] getMessage(): [SX1262] R [D4] RSSI: -106.5
11:03:41.495 -> [ 64545][D][WeatherSensorDecoders.cpp:723] decodeBresser7In1Payload(): Digest check failed - [F690] vs [68C8] (9E58)
11:03:41.603 -> [ 64551][D][WeatherSensorDecoders.cpp:449] decodeBresser6In1Payload(): Digest check failed - [5C3A] != [2590]
11:03:41.603 -> [ 64561][D][WeatherSensorDecoders.cpp:242] decodeBresser5In1Payload(): Parity wrong at column 0
11:03:41.603 -> [ 64569][D][WeatherSensorDecoders.cpp:880] decodeBresserLightningPayload(): Digest check failed - [F690] vs [477C] (B1EC)
11:03:41.603 -> [ 64580][D][WeatherSensorDecoders.cpp:987] decodeBresserLeakagePayload(): CRC16 check failed - [A785] vs [5C3A]
11:03:48.200 -> [ 71222][D][WeatherSensor.cpp:351] getMessage(): [SX1262] R [D4] RSSI: -107.5
11:03:48.200 -> [ 71222][D][WeatherSensorDecoders.cpp:723] decodeBresser7In1Payload(): Digest check failed - [2C93] vs [1BF1] (3762)
11:03:48.200 -> [ 71228][D][WeatherSensorDecoders.cpp:449] decodeBresser6In1Payload(): Digest check failed - [8639] != [DD0B]
11:03:48.200 -> [ 71238][D][WeatherSensorDecoders.cpp:242] decodeBresser5In1Payload(): Parity wrong at column 0
11:03:48.200 -> [ 71246][D][WeatherSensorDecoders.cpp:880] decodeBresserLightningPayload(): Digest check failed - [2C93] vs [6A28] (46BB)
11:03:48.237 -> [ 71257][D][WeatherSensorDecoders.cpp:987] decodeBresserLeakagePayload(): CRC16 check failed - [DB17] vs [8639]
11:04:12.571 -> [ 95627][I][BresserWeatherSensorMQTT_T190.ino:827] loop(): ESPWeather-3C5268/status: online
11:04:12.571 -> 
11:04:12.605 -> [ 95630][I][BresserWeatherSensorMQTT_T190.ino:740] publishRadio(): ESPWeather-3C5268/radio: {"rssi":-102}
11:05:25.977 -> [168654][D][WeatherSensor.cpp:351] getMessage(): [SX1262] R [D4] RSSI: -108.0
11:05:25.977 -> [168655][D][WeatherSensorDecoders.cpp:723] decodeBresser7In1Payload(): Digest check failed - [329E] vs [863F] (B4A1)
11:05:25.977 -> [168661][D][WeatherSensorDecoders.cpp:449] decodeBresser6In1Payload(): Digest check failed - [9834] != [DCD]
11:05:25.977 -> [168670][D][WeatherSensorDecoders.cpp:242] decodeBresser5In1Payload(): Parity wrong at column 0
11:05:25.977 -> [168678][D][WeatherSensorDecoders.cpp:880] decodeBresserLightningPayload(): Digest check failed - [329E] vs [5A04] (689A)
11:05:25.977 -> [168689][D][WeatherSensorDecoders.cpp:987] decodeBresserLeakagePayload(): CRC16 check failed - [0DC1] vs [9834]
11:05:42.615 -> [185635][I][BresserWeatherSensorMQTT_T190.ino:827] loop(): ESPWeather-3C5268/status: online
Radiolib 7.02
11:21:12.928 ->     20 : USB_DP
11:21:12.928 ->     43 : UART_TX[0]
11:21:12.928 ->     44 : UART_RX[0]
11:21:12.928 -> ============ After Setup End =============
11:21:14.414 -> [  6574][D][WeatherSensor.cpp:351] getMessage(): [SX1262] R [D4] RSSI: -105.5
11:21:14.447 -> [  6575][D][WeatherSensor.h:598] log_message():           Byte #: 00 01 02 03 04 05 06 07 08 09 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 
11:21:14.447 -> [  6583][D][WeatherSensor.h:608] log_message(): De-whitened Data: E7 47 AB B9 13 40 B2 00 90 09 01 00 45 70 05 20 92 00 79 27 00 00 00 00 00 AA 
11:21:14.447 -> [  6596][D][WeatherSensorDecoders.cpp:105] findSlot(): sensor[0]: v=0 id=0x00000000 t=0 c=0
11:21:14.482 -> [  6604][D][WeatherSensorDecoders.cpp:105] findSlot(): sensor[1]: v=0 id=0x00000000 t=0 c=0
11:21:14.482 -> [  6612][D][WeatherSensorDecoders.cpp:105] findSlot(): sensor[2]: v=0 id=0x00000000 t=0 c=0
11:21:14.482 -> [  6620][D][WeatherSensorDecoders.cpp:105] findSlot(): sensor[3]: v=0 id=0x00000000 t=0 c=0
11:21:14.482 -> [  6628][D][WeatherSensorDecoders.cpp:105] findSlot(): sensor[4]: v=0 id=0x00000000 t=0 c=0
11:21:14.482 -> [  6636][D][WeatherSensorDecoders.cpp:105] findSlot(): sensor[5]: v=0 id=0x00000000 t=0 c=0
11:21:26.934 -> [ 19075][D][WeatherSensor.cpp:351] getMessage(): [SX1262] R [D4] RSSI: -102.5
11:21:26.934 -> [ 19075][D][WeatherSensor.h:598] log_message():           Byte #: 00 01 02 03 04 05 06 07 08 09 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 
11:21:26.934 -> [ 19084][D][WeatherSensor.h:608] log_message(): De-whitened Data: 16 E6 AB B9 15 00 B2 00 80 08 01 00 45 70 05 20 91 00 78 33 00 00 00 00 00 AA 
11:21:27.047 -> [ 19096][D][WeatherSensorDecoders.cpp:105] findSlot(): sensor[0]: v=0 id=0x0000ABB9 t=1 c=0
11:21:27.047 -> [ 19104][D][WeatherSensorDecoders.cpp:105] findSlot(): sensor[1]: v=0 id=0x00000000 t=0 c=0
11:21:27.047 -> [ 19112][D][WeatherSensorDecoders.cpp:105] findSlot(): sensor[2]: v=0 id=0x00000000 t=0 c=0
11:21:27.047 -> [ 19120][D][WeatherSensorDecoders.cpp:105] findSlot(): sensor[3]: v=0 id=0x00000000 t=0 c=0
11:21:27.047 -> [ 19128][D][WeatherSensorDecoders.cpp:105] findSlot(): sensor[4]: v=0 id=0x00000000 t=0 c=0
11:21:27.047 -> [ 19136][D][WeatherSensorDecoders.cpp:105] findSlot(): sensor[5]: v=0 id=0x00000000 t=0 c=0
11:21:27.047 -> [ 19150][D][RainGauge.cpp:194] prefs_load(): lastUpdate        =1733307587
11:21:27.047 -> [ 19151][D][RainGauge.cpp:195] prefs_load(): startupPrev       =0
11:21:27.048 -> [ 19157][D][RainGauge.cpp:196] prefs_load(): rainPreStartup    =1004.500000
11:21:27.048 -> [ 19164][D][RainGauge.cpp:197] prefs_load(): tsDayBegin        =3
11:21:27.048 -> [ 19169][D][RainGauge.cpp:198] prefs_load(): rainDayBegin      =1004.500000

matthias-bs, 2024-12-04:
Interesting... I did not find any clue in the RadioLib issues or related examples.
I have a few different boards, but none with an SX1262, so I cannot test and debug it.

tonbor, 2024-12-05:
Well till there is a solution I will stick with Radioalib version 7.02

tonbor, 2024-12-09:
> Interesting... I did not find any clue in the RadioLib issues or related examples. I have a few different boards, but none with an SX1262, so I cannot test and debug it.

solution?

matthias-bs, 2024-12-16:
Fixed in https://github.com/jgromes/RadioLib/issues/1350

closed 2024-12-16 by matthias-bs

tonbor, 2024-12-17:
Well not fixed, with Radiolib 7.1 still problems, had to go back to 6.60 to get everything working agian. RadioLib/issues/1268 still present

Radiolib: Pushed a commit adding the automated configuration - with that the issue is resolved. I will keep digging further into [#1268](https://github.com/jgromes/RadioLib/issues/1268) since that very much feels related, but still present with this fix.

1:51:21.055 -> [212563][D][WeatherSensor.cpp:351] getMessage(): [SX1262] R [D4] RSSI: -101.5
11:51:21.089 -> [212564][D][WeatherSensorDecoders.cpp:723] decodeBresser7In1Payload(): Digest check failed - [45EA] vs [C7E7] (820D)
**_11:51:21.089 -> [212569][D][WeatherSensorDecoders.cpp:449] decodeBresser6In1Payload(): Digest check failed - [EF40] != [58AD]_**
11:51:21.089 -> [212579][D][WeatherSensorDecoders.cpp:242] decodeBresser5In1Payload(): Parity wrong at column 0
11:51:21.089 -> [212587][D][WeatherSensorDecoders.cpp:880] decodeBresserLightningPayload(): Digest check failed - [45EA] vs [E732] (A2D8)
11:51:21.124 -> [212598][D][WeatherSensorDecoders.cpp:987] decodeBresserLeakagePayload(): CRC16 check failed - [E64F] vs [EF40]
Replacing RadioLib@7.1.0 with RadioLib@7.0.2 (not working)
Installed [RadioLib@6.6.0](mailto:RadioLib@6.6.0)
2:31:33.091 -> [  9375][D][WeatherSensorDecoders.cpp:582] decodeBresser6In1Payload(): Flags: Temp=1  Hum=0  Wind=0  Rain=0  UV=0
12:31:37.692 -> [ 13984][D][WeatherSensor.cpp:351] getMessage(): [SX1262] R [D4] RSSI: -100.5
12:31:37.692 -> [ 13985][D][WeatherSensor.h:598] log_message():           Byte #: 00 01 02 03 04 05 06 07 08 09 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 
12:31:37.692 -> [ 13993][D][WeatherSensor.h:608] log_message(): De-whitened Data: 5A B0 AB B9 20 00 B2 00 00 00 01 04 41 94 08 60 96 00 45 64 00 00 00 00 00 AA 
12:31:37.692 -> [ 14006][D][WeatherSensorDecoders.cpp:105] findSlot(): sensor[0]: v=0 id=0x18839309 t=4 c=0
12:31:37.800 -> [ 14014][D][WeatherSensorDecoders.cpp:105] findSlot(): sensor[1]: v=0 id=0x00000000 t=0 c=0
12:31:37.800 -> [ 14022][D][WeatherSensorDecoders.cpp:105] findSlot(): sensor[2]: v=0 id=0x00000000 t=0 c=0
12:31:37.800 -> [ 14030][D][WeatherSensorDecoders.cpp:105] findSlot(): sensor[3]: v=0 id=0x00000000 t=0 c=0
12:31:37.800 -> [ 14038][D][WeatherSensorDecoders.cpp:105] findSlot(): sensor[4]: v=0 id=0x00000000 t=0 c=0
12:31:37.800 -> [ 14046][D][WeatherSensorDecoders.cpp:105] findSlot(): sensor[5]: v=0 id=0x00000000 t=0 c=0
12:31:50.184 -> [ 26483][D][WeatherSensor.cpp:351] getMessage(): [SX1262] R [D4] RSSI: -95.0
12:31:50.184 -> [ 26483][D][WeatherSensor.h:598] log_message():           Byte #: 00 01 02 03 04 05 06 07 08 09 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 
12:31:50.184 -> [ 26491][D][WeatherSensor.h:608] log_message(): De-whitened Data: 6A 86 AB B9 17 20 B2 00 60 06 01 04 41 94 08 60 96 00 46 15 00 00 00 00 00 AA 
12:31:50.222 -> [ 26504][D][WeatherSensorDecoders.cpp:105] findSlot(): sensor[0]: v=0 id=0x0000ABB9 t=1 c=0
12:31:50.222 -> [ 26512][D][WeatherSensorDecoders.cpp:105] findSlot(): sensor[1]: v=0 id=0x00000000 t=0 c=0
12:31:50.222 -> [ 26520][D][WeatherSensorDecoders.cpp:105] findSlot(): sensor[2]: v=0 id=0x00000000 t=0 c=0
12:31:50.222 -> [ 26528][D][WeatherSensorDecoders.cpp:105] findSlot(): sensor[3]: v=0 id=0x00000000 t=0 c=0
12:31:50.222 -> [ 26536][D][WeatherSensorDecoders.cpp:105] findSlot(): sensor[4]: v=0 id=0x00000000 t=0 c=0
12:31:50.256 -> [ 26544][D][WeatherSensorDecoders.cpp:105] findSlot(): sensor[5]: v=0 id=0x00000000 t=0 c=0
12:31:50.256 -> [ 26572][D][RainGauge.cpp:194] prefs_load(): lastUpdate        =1734432347
12:31:50.256 -> [ 26573][D][RainGauge.cpp:195] prefs_load(): startupPrev       =0
12:31:50.290 -> [ 26574][D][RainGauge.cpp:196] prefs_load(): rainPreStartup    =1044.099976
12:31:50.290 -> [ 26580][D][RainGauge.cpp:197] prefs_load(): tsDayBegin        =2
12:31:50.290 -> [ 26586][D][RainGauge.cpp:198] prefs_load(): rainDayBegin      =1044.099976
12:31:50.290 -> [ 26593][D][RainGauge.cpp:199] prefs_load(): tsWeekBegin       =1
12:31:50.290 -> [ 26599][D][RainGauge.cpp:200] prefs_load(): rainWeekBegin     =1043.099976
12:31:50.290 -> [ 26605][D][RainGauge.cpp:201] prefs_load(): wdayPrev          =2
12:31:50.326 -> [ 26611][D][RainGauge.cpp:202] prefs_load(): tsMonthBegin      =11
12:31:50.326 -> [ 26617][D][RainGauge.cpp:203] prefs_load(): rainMonthBegin    =997.200012
12:31:50.326 -> [ 26624][D][RainGauge.cpp:204] prefs_load(): rainPrev          =1044.099976
12:31:50.326 -> [ 26630][D][RainGauge.cpp:205] prefs_load(): rainAcc           =0.000000
12:31:50.326 -> [ 26637][D][RainGauge.cpp:280] update(): rainDelta: 0.0

matthias-bs, 2024-12-17:
Release v7.1.0 remains unchanged. You have to pull the latest version from the main branch or wait for the next release.

tonbor, 2024-12-25:
With radiolib 7.1.1 all ok