matthias-bs/BresserWeatherSensorReceiver #205: I can't get version 0.28.10 for HelTec WiFi LoRa 32(V2) to compile.

gitpower2017, 2024-12-04:
I have found a HelTec WiFi LoRa 32(V2) and thought the optimal basis for this project
In Arduino 1.8.19 I have entered this URL under “Voreinstellungen”: “https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/releases/download/0.0.5/package_heltec_esp32_index.json”
Then I tried to compile the project according to the instructions.
I used “BresserWeatherSensorMQTTCustomer”.  (or BresserWeatherSensorMQTT)

Edit WeatherSensorCfg.h
<code>
// This define is set by selecting "Board: Heltec WiFi LoRa 32(V2)" in the Arduino IDE:
#define ARDUINO_HELTEC_WIFI_LORA_32_V2
</code> 

I had to make a few changes in the file “BresserWeatherSensorMQTTCustom.ino”.
Line 184...
<code>
#if defined(ESP32)
  #if defined(USE_WIFI)
      #include <WiFi.h>
  #elif defined(USE_SECUREWIFI)
      //#include <NetworkClientSecure.h> 
      #include <WiFiClientSecure.h>
  #endif
#elif defined(ESP8266)
  #include <ESP8266WiFi.h>
#endif
</code>
Zeile 321..
<code>
// Generate WiFi network instance                                                                  
#if defined(ESP32)                                                                                 
  #if defined(USE_WIFI)
    WiFiClient net;
  #elif defined(USE_SECUREWIFI)
    //NetworkClientSecure net;
    WiFiClientSecure net;
  #endif
#elif defined(ESP8266)
  #if defined(USE_WIFI)
    WiFiClient net;
  #elif defined(USE_SECUREWIFI)
    BearSSL::WiFiClientSecure net;
  #endif
#endif
</code>
Zeile 414
<code>
 //WiFi.hostname(Hostname); //Habe keine andere Lösung gefunden
</code>
Zeile 732
<code>
//JsonDocument payload;
StaticJsonDocument<64> payload;
</code>

After all these changes I now get the error

<code>
/home/ls/Arduino/libraries/BresserWeatherSensorReceiver/src/WeatherSensor.cpp: In member function 'int16_t WeatherSensor::begin(uint8_t, bool)':
/home/ls/Arduino/libraries/BresserWeatherSensorReceiver/src/WeatherSensor.cpp:226:11: error: 'class SX1276' has no member named 'setPacketReceivedAction'
     radio.setPacketReceivedAction(setFlag);
           ^
Mehrere Bibliotheken wurden für "Preferences.h" gefunden
 Benutzt: /home/ls/.arduino15/packages/Heltec-esp32/hardware/esp32/0.0.5/libraries/Preferences
 Nicht benutzt: /home/ls/Arduino/libraries/Preferences
exit status 1
Fehler beim Kompilieren für das Board WiFi LoRa 32(V2).
</code

Now I am at a complete loss as to how to get this compiled.
I hope someone can help me

matthias-bs, 2024-12-04:
Hi,

please use the board package from https://github.com/espressif/arduino-esp32 (in the Arduino IDE: https://espressif.github.io/arduino-esp32/package_esp32_index.json) and a more recent version of the Arduino IDE (v2.3.3).
Additionally, please check the required library versions in https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/package.json.

Best regards,
Matthias

gitpower2017, 2024-12-04:
Many thanks for the tip.
I have tried to compile this under 2.3.3.
Unfortunately I have now received this error.
Too bad that there are no suitable instructions.
For me as a beginner this is very complicated.

Unter Einstellungen:
https://espressif.github.io/arduino-esp32/package_esp32_index.json
https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/package.json

 Error:
<code>
In file included from /home/ls/Dokumente/Bresser2024_ide2.x/BresserWeatherSensorMQTT/BresserWeatherSensorMQTT.ino:203:
/home/ls/Dokumente/Bresser2024_ide2.x/BresserWeatherSensorMQTT/WeatherSensorCfg.h:215:89: note: #pragma message: ARDUINO_HELTEC_WIFI_LORA_32_V2 defined; using on-board transceiver
     #pragma message("ARDUINO_HELTEC_WIFI_LORA_32_V2 defined; using on-board transceiver")
                                                                                         ^
In file included from /home/ls/Dokumente/Bresser2024_ide2.x/BresserWeatherSensorMQTT/BresserWeatherSensorMQTT.ino:203:
/home/ls/Dokumente/Bresser2024_ide2.x/BresserWeatherSensorMQTT/WeatherSensorCfg.h:611:48: note: #pragma message: Receiver chip: [SX1276]
 #pragma message("Receiver chip: " RECEIVER_CHIP)
                                                ^
/home/ls/Dokumente/Bresser2024_ide2.x/BresserWeatherSensorMQTT/WeatherSensorCfg.h:612:170: note: #pragma message: Pin config: RST->RST_LoRa, CS->SS, GD0/G0/IRQ->DIO0, GDO2/G1/GPIO->DIO1
 #pragma message("Pin config: RST->" STR(PIN_RECEIVER_RST) ", CS->" STR(PIN_RECEIVER_CS) ", GD0/G0/IRQ->" STR(PIN_RECEIVER_IRQ) ", GDO2/G1/GPIO->" STR(PIN_RECEIVER_GPIO) )
                                                                                                                                                                          ^
In file included from /home/ls/.arduino15/packages/arduino/hardware/esp32/2.0.18-arduino.5/cores/esp32/Arduino.h:235,
                 from /home/ls/Dokumente/Bresser2024_ide2.x/BresserWeatherSensorMQTT/BresserWeatherSensorMQTT.ino:134:
/home/ls/.arduino15/packages/arduino/hardware/esp32/2.0.18-arduino.5/cores/esp32/io_pin_remap.h:47:61: error: 'digitalPinToGPIONumber' is not a type
 #define pinMode(pin, mode)                          pinMode(digitalPinToGPIONumber(pin), mode)
                                                             ^~~~~~~~~~~~~~~~~~~~~~
/home/ls/Arduino/libraries/RadioLib/src/Hal.h:68:18: note: in expansion of macro 'pinMode'
     virtual void pinMode(uint32_t pin, uint32_t mode) = 0;
                  ^~~~~~~
/home/ls/.arduino15/packages/arduino/hardware/esp32/2.0.18-arduino.5/cores/esp32/io_pin_remap.h:46:66: error: 'digitalPinToGPIONumber' is not a type
 #define digitalWrite(pin, val)                      digitalWrite(digitalPinToGPIONumber(pin), val)
                                                                  ^~~~~~~~~~~~~~~~~~~~~~
/home/ls/Arduino/libraries/RadioLib/src/Hal.h:76:18: note: in expansion of macro 'digitalWrite'
     virtual void digitalWrite(uint32_t pin, uint32_t value) = 0;
                  ^~~~~~~~~~~~
/home/ls/.arduino15/packages/arduino/hardware/esp32/2.0.18-arduino.5/cores/esp32/io_pin_remap.h:42:65: error: 'digitalPinToGPIONumber' is not a type
 #define digitalRead(pin)                            digitalRead(digitalPinToGPIONumber(pin))
                                                                 ^~~~~~~~~~~~~~~~~~~~~~
/home/ls/Arduino/libraries/RadioLib/src/Hal.h:84:22: note: in expansion of macro 'digitalRead'
     virtual uint32_t digitalRead(uint32_t pin) = 0;
                      ^~~~~~~~~~~
/home/ls/.arduino15/packages/arduino/hardware/esp32/2.0.18-arduino.5/cores/esp32/io_pin_remap.h:44:69: error: 'digitalPinToGPIONumber' is not a type
 #define attachInterrupt(pin, fcn, mode)             attachInterrupt(digitalPinToGPIONumber(pin), fcn, mode)
                                                                     ^~~~~~~~~~~~~~~~~~~~~~
/home/ls/Arduino/libraries/RadioLib/src/Hal.h:93:18: note: in expansion of macro 'attachInterrupt'
     virtual void attachInterrupt(uint32_t interruptNum, void (*interruptCb)(void), uint32_t mode) = 0;
                  ^~~~~~~~~~~~~~~
/home/ls/.arduino15/packages/arduino/hardware/esp32/2.0.18-arduino.5/cores/esp32/io_pin_remap.h:45:69: error: 'digitalPinToGPIONumber' is not a type
 #define detachInterrupt(pin)                        detachInterrupt(digitalPinToGPIONumber(pin))
                                                                     ^~~~~~~~~~~~~~~~~~~~~~
/home/ls/Arduino/libraries/RadioLib/src/Hal.h:100:18: note: in expansion of macro 'detachInterrupt'
     virtual void detachInterrupt(uint32_t interruptNum) = 0;
                  ^~~~~~~~~~~~~~~
/home/ls/.arduino15/packages/arduino/hardware/esp32/2.0.18-arduino.5/cores/esp32/io_pin_remap.h:19:45: error: 'digitalPinToGPIONumber' is not a type
 #define pulseIn(pin, args...)       pulseIn(digitalPinToGPIONumber(pin), args)
                                             ^~~~~~~~~~~~~~~~~~~~~~
/home/ls/Arduino/libraries/RadioLib/src/Hal.h:138:18: note: in expansion of macro 'pulseIn'
     virtual long pulseIn(uint32_t pin, uint32_t state, RadioLibTime_t timeout) = 0;
                  ^~~~~~~
/home/ls/.arduino15/packages/arduino/hardware/esp32/2.0.18-arduino.5/cores/esp32/io_pin_remap.h:21:42: error: 'digitalPinToGPIONumber' is not a type
 #define tone(_pin, args...)         tone(digitalPinToGPIONumber(_pin), args)
                                          ^~~~~~~~~~~~~~~~~~~~~~
[... cut ...]
 #define tone(_pin, args...)         tone(digitalPinToGPIONumber(_pin), args)
                                                                            ^
/home/ls/Arduino/libraries/RadioLib/src/protocols/FSK4/FSK4.h:91:10: note: in expansion of macro 'tone'
     void tone(uint8_t i);
          ^~~~
Mehrere Bibliotheken wurden für "Preferences.h" gefunden
  Benutzt: /home/ls/.arduino15/packages/arduino/hardware/esp32/2.0.18-arduino.5/libraries/Preferences
  Nicht benutzt: /home/ls/Arduino/libraries/Preferences
exit status 1

Compilation error: exit status 1
</code>

gitpower2017, 2024-12-04:
Many thanks for your help

I had to add this URL to the IDE
https://espressif.github.io/arduino-esp32/package_esp32_index.json
https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/releases/download/3.0.0/package_heltec_esp32_index.json
https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/package.json

I hope everything works now.
So far I don't see any 5.1 sensor

matthias-bs, 2024-12-04:
Please remove https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/releases/download/3.0.0/package_heltec_esp32_index.json from your preferences, it might cause conflicts with the Espressif package.

The file https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/package.json is used by the library manager when you install BresserWeatherSensorReceiver, but you can also view it and install the indicated libraries manually (with the library manager). In any case, it is not meant for listing in the preferences section.

Please see here for getting debug output: https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/DEBUG_OUTPUT.md

matthias-bs, 2024-12-04:
Please try to get https://github.com/matthias-bs/BresserWeatherSensorReceiver/tree/main/examples/BresserWeatherSensorBasic running first.

gitpower2017, 2024-12-05:
First Step:
Compile:  BresserWeatherSensorBasic
<code>
Id: [      74] Typ: [0] Ch: [0] St: [0] Bat: [OK ] RSSI: [-100.5dBm] Temp: [ -1.7C] Hum: [ 99%] Wmax: [ 0.0m/s] Wavg: [ 0.0m/s] Wdir: [ 90.0deg] Rain: [    0.0mm] UVidx: [--.-] Light: [--.-klx] 
Id: [      74] Typ: [0] Ch: [0] St: [0] Bat: [OK ] RSSI: [-101.0dBm] Temp: [ -1.6C] Hum: [ 99%] Wmax: [ 0.0m/s] Wavg: [ 0.0m/s] Wdir: [ 90.0deg] Rain: [    0.0mm] UVidx: [--.-] Light: [--.-klx] 
Id: [      74] Typ: [0] Ch: [0] St: [0] Bat: [OK ] RSSI: [ -95.5dBm] Temp: [ -1.5C] Hum: [ 99%] Wmax: [ 0.0m/s] Wavg: [ 0.0m/s] Wdir: [ 90.0deg] Rain: [    0.0mm] UVidx: [--.-] Light: [--.-klx] 
Id: [      74] Typ: [0] Ch: [0] St: [0] Bat: [OK ] RSSI: [ -94.5dBm] Temp: [ -1.7C] Hum: [ 99%] Wmax: [ 0.0m/s] Wavg: [ 0.0m/s] Wdir: [ 90.0deg] Rain: [    0.0mm] UVidx: [--.-] Light: [--.-klx] 
Id: [      74] Typ: [0] Ch: [0] St: [0] Bat: [OK ] RSSI: [-106.5dBm] Temp: [ -1.7C] Hum: [ 99%] Wmax: [ 0.0m/s] Wavg: [ 0.0m/s] Wdir: [ 90.0deg] Rain: [    0.0mm] UVidx: [--.-] Light: [--.-klx] 
Id: [      74] Typ: [0] Ch: [0] St: [0] Bat: [OK ] RSSI: [-106.0dBm] Temp: [ -1.6C] Hum: [ 99%] Wmax: [ 0.0m/s] Wavg: [ 0.0m/s] Wdir: [ 90.0deg] Rain: [    0.0mm] UVidx: [--.-] Light: [--.-klx] 
Id: [      74] Typ: [0] Ch: [0] St: [0] Bat: [OK ] RSSI: [-107.0dBm] Temp: [ -1.6C] Hum: [ 99%] Wmax: [ 0.0m/s] Wavg: [ 0.0m/s] Wdir: [ 90.0deg] Rain: [    0.0mm] UVidx: [--.-] Light: [--.-klx] 
Id: [      74] Typ: [0] Ch: [0] St: [0] Bat: [OK ] RSSI: [-104.0dBm] Temp: [ -1.6C] Hum: [ 99%] Wmax: [ 0.0m/s] Wavg: [ 0.0m/s] Wdir: [ 90.0deg] Rain: [    0.0mm] UVidx: [--.-] Light: [--.-klx] 
Id: [      74] Typ: [0] Ch: [0] St: [0] Bat: [OK ] RSSI: [-105.5dBm] Temp: [ -1.6C] Hum: [ 99%] Wmax: [ 0.0m/s] Wavg: [ 0.0m/s] Wdir: [ 90.0deg] Rain: [    0.0mm] UVidx: [--.-] Light: [--.-klx] 
Id: [      74] Typ: [0] Ch: [0] St: [0] Bat: [OK ] RSSI: [-105.5dBm] Temp: [ -1.7C] Hum: [ 99%] Wmax: [ 0.0m/s] Wavg: [ 0.0m/s] Wdir: [ 90.0deg] Rain: [    0.0mm] UVidx: [--.-] Light: [--.-klx] 
Id: [      74] Typ: [0] Ch: [0] St: [0] Bat: [OK ] RSSI: [-106.5dBm] Temp: [ -1.7C] Hum: [ 99%] Wmax: [ 0.0m/s] Wavg: [ 0.0m/s] Wdir: [ 90.0deg] Rain: [    0.0mm] UVidx: [--.-] Light: [--.-klx] 
Id: [      74] Typ: [0] Ch: [0] St: [0] Bat: [OK ] RSSI: [-100.0dBm] Temp: [ -1.6C] Hum: [ 99%] Wmax: [ 6.6m/s] Wavg: [ 0.6m/s] Wdir: [  0.0deg] Rain: [    0.0mm] UVidx: [--.-] Light: [--.-klx] 
Id: [      74] Typ: [0] Ch: [0] St: [0] Bat: [OK ] RSSI: [-107.0dBm] Temp: [ -1.6C] Hum: [ 99%] Wmax: [ 0.8m/s] Wavg: [ 0.8m/s] Wdir: [  0.0deg] Rain: [    0.0mm] UVidx: [--.-] Light: [--.-klx] 
Id: [      74] Typ: [0] Ch: [0] St: [0] Bat: [OK ] RSSI: [ -97.0dBm] Temp: [ -1.7C] Hum: [ 99%] Wmax: [ 0.0m/s] Wavg: [ 0.8m/s] Wdir: [  0.0deg] Rain: [    0.0mm] UVidx: [--.-] Light: [--.-klx] 
</code>
This is OK

gitpower2017, 2024-12-05:
I have compiled BresserWeatherSensorMQTT with DEBUG.

I get a message on the console:
<code>
[ 21448][I][BresserWeatherSensorMQTT.ino:711] publishWeatherdata(): WetterKorkESP-DABDAC/74/data: {"id":116,"ch":0,"battery_ok":1,"temp_c":-0.4,"humidity":99,"wind_gust":0.0,"wind_avg":0.0,"wind_dir":90.0,"rain":0.0,"rain_h":0.0,"rain_d":0.0,"rain_w":0.0,"rain_m":0.0}

[ 21476][I][BresserWeatherSensorMQTT.ino:723] publishWeatherdata(): WetterKorkESP-DABDAC/extra: {"wind_dir_txt":"E","wind_gust_bft":0,"wind_avg_bft":0,"dewpoint_c":-0.5,"perceived_temp_c":-0.4}
</code>

but the MQTT broker did not receive this two packet.
RADIO and Status are displayed.

I also get the same result with the MQTTX client

But I see status & radio

Debug output:
[ 6409][D][WeatherSensor.cpp:351] getMessage(): [SX1276] R [D4] RSSI: -104.5

MQTTX:
{“rssi”:-104.5}

matthias-bs, 2024-12-05:
Are you sure that the broker did not receive the messages or did actually a client not receive the messages?
Is it possible that the client was not configured correctly?
Please note:
- RSSI and Status are numbers or strings, respectively.
- "extra" and "data" are sent as JSON encoded messages.
- The topic "data" has a different level of hierarchy: `<hostname>/<sensor-id>/data`

I recommend to use MQTT Explorer.

gitpower2017, 2024-12-05:
As an MQTT broker, I use the implementation of ioBroker.
It is either due to this implementation or to the software.
For example, if I use the MQTTX to send "extra" with a JSON as content, it is immediately displayed in the ionroker. Unfortunately, I cannot sniff at network level

gitpower2017, 2024-12-05:
I adjusted the timings and monitored this with Node-Red.
The system wakes up every 5 minutes and sometimes sends data.
The SLEEP mode is Enable . I reduced the sleep mode to 30 seconds and the result is better

I think my actual problem is solved and the ticket can be closed.

Thanks for the help
[image]
[image]

gitpower2017, 2024-12-05:
I  have disable the SLEEP MODE and I see the DATA on the flow

closed 2024-12-05 by matthias-bs