matthias-bs/BresserWeatherSensorReceiver #211: Sleep Mode for CC1101

pcfritz, 2024-12-16:
Hi I am using BresserWeatherSensorBasic in a project to receive every 15 minutes weather data and to send it to an Influxdb server via mobile network, which works great. But as the esp32 is connected to a LiPo battery, I wonder if it would be possible to not only use deep sleep of esp32 but also sleep mode of CC1101? Would be awesome if you could implement something like this!

matthias-bs, 2024-12-17:
Hi,
I think this is already available. You can call `radio.standby()` at the end of each reception cycle before the ESP32 enters sleep mode. You can also have a look at the other examples, which use `getData()` instead of `getMessage()`:
https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/90663f3e7b20bbfce9d295a6bba86bcbc1fbb783/src/WeatherSensor.cpp#L269
`getData()` calls `radio.standby()` after all required messages have been received.
[RadioLib](https://github.com/jgromes/RadioLib) also provides a `sleep()` method for some modules, but not for the CC1101. If CC1101 has a sleep function in addition to the standby function, you might file an issue in the RadioLib project.

pcfritz, 2024-12-19:
thanks @matthias-bs, that helped. Issue can be closed

matthias-bs, 2025-01-11:
@pcfritz See https://github.com/jgromes/RadioLib/discussions/1382, maybe sleep mode will be implemented soon.

pcfritz, 2025-01-18:
Fingers crossed. If I call ws.sleep(), change the according block in WeatherSensor.cpp to this and move void SPIsendCommand(uint8_t cmd); to public, I still only reach about 1.65ma:

``  void WeatherSensor::sleep(void)
  {
      radio.SPIsendCommand(RADIOLIB_CC1101_CMD_POWER_DOWN);
  }``

matthias-bs, 2025-01-18:
To which extend did the CC1101 sleep command reduce the current?
Is the 1.65 mA only the CC1101 supply current or anything else? For low power/quiescent current, you have to take the entire circuit into account.
Anyway, I will integrate the CC1101 sleep comand in the next release.

pcfritz, 2025-01-19:
Great!

Meanwhile I've changed cc1101.h and cc11011.cpp according to [https://github.com/jgromes/RadioLib/pull/1389](url). 
I also found out, that cc1101 current after ws.sleep() was lower compared to to current after beginning of deepsleep. The reason for it might be the CS pin, because after setting CS as input, high and remaining state in deepsleep, the current drops to <0.1ma.

closed 2025-01-19 by matthias-bs