matthias-bs/BresserWeatherSensorReceiver #217: Separate basic decoding functions from complete weather sensor runtime application

SchoMa1, 2025-01-02:
I normally use Tasmota as code base for ESP/ESP32 application, so I tried to integrate "WeatherSensorDecoders" in a "HELTEC LoRa 32" system together with a BMP280 pressure sensor. The processing host for weather data is a linux node-red machine. ESP32 should be only a mqtt gateway (868Mhz-Mqtt), rain/lightning statistics stored at node-red system (not necessary at ESP32). The display of "HELTEC LoRa 32" should be controlled by node-red.

This means I need only raw Bresser decoding functions from library!

In the end, it works, all features of Tasmota (rule, display, mqtt, wifi,..) are available:

09:26:03.469 MQT: tasmota/stat/Weather/SENSOR = {"Time":"2025-01-02T09:26:03","Bresser":{"123456":{"ch":0,"battery_ok":1,"rssi":-113.00,"snr":-20.00,"stype":"1","temp_c":4.1,"humidity":95,"wind_gust":1.2,"wind_avg":1.2,"wind_dir":192.0,"uv":0.0,"light_lux":796.0,"rain":1273.3}}}

09:26:04.735 MQT: tasmota/tele/Weather/SENSOR = {"Time":"2025-01-02T13:55:55","BMP280":{"Temperature":27.9,"Pressure":979.7},"PressureUnit":"hPa","TempUnit":"C"}

The PROBLEM is, that I need to use nearly all sources and includes, for decoding a single 868-Bresser-weather frame.

The request would be to have an encapsulated decode "library" (for example use in Tasmota) inside a full blown "Mathias-bs/WeatherSensor" library. I think the first step is done with WeatherSensorDecoders.cpp, but there are still too many dependencies.

I would like that this example would be possible, without too much overhead:

```
> #ifdef USE_BRESSER_WEATHER_BRIDGE
>   if ( (packet_size>=MSG_BUF_SIZE) && (0xD4 == rcvd_data[0]) )
>   {
> 	for (size_t i=0; i<weatherSensor.sensor.size(); i++) 
>       weatherSensor.sensor[i].complete = false;
> 	  
>     DecodeStatus decode_status = weatherSensor.decodeMessage((const uint8_t *)&rcvd_data[1], MSG_BUF_SIZE - 1);	
> 	
>     if (DECODE_OK == decode_status)
>     {
>        String mqtt_payload;  // sensor data
> 	  
> 	  size_t i=0;
>   	  for (;(i<weatherSensor.sensor.size())&&!weatherSensor.sensor[i].complete; i++);
> 	  
> 	  if (i<weatherSensor.sensor.size())
> 	  {
>           ......
```

1) in this case Preferences.h is not necessary, should not be referenced (at the moment replaced by dummy)
2) the following "hack" in WeatherSensorCfg.h should not be necessary (radio lib access):
```
> class RadioDmy { 
>   public: 
>     int16_t beginFSK(float freq=0.0, float br=0.0, float freqDev=0.0, float rxBw=0.0, int8_t power=0, uint16_t preambleLength=0, float tcxoVoltage=false, bool useRegulatorLDO=false) {return -1;}
> 	int16_t fixedPacketLengthMode(int v) {return -1;};
> 	int16_t setCrcFiltering(bool v) {return -1;};
> 	int16_t getRSSI(){return -1;};
> 	void radioReset(){};
> 	void sleep(){};	
>     int16_t setSyncWord(uint8_t syncWord, uint8_t controlBits = 0){return -1;};
>     int16_t setSyncWord(uint8_t* syncWord, size_t len){return -1;};
> 	void setPacketReceivedAction(void (*func)(void)){};
> 	int16_t startReceive(){return -1;};
> 	void reset(){};
> 	void standby(){};
> 	int16_t readData(uint8_t *p, int v){return -1;};
> };
> static RadioDmy radio;	
> 
> #define RECEIVER_CHIP "[no]"
> 

```

OVERAL: Integration in Tasmota is not a big deal, all we need is still there ("LoRa Bridge", xdrv_73_9_lora.ino). 
Decoding Bresser protocol plus Tasmota features (Display, Sensors, Scripting, Rules, Mqtt,...) brings a lot of advantages. I am running "Mathias-bs Bresser decoding" inside Tasmota on a "HELTEC LoRa 32" with a bm280 pressure sensor. The "Weather calculations" (rain, pressure, wind, ..) are done with node-red via mqtt. Some weather parameters (temperature, pressure, rain) are set from node-red to tasmota-display (HELTEC LoRa 32). I am using a 6in1 Bresser sensor plus an additional hygrometer sensor.  Data send to "wunderground.com" ...

matthias-bs, 2025-01-02:
The required interface to the decoding part is not 100% clear to me. If you replace the radio object by a dummy, where do you get the sensor messages from?

In general, it should be possible to exclude some parts of the code (including unwanted dependencies) by adding another `#define` to [WeatherSensorCfg.h](https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/src/WeatherSensorCfg.h).

SchoMa1, 2025-01-02:
The whole receive infrastructure (also init and configuration of chip) is done within standard Tasmota system. Tasmota gets frame from Bresser Weather sensor. This frame will be injected into your library and the result is handled in standard Tasmota way.

To make it short, best would be a only decoding module without operating system, radio library or storage stuff.

SchoMa1, 2025-01-03:
_"Did you consider using source code from https://github.com/merbanan/rtl_433 as an alternative?"_
Hm, I never saw this source, not only Bresser sensors are supported, this is a good hint, I will check this maybe this is a better/more universal way.

matthias-bs, 2025-01-25:
@SchoMa1 Did you find an alternative solution?

matthias-bs, 2025-01-31:
Closed due to inactivity

closed 2025-01-31 by matthias-bs