matthias-bs/BresserWeatherSensorReceiver #219: [HELP] I can't get FRobot FireBeetle ESP3 to compile.

gitpower2017, 2025-01-25:
I would like to compile “FRobot FireBeetle ESP3” with “DFRobot FireBeetle Covers-LoRa Radio”.
I am not able to do this. Can't find any instructions either.
Maybe someone can help me?

I used the Ardunio IDE and selected DFRobot Firebeetle 2 as board

Error: [  5272][D][WeatherSensor.cpp:161] begin(): [SX1276] Initializing ... 

However, I cannot find out how to use the Cover-LoRa Radio.

matthias-bs, 2025-01-26:
Filing a compile error issue without posting the compiler output doesn't make much sense...

I tried to create a configuration for this setup, but
as far as I remember, the FireBeetle2 is not pin compatible to the FireBeetle Cover LoRa Radio.

matthias-bs, 2025-01-26:
Also see https://github.com/matthias-bs/BresserWeatherSensorReceiver/issues/165

closed 2025-01-31 by matthias-bs