matthias-bs/BresserWeatherSensorReceiver #220: Request for support for 7003150 8 in 1 station

jpdias, 2025-01-26:
The library does not seem to support the [Bresser 7003150 8 in 1 Weather station](https://www.bresseruk.com/p/bresser-8-in-1-clearviewtb-weather-station-7003150).  It identifies the station, transmission signal but no data. 

Here is some of the debug logs with some data

`
Id: [    6241] Typ: [D] Ch: [0] St: [0] Bat: [Low] RSSI: [-107.5dBm] Temp: [---.-C] Hum: [---%] Wmax: [--.-m/s] Wavg: [--.-m/s] Wdir: [---.-deg] Rain: [-----.-mm] UVidx: [--.-] Light: [--.-klx] 
[334842][V][WeatherSensor.cpp:392] getMessage(): [SX1276] Data: D4 6E A7 C8 EB 88 2A D8 AD AA FD AA A8 98 AA BF FC 3E AA 82 22 AA AA BE 3A AA 00 
[334854][D][WeatherSensor.cpp:394] getMessage(): [SX1276] R [D4] RSSI: -70.5
[334861][D][WeatherSensor.h:647] log_message():           Byte #: 00 01 02 03 04 05 06 07 08 09 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 
[334874][D][WeatherSensor.h:657] log_message(): De-whitened Data: C4 0D 62 41 22 80 72 07 00 57 00 02 32 00 15 56 94 00 28 88 00 00 14 90 00 AA 
[334887][V][WeatherSensorDecoders.cpp:67] findSlot(): find_slot(): ID=00006241
[334894][D][WeatherSensorDecoders.cpp:105] findSlot(): sensor[0]: v=0 id=0x00006241 t=13 c=0
[334902][V][WeatherSensorDecoders.cpp:130] findSlot(): find_slot(): Storing into slot #0
Id: [    6241] Typ: [D] Ch: [0] St: [0] Bat: [Low] RSSI: [ -70.5dBm] Temp: [---.-C] Hum: [---%] Wmax: [--.-m/s] Wavg: [--.-m/s] Wdir: [---.-deg] Rain: [-----.-mm] UVidx: [--.-] Light: [--.-klx] 
[346923][V][WeatherSensor.cpp:392] getMessage(): [SX1276] Data: D4 3D 62 C8 EB 88 8A D8 AE 2A EB AA A8 98 AA BF FC 3E AA 82 22 AA AA BE 3A AA 00 
[346935][D][WeatherSensor.cpp:394] getMessage(): [SX1276] R [D4] RSSI: -107.0
[346943][D][WeatherSensor.h:647] log_message():           Byte #: 00 01 02 03 04 05 06 07 08 09 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 
[346955][D][WeatherSensor.h:657] log_message(): De-whitened Data: 97 C8 62 41 22 20 72 04 80 41 00 02 32 00 15 56 94 00 28 88 00 00 14 90 00 AA 
[346968][V][WeatherSensorDecoders.cpp:67] findSlot(): find_slot(): ID=00006241
[346975][D][WeatherSensorDecoders.cpp:105] findSlot(): sensor[0]: v=0 id=0x00006241 t=13 c=0
[346983][V][WeatherSensorDecoders.cpp:130] findSlot(): find_slot(): Storing into slot #0
Id: [    6241] Typ: [D] Ch: [0] St: [0] Bat: [Low] RSSI: [-107.0dBm] Temp: [---.-C] Hum: [---%] Wmax: [--.-m/s] Wavg: [--.-m/s] Wdir: [---.-deg] Rain: [-----.-mm] UVidx: [--.-] Light: [--.-klx] 
[358904][V][WeatherSensor.cpp:392] getMessage(): [SX1276] Data: D4 EF 50 C8 EB 82 2A D8 AC DA FC AA A8 98 AA BF FC 3E AA 82 22 AA AA BE 3A AA 00 
[358916][D][WeatherSensor.cpp:394] getMessage(): [SX1276] R [D4] RSSI: -101.0
[358924][D][WeatherSensor.h:647] log_message():           Byte #: 00 01 02 03 04 05 06 07 08 09 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 
[358936][D][WeatherSensor.h:657] log_message(): De-whitened Data: 45 FA 62 41 28 80 72 06 70 56 00 02 32 00 15 56 94 00 28 88 00 00 14 90 00 AA 
[358949][V][WeatherSensorDecoders.cpp:67] findSlot(): find_slot(): ID=00006241
[358956][D][WeatherSensorDecoders.cpp:105] findSlot(): sensor[0]: v=0 id=0x00006241 t=13 c=0
[358964][V][WeatherSensorDecoders.cpp:130] findSlot(): find_slot(): Storing into slot #0
Id: [    6241] Typ: [D] Ch: [0] St: [0] Bat: [Low] RSSI: [-101.0dBm] Temp: [---.-C] Hum: [---%] Wmax: [--.-m/s] Wavg: [--.-m/s] Wdir: [---.-deg] Rain: [-----.-mm] UVidx: [--.-] Light: [--.-klx] 
[370885][V][WeatherSensor.cpp:392] getMessage(): [SX1276] Data: D4 D9 B7 C8 EB 8F 2A D8 A2 2A D2 AA A8 98 AA BF FC 3E AA 82 22 AA AA BE 3A AA 00 
[370897][D][WeatherSensor.cpp:394] getMessage(): [SX1276] R [D4] RSSI: -104.0
[370905][D][WeatherSensor.h:647] log_message():           Byte #: 00 01 02 03 04 05 06 07 08 09 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 
[370917][D][WeatherSensor.h:657] log_message(): De-whitened Data: 73 1D 62 41 25 80 72 08 80 78 00 02 32 00 15 56 94 00 28 88 00 00 14 90 00 AA 
[370930][V][WeatherSensorDecoders.cpp:67] findSlot(): find_slot(): ID=00006241
[370937][D][WeatherSensorDecoders.cpp:105] findSlot(): sensor[0]: v=0 id=0x00006241 t=13 c=0
[370945][V][WeatherSensorDecoders.cpp:130] findSlot(): find_slot(): Storing into slot #0
Id: [    6241] Typ: [D] Ch: [0] St: [0] Bat: [Low] RSSI: [-104.0dBm] Temp: [---.-C] Hum: [---%] Wmax: [--.-m/s] Wavg: [--.-m/s] Wdir: [---.-deg] Rain: [-----.-mm] UVidx: [--.-] Light: [--.-klx] 
`

matthias-bs, 2025-01-27:
Please try https://github.com/matthias-bs/BresserWeatherSensorReceiver/tree/20250127-feat-bresser8in1

Fortunately, the 8-in-1 sensor uses the 7-in-1 protocol. I added the new type definition and put the first message of your log file into examples/BresserWeatherSensorTest.ino and got:

```
Id: [    6241] Typ: [D] Ch: [0] St: [0] Bat: [Low] RSSI: [ -56.0dBm] Temp: [ 15.5C] Hum: [ 94%] Wmax: [ 7.0m/s] Wavg: [ 5.7m/s] Wdir: [228.0deg] Rain: [   23.2mm] UVidx: [0.0] Light: [2.9Klux]
```

Do those values make sense?

To proceed further, we need the actual values shown on your display alongside with the log file. If the decoded values above are correct, we still need to find and decode the additional signal (heat index?).

jpdias, 2025-01-27:
Hello @matthias-bs,

That's amazing, I will try this out asap, the values seem to make perfect sense and match the display.

Regarding the missing value it seems to be the Heat Index as per the manual. I will send you some photos of the dashboard with the data too to confirm this. 

Thanks again for the effort.

jpdias, 2025-01-27:
Hello @matthias-bs , did you pushed your changes into the branch? I'm not seeing any new commit

matthias-bs, 2025-01-27:
@jpdias My bad, I forgot to push my changes this morning. Here we go: https://github.com/matthias-bs/BresserWeatherSensorReceiver/tree/20250127-feat-bresser8in1

matthias-bs, 2025-01-27:
Ah, excellent!
I assume that the sensor transmits the "globe thermometer temperature" as an additional value. The 8-in-1 sensor has a black bulb on top, which the other units do not have.

According to https://www.omnicalculator.com/physics/wet-bulb#how-to-calculate-the-wet-bulb-temperature, you can calculate the wet bulb globe temperature from a temperature sensor exposed to sunlight (inside the bulb, T<sub>g</sub>), another temperature not exposed to sunlight (T) and the humidity.

Also see https://www.weather-above.com/blackball%20sensor.html and https://arduinoproyek.wordpress.com/thermohygrometer-by-wet-and-dry-thermometer/.

Looking at your previous data, the additional temperature could be encoded in bytes 22 and 23 (only upper 4 bytes). This would have been T<sub>g</sub>=14.9°C as compared to T=15.5°C. With little sun and considering some tolerance, this could make sense.

Today's reading 0x00E could mean "invalid", because the light was very low at 518lx.

jpdias, 2025-01-27:
I'll test this as soon as possible and give you feedback. Thanks again for addressing this so quickly!

jpdias, 2025-01-27:
Just had some minutes to deploy this and seems to be working. One thing that I find weird, how is there a value for barometric pressure on the screen but it is not transmitted? Or maybe that last value is the pressure?

matthias-bs, 2025-01-28:
The pressure sensor is integrated in the display unit! See manual - **9.1 Console**: _Barometer (Note: Data detected by console)_.

jpdias, 2025-01-28:
Uh, ok! Then I think that's it, everything seems to be working!

closed 2025-01-28 by jpdias

matthias-bs, 2025-01-29:
@jpdias Hi, please see https://github.com/matthias-bs/BresserWeatherSensorReceiver/pull/225. It would be great if you could check if the calculated WBGT (if available) matches the value shown on the console.

matthias-bs, 2025-01-30:
Sure! To be clear: The MQTT examples (excluding the Domoticz variant) should provide the calculated WBGT value if the globe thermometer temperature provides valid data, i.e. there is some daylight and maybe the air temperature is within a certain range.

jpdias, 2025-03-11:
Hello @matthias-bs . I just had time to check (and finally good weather) to see the wbgt values

Temp: 16.0 C |Wind: 1.2m/s | Gust: 1.2m/s | Dir: 295.0m/s |Humidity: 62% |Rain(1h):     0.0 mm | Rain(1d):     8.8 mmDewpoint: 8.7 C |t_wet: 11.6 C |wbgt: 13.2 |UV: 5.5 Klux: 97.9 

But the reception unit reports ~15.9 for wbgt. 

Another example

Temp: 17.0 C |Wind: 0.0m/s | Gust: 0.0m/s | Dir: 276.0m/s |Humidity: 63% |Rain(1h):     0.0 mm | Rain(1d):     8.8 mmDewpoint: 9.9 C |t_wet: 12.7 C |wbgt: 14.0 |UV: 2.4 Klux: 25.8 

Reports on the screen ~ 17.3

matthias-bs, 2025-03-11:
@jpdias Thanks for the update! Could you please provide `tglobe_c` alongside those other values?

jpdias, 2025-03-11:
The tglobe_c closely matches the wbgt value on the screen.
Temp: 17.8 C |Wind: 1.0m/s | Gust: 1.0m/s | Dir: 234.0m/s |Humidity: 53% |Rain(1h):     0.0 mm | Rain(1d):     8.8 mm | Dewpoint: 8.1 C | t_wet: 12.2 C | tglobe_c: 17.7 C | wbgt: 13.8 | UV: 4.1 Klux: 75.2 

screen gives 17.4

matthias-bs, 2025-03-11:
This might be a coincidence with the current data (I did not check this yet) or  the sensor already does the calculation internally and actually transmits `wbgt` instead of `tglobe_c` (as assumed)?

Could you please compare a few more data points?