matthias-bs/BresserWeatherSensorReceiver #227: autodiscovery handles negative integer id wrong

alexxx-k, 2025-02-16:
**Context**
My sensor is a Bresser 5in1 that uses the 6in1 Protocoll

**Describe the bug**
The autodiscovery message is

{
  **"stat_t": "+/+/RTL_433toMQTT/Bresser-6in1/0//2059337613",**
  "dev_cla": "precipitation",
  "unit_of_meas": "mm",
  "name": "Rain",
  "uniq_id": "Bresser-6in1-0--2059337613-rain_mm",
  "val_tpl": "{{ value_json.rain_mm | is_defined }}",
  "stat_cla": "measurement",
  "device": {
    "ids": [
      "Bresser-6in1-0--2059337613"
    ],
    "cns": [
      [
        "mac",
        "Bresser-6in1-0--2059337613"
      ]
    ],
    "mdl": "Bresser-6in1",
    "name": "Bresser-6in1-0--2059337613",
    "via_device": "OMG_lilygo_rtl_433_ESP_FSK"
  }
}

the message is then published to topic:
home/OMG_lilygo_rtl_433_ESP_FSK/RTL_433toMQTT/Bresser-6in1/0/-2059337613

manually fixing the autodiscovery message fixes the issue.

**Expected behavior**
handle negative integer values in ids correctly or maybe the negative value itself is an integer overflow

**Embedded Device (please complete the following information):**
Running on LilyGo_SSD1306
using [Theengs OpenMQTTGateway v1.8.1]
https://docs.openmqttgateway.com/prerequisites/boards/lilygo-rtl_433-fsk.html

matthias-bs, 2025-02-17:
You wrote that you are using OpenMQTTGateway, but you created an issue in BresserWeatherSensorReceiver. I don't see that OpenMQTTGateway is using BresserWeatherSensorReceiver. Could you please explain how this issue is related to BresserWeatherSensorReceiver?

alexxx-k, 2025-02-17:
Hi, thanks for the response. I thought BresserWeatherSensorReceiver is used by rtl433 what again is used by OpenMQTTGateway.

If not forget about this issue, i will open a issue in OpenMQTTGateway then.

Sorry for th confusion.
Greets
Alex

closed 2025-02-17 by matthias-bs